felis/USB_Host_Shield_2.0 #219: Arduino 101 board issue

KitPan, 2016-03-18:
When using with Arduino 101, running board_qc example, it always return invalid value FF after Reading REVISION register....
tried change the clock divider and spimode, does not work, same code works fine on UNO, would you mind give some hint on making the intel curie compatible with the shield? any help will be highly appreciated!

Thanks!

xxxajk, 2016-03-18:
Is this an official USB Host Shield board?

KitPan, 2016-03-19:
I guess not, brought from a local dealer. is this a hardware compatible issue? if so I could  managed to get an official one

xxxajk, 2016-03-19:
Yes. Other vendors are UNO only and/or use different pins making them
incompatible. You also may want the 3.3v version for this board. In my own
experience with 3.3v boards, using 5v shields on 3.3v boards, even if 5v
tolerant, sometimes causes communications problems to the shield. Not
always, though. It is kind of hit-and-miss.

KitPan, 2016-03-19:
Thanks xxxajk ! I had checked the board,it seemed exact same with the official one, and passed test on 3.3v boards Due and M0, I will digging the codes to see if it is a software issue.

xxxajk, 2016-03-19:
Yeh, it really depends. The other thing is that nobody on the team has this
specific board, so you may be in for a few surprises.

Lauszus, 2016-03-21:
@KitPan have you published your modified library somewhere? It's hard to help you when I can't see what you have changed.

KitPan, 2016-03-22:
@Lauszus Thanks! I have create a fork and publish my changes in it at :https://github.com/KitPan/USB_Host_Shield_2.0, really appreciated.

oxesoft, 2017-01-18:
I'm trying to use [USB Host Shield](http://en.keyes-robot.com/productshow.aspx?id=137) with Android 101 after applying the relevant changes made by @KitPan that I put [HERE](https://gist.github.com/oxesoft/2e4b22660f163a0574e89f8fc3b02f69). The shield has exactly the same schematic of the [official](https://www.circuitsathome.com/usb-host-shield-hardware-manual) one and it works perfectly with Arduino Uno. Do you guys have some tips about what could I do?

mjs513, 2017-01-28:
hi oxesoft.  I am having  the same issue but I think one of the problems may be the power configuration.  I am using a BT dongle attached to the host shield and on a light illuminates when I connect the shield to the Uno and it does not on the Arduino101.  So I am not sure if its the lib or the board with the 101.  I am thinking about configuring it for 3.3v operation and powering the vbus from an external power source.  Not sure if this is visible but I think it is.  I may give it a try with my mini shield but I have to pull it out of another application I am using it on.  Any comments would be appreciated.

xxxajk, 2017-01-29:
Yes, vbus should be 5v, it trips up many people. I have no idea why Oleg designed it this way.
Some people can run it in the 5v config on 3.3v boards, some can't. it depends on the board and the capabilities of the board. Some boards are 5v tolerant, these usually work fine, but not always.

Also can depend on the chip used, and the board layout.

Other things to try:
Since the 101 is 3.3v, and even though it is 5v tolerant, you may want to try modified to the 3.3v config.
Lower the SPI speed in the library, some boards have awful SPI bugs at high speeds.
Make sure that there are no pin conflicts.

xxxajk, 2017-01-29:
One other hint--- use an external supply, not just USB supply from your PC.

mjs513, 2017-01-29:
Thanks for getting back to me this.  Thank god I am a glutton for punishment.  Anyway I think I put into the 3.3v config (the jumpers match whats in the picture with the boost suppy).  I also tried lowering the SPI Clock speed in the library as well.  Don't think there are any pin conflicts as far as I can see.  In all cases still received a revision error with 09 sometimes a 00.

By the way I put 5v on vbus from an external power supply (5v) and attached gnd to gnd.  Well the thing started to smoke.  So I don't know if I fried the board or not.  Still seems to work and nothing got hot..

Just funny that the min works fine.  By the way the board looks identical to the official one even down to the layout.  Just funny how the mini works but not the full size shield.  May just make a small shield to put the mini on and be done with it although, like I said I am a glutton for punishment.

mjs513, 2017-02-04:
Hi all just wanted to give you all an final update.  I received another Keyes host shield and no matter what I tried I could not get it work.  I finally received the OFFICIAL host shield board today and guess what it worked out of the box (after I soldered the headers on :)) of course.

Thanks for every ones help.

joebowbeer, 2017-02-04:
@KitPan Can you make a PR to add your changes for the Arduino 101 (ARC)?

xxxajk, 2017-02-05:
@mjs513 Not surprised by that at all. Their design is completly different, and you can't make it work without major rework efforts. Thanks for helping to support the library by buying an official board.

mjs513, 2017-02-06:
Thank you. Only reason I got the keyed was because I was waiting for the official one. Thanks for the lib

mjs513, 2017-02-07:
Suggest you hold off on incorporating the Arduino 101 update.  While I tested it the 101 I did not do a regression test, e.g., run the sketch on teensy 3.5 for instance, until now.  When I change the board back to the Teensy 3.5 I get the following error:

```
Arduino: 1.8.1 (Windows 10), TD: 1.35, Board: "Teensy 3.5, Serial, 120 MHz, Fast, US English"

In file included from C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0/Usb.h:34:0,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0/BTD.h:21,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0/BTHID.h:21,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0/PS4BT.h:21,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0\examples\Bluetooth\PS4BT\PS4BT.ino:7:

C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0/avrpins.h:1249:2: error: #error "Please define board in avrpins.h"
 #error "Please define board in avrpins.h"
  ^

In file included from C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0/Usb.h:36:0,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0/BTD.h:21,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0/BTHID.h:21,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0/PS4BT.h:21,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0\examples\Bluetooth\PS4BT\PS4BT.ino:7:

C:\Users\CyberPalin\Documents\Arduino\libraries\USB_Host_Shield_2.0/usbhost.h:111:2: error: #error "No SPI entry in usbhost.h"
 #error "No SPI entry in usbhost.h"
  ^
```
For the Arduino Uno and Mega 2560 it compiles no problem. I am running the PS4BT example.  I will try and track down the error.  Any hints would be appreciated.

Lauszus, 2017-02-07:
@mjs513 I can compile everything just fine and so can PlatformIO, which is run for every push: https://github.com/felis/USB_Host_Shield_2.0/blob/master/.travis.yml, so I think the problem is on your side?

Are you using the newest version of the code from Github?

mjs513, 2017-02-07:
@Lauszus You are probably right.  Didn't see the latest update - should have checked.  Was using kitpans library because it wasn't incorporated yet.  I will update to the latest. Should of checked first.
Thanks
Mike

Lauszus, 2017-02-07:
No problem ;) I will close this issue now, as I believe the problem has been solved.

closed 2017-02-07 by Lauszus

Lauszus, 2017-02-07:
@mjs513 I will tag a new release once #281 is merged, then you will no longer have to manually install the library and can just use the Arduino library manager: https://www.arduino.cc/en/Guide/Libraries.