felis/USB_Host_Shield_2.0 #223: Could not detect the shield when: Android reboot and unplug/replug

KitPan, 2016-03-23:
Hi Guys, have you meet these issues?
A. reboot Android with shield attached, the debug output is:
`Disconnected from accessory
ADK Init
Addr:01

ADK protocol rev. 0002

Accessory mode switch attempt
ADK Init
Addr:01

ADK protocol rev. 0002

Accessory mode switch attempt
ADK Init
Addr:01

Acc.mode device detected
Endpoint Structure:
EP0:
Addr: 00

Max.pkt.size: 40

Attr: 3C

Epout:
Addr: 02

Max.pkt.size: 40

Attr: 04

Epin:
Addr: 01

Max.pkt.size: 40

Attr: 06

Configuration successful
ADK Init
Address in use
Connected to accessory
Data rcv: D
Data rcv: D
..........
`
B. when unplug and then replug the cable,the debug output is:
`ADK Init
Addr:01

Acc.mode device detected
Got 1st bad code for config: 0D

getConf 0D

ADK Init
getDevDescr 0D

ADK Init
getDevDescr 0D

ADK Init
getDevDescr 0D
`
thinking this may due to the timing issue when android reset to accessory mode, would you mind give some guidance or advice on resolve this?

xxxajk, 2016-03-23:
0x0D should be hrJERR.
This particular error due to the peripheral expecting a different
line-assertion than what was sent by the MAX3421E. Specifically a J-state
of the D+/D- lines was seen instead of a response bit pattern. The J and K
states are defined based on the USB configuration (opposite D+/D- for Full-
versus Low- speed USB).

Hi Guys, have you meet these issues?
A. reboot Android with shield attached, the debug output is:
`Disconnected from accessory
ADK Init
Addr:01

ADK protocol rev. 0002

Accessory mode switch attempt
ADK Init
Addr:01

ADK protocol rev. 0002

Accessory mode switch attempt
ADK Init
Addr:01

Acc.mode device detected
Endpoint Structure:
EP0:
Addr: 00

Max.pkt.size: 40

Attr: 3C

Epout:
Addr: 02

Max.pkt.size: 40

Attr: 04

Epin:
Addr: 01

Max.pkt.size: 40

Attr: 06

Configuration successful
ADK Init
Address in use
Connected to accessory
Data rcv: D
Data rcv: D
..........

B. when unplug and then replug the cable,the debug output is:
ADK Init
Addr:01

Acc.mode device detected
Got 1st bad code for config: 0D

getConf 0D

ADK Init
getDevDescr 0D

ADK Init
getDevDescr 0D

ADK Init
getDevDescr 0D
`
thinking this may due to the timing issue when android reset to accessory
mode, would you mind give some guidance or advice on resolve this?

Lauszus, 2016-03-23:
@KitPan so this means that issue https://github.com/felis/USB_Host_Shield_2.0/issues/219 is resolved? Have you tried it with any other device than a Android phone?

KitPan, 2016-03-23:
@Lauszus still no lucky on issue #219, so I am switching to a Arduino.org M0 pro board. did not tried with other device, since the debug output indicate the error is during adk init, still looking at the adk part

KitPan, 2016-03-23:
@xxxajk Thanks! thinking this may mainly due to the adk protocol implement still working on it

KitPan, 2016-03-24:
Finally find a way to deal with the android device reboot problem! 
take a note here incase someone may run in the same problem. 
when you reboot your android device with the shield attached, after android boot up android will never detect the shield unless you push the reset button or unplug/replug. 
during the android reboot process it will make the Vbusstate change to SE0(disconnect) unfortunately this Vbusstate change is taking place during the shield doing the "Accessory mode switch attempt" in adk.cpp, this will trigger the USB_ERROR_CLASS_INSTANCE_ALREADY_IN_USE error 0xD9, tried to do device release but dose work, it seemed the interruption of the acc mode switching makes the android and arduino missing sync in usb state.
and here is the easiest and brute solution: trigger a reset by code when detect 0xD9 error! my board is a M0, could use sysresetreq to perform a reset, should work in all cortex-m boards like due. declare:
`#define SYSRESETREQ    (1<<2)
#define VECTKEY        (0x05fa0000UL)
#define VECTKEY_MASK   (0x0000ffffUL)
#define AIRCR          (*(uint32_t*)0xe000ed0cUL) // fixed arch-defined address
#define REQUEST_EXTERNAL_RESET (AIRCR=(AIRCR&VECTKEY_MASK)|VECTKEY|SYSRESETREQ)`
then call REQUEST_EXTERNAL_RESET;
if you have any better solution please let me know. Thanks for all you help on this!

xxxajk, 2016-03-24:
Install the VBUS power switch then.
https://www.circuitsathome.com/mcu/vbus-power-control-on-usb-host-shield

Lauszus, 2017-10-12:
A VBUS power switch should indeed fix this issue. Please reopen if needed.

closed 2017-10-12 by Lauszus