felis/USB_Host_Shield_2.0 #224: Problem with a PS4 controller [USB host library]

Fringergueber, 2016-03-31:
Hello arduino friends;

I've been trying to conect a ps4 controller to an arduino, and I successfully achieve that objective with a USB cable. The thing is that I am not able to conect it by bluetooth and I would like to ask you for help.

I'm using a USB host, a bluetooth dongle and the USB host library 2.0, first of all, I'm the code i'm working with is the PS4BT example. With that being said I will describe you what I've done and what I've tried.

The first thing I did was to plug the dongle in the USB host and see what message the console showed, I only got this message:

PS4 Bluetooth LibraD: 0A5C PID: 4500
BTD Init Failed, error code: D1

PS4 Bluetooth Library Started
Unknown Device Connected - VID: 0A5C PID: 4500
BTD Init Failed, error code: D1

Reading the code I found this "//USBHub Hub1(&Usb); //Some dongles have a hub inside", I deleted the first "//" so the code wasn't undestood as a comment and then, when I uploaded to the arduino, I got this message in the console:

PS4 Bluet
Unknown Device Connected - VID: 0A5C PID: 4502
BTD Init Failed, error code: D1

PS4 Bluetooth Library Started
Unknown Device Connected - VID: 0A5C PID: 4502
BTD Init Failed, error code: D1

Unknown Device Connected - VID: 0A5C PID: 4503
BTD Init Failed, error code: D1

Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:0A:C4:C3
Please enable discovery of your device
Couldn't find HID device
Wait For Incoming Connection Request

Then I have tried to connect the PS4 controller by holding down the PS and Share button at the same time but nothing happened, I have also tried to disconnect the dongle, connect the PS4 controller by USB and then disconnecting the controller and connecting the dongle but I only got this message:

Unknown Device Connected - VID: 054C PID: 05C4
BTD Init Failed, error code: D1

Do you have any idea fix it? Thank you and sorry for my grammar, english is not my native language.

Lauszus, 2016-04-14:
@Fringergueber can you try to run the following example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USB_desc/USB_desc.ino and plug in your dongle, then please post the output here.

Did you following the instructions here: https://github.com/felis/USB_Host_Shield_2.0#ps4-library in order to put the PS4 controller into pairing mode?

harshaperni, 2016-12-12:
Hello,
I am also getting similar error..
Initially I uploaded the PS4BT sketch with "PS4BT PS4(&Btd, PAIR)" and go the below message:

PS4 Bluetooth Library Started
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1B:10:00:2A:EC
Please enable discovery of your device
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Connecting to HID device
Connected to HID device
Received Key Request
Bluetooth pin is set too: 0000
Pairing successful with HID device
Send HID Control Connection Request

Later uploaded the sketch PS4BT sketch with "PS4BT PS4(&Btd)" and go the below message:

PS4 Bluetooth Library Started
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1B:10:00:2A:EC
Wait For Incoming Connection Request

I tried holding PS button and Share+PS button but couldn't connect.
I then have uploaded the USB_desc and got the below message:

Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		E0
Device Subclass:	01
Device Protocol:	01
Max.packet size:	40
Vendor  ID:		0A12
Product ID:		0001
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		00D8
Num.intf:		04
Conf.value:		01
Conf.string:		00
Attr.:			E0
Max.pwr:		00

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		03
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0010
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Endpoint descriptor:
Endpoint address:	02
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			02
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			03
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			04
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			05
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		FF
Intf. Protocol:		FF
Intf.string:		00

Endpoint descriptor:
Endpoint address:	84
Attr.:			02
Max.pkt size:		0020
Polling interval:	01

Endpoint descriptor:
Endpoint address:	04
Attr.:			02
Max.pkt size:		0020
Polling interval:	01

Interface descriptor:
Intf.number:		03
Alt.:			00
Endpoints:		00
Intf. Class:		FE
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00
Unknown descriptor:
Length:		07
Type:		21
Contents:	0588134000AF00

Addr:1(0.0.1)

Please help me understand where the problem is ?

Thanks,
Harsha

Lauszus, 2017-01-23:
@harshaperni it looks like they might have changed something since the first release of the DS4 controller. I will close this issue as a duplicate of #273, please move further discussion to that issue.

closed 2017-01-23 by Lauszus