felis/USB_Host_Shield_2.0 #225: Weird performance of usb shield suddenly!

mamech, 2016-04-05:
hello

I used the testusbhostFAT for around 1 year and it works great with my arduino + usb shield. I took the code and I made some modifications to suit my interest, and it also worked fine with me. but somedays ago, when I was trying the arduino and the shieldin some circuit, I noticed that my code does not behave as usual. I bean to suspect my code, so I uploaded again testusbhostFAT, and I found also weird performance (really I do not know what changed!)

in the past, the arduino showed me messages related to opening a file named "message.txt" that exist in usb flash, and creating another file in the flash named "Hello.txt". but now this does not happen. currently, this is what I get on serial monitor:
Start
Current UsbDEBUGlvl 81
'+' and '-' increase/decrease by 0x01
'.' and ',' increase/decrease by 0x10
't' will run a 10MB write/read test and print out the time it took.
'e' will toggle vbus off for a few moments.

Long filename support: Enabled
Total EXT RAM banks 0
Available heap: 7124 Bytes
SP 21ef
Available heap: 7097 Bytes
VBUS on
USB state = 20
USB state = 40
USB state = 50
USB state = 51
USB state = 90
Available heap: 6650 Bytes
Available heap: 6650 Bytes
Available heap: 6650 Bytes
Available heap: 6650 Bytes
Available heap: 6650 Bytes
Available heap: 6650 Bytes

and it keeps showing more and more without showing anything from the flash memory!

I suspected that my flash is defective, and i replaced it, but the result is the same.

does the above mentioned behavior means that my shield became defective for some reason??

xxxajk, 2016-04-05:
try contact cleaner first.

xxxajk, 2016-04-05:
Oh and see if it happens in UHS3.0, which is better anyway.

mamech, 2016-04-05:
thank you for your reply. regarding to UHS 3.0 , it will not be good in my application , because my application is based on making converter from Flash memory usb stick to rs232 using arduino and max232. 

regarding your first reply, you said : "try contact cleaner first."
sorry, I did not understand. may you please explain to me what it the cleaner that I need to contact first?

thanks for your effort

xxxajk, 2016-04-05:
CRC QD Contact cleaner, if you can get it works very well.
http://delivr.com/2fcd4_qr

mamech, 2016-04-05:
I do not have this kind of cleaner currently, so I used a kind of cleaner that is used for LCD. but still I am getting same message.

so from your answer, I understand now that this problem seems to be a problem of dirty/rusty usb connector of the shield? if I desolder the usb connector of the shield and replace it, this may solve the problem??

xxxajk, 2016-04-05:
It might but I would try cleaning first because it is easier and usually
solves the problem for me. Sometimes it is a tension issue and that would
require replacement of the socket.

mamech, 2016-04-05:
thank you for your suggestions , I will test and see what is the cause.

one more question please. how did you determine that it is a connector problem? do the numbers of "USB State = ##" have a meaning? for example, I noticed that if i unplug the flash usb stick, I get "USB State = 12".
 what is the meaning of these numbers?

xxxajk, 2016-04-05:
Partially, The code means that it had an electrical problem.
I have had it happen to me many times.
The usual causes I have seen are:
- Oxidation on the contacts
- Socket tension/retainers out of spec.
- Finally rare situation of a bad solder joint on 33ohm resistor.

xxxajk, 2016-04-05:
Error codes are listed here:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/max3421e.h#L206

xxxajk, 2016-04-05:
State codes are here:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/UsbCore.h#L106

mamech, 2016-04-05:
thanks for the codes. I see this now:
#define USB_ATTACHED_SUBSTATE_SETTLE                        0x20
#define USB_ATTACHED_SUBSTATE_WAIT_RESET_COMPLETE           0x40
#define USB_ATTACHED_SUBSTATE_WAIT_SOF                      0x50
#define USB_ATTACHED_SUBSTATE_WAIT_RESET                    0x51
#define USB_STATE_RUNNING                                   0x90

these are the 5 numbers that appear to me, and as the names indicate, they do not show an error. the last USB state that I see is 90, which indicates "USB_STATE_RUNNING  " as shown above. how does this indicate a missing contact/oxidized pin?

I ask this question, because I want to handle this electrical connection error in my code, so the support team may be aware that the connector is in need for replacement.

xxxajk, 2016-04-06:
Doesn't, but there was other errors. Is the usb stick formatted as FAT32?
Legacy partition table and full device formatted are supported.
NTFS is _NOT_ supported.

mamech, 2016-04-06:
yes sure, it is formatted as FAT32

xxxajk, 2016-04-06:
is there no volume label?

mamech, 2016-04-07:
the volume label that appears on computer is ADATA UFD (J:)

xxxajk, 2016-04-07:
Erase the label, and it should work and mount on '/'.

mamech, 2016-04-07:
I erased the label so the windows replaced it automatically to "Removable Disk (J:)".
sorry, I did not understand "mount on '/' " . how can I do this?

xxxajk, 2016-04-07:
Use the shell (command) and do this manually.
See here:
https://technet.microsoft.com/en-us/library/bb490925.aspx

mamech, 2016-04-07:
I have read, and I think I did what is needed , and still the result is the same.
actually, I used the flash usb stick without any modifications or without removing the label in the past, and it worked perfectly with the usb shield, so I think that may be it is a hardware problem as you said in beginning, but I am not sure ofcourse.

mamech, 2016-04-08:
Update: I asked in Edaboard site about how to test the meshing between male and female usb connectors, and they told me a good suggestion. I cut an old usb cable with male terminal and I plugged it into the usb female terminal of the shield, and then I used the multimeter to be sure that every wire in the usb is connected to a pin in the shield. The strange thing now, is that all pins are connected to wires, which means that our initial guessing about the dirty/rusty usb female terminal in the shield was not correct.

I am totally confused now... what is the problem? it was working very well short time ago! 

do not you have any suggestions to me?

xxxajk, 2016-04-08:
Does it work on other devices? Have you run the quality control program?

xxxajk, 2018-05-24:
Closed due to inactivity.

closed 2018-05-24 by xxxajk