felis/USB_Host_Shield_2.0 #228: Issue with setLedOn() in PS3USB (works in PS3BT)

bbf3, 2016-04-20:
Hello,
I'm having an issue turning on the PS3 controller's LEDs when in USB (corded) mode. Using just the examples provided (PS3BT and PS3USB), it seems like my controller can turn LEDs on and off when using the PS3BT example, but not when using the PS3USB example program. 

An exception: If I add an "attachOnInit" in my main loop:
`void loop() { // begin main loop
  Usb.Task(); 
  PS3.attachOnInit(USBonInit); // this added

...main loop code...`
Then after my main loop: 
`...main loop code...
} // end of main loop

void USBonInit(){  // runs only once at moment of controller initialization
  Serial.print(" JUST CONNECTED ");
   PS3.setLedOn(LED3);
}`
...then I can make an LED (#3) turn on. This will only happen if I unplug the controller and replug it back in. It won't happen on first programming the Arduino if the controller is plugged in, or by resetting the arduino while keeping the controller plugged. Additionally, pressing one of the D-pad keys (to change an LED) makes the LED go out and turns none of them back on. 

Changing the Arduino program back to the PS3BT example fixes the issue and the LEDs can be programmed by the D-pad buttons as intended. I've switched back and forth between the two examples multiple times and have the same issue with PS3USB each time. 

Any ideas? Thank you for all the effort that clearly went into this hardware and software. 

Blake

Lauszus, 2016-04-20:
@bbf3 can you try to revert this commit: https://github.com/felis/USB_Host_Shield_2.0/commit/627c0ebc777c43ea452eb5508a9c5372e2382cda and see if it fixes the issue?

bbf3, 2016-04-21:
Hey,
So I've only been on Github less than 24 hours and am trying to understand what you mean. I went to the commit you specified (a branch in the code, right?), downloaded the PS3USB.cpp file from that branch, and put it into the host shield 2.0 library on my local system, overwriting the previous file. I tried the program again with the same results: no LEDs except for LED3 programmed in the attachOnInit when the cord is unplugged/replugged. 

Maybe I'm not understanding what exactly you want me to try? Should I download the entire library from that branch and replace every file? I'm reading about Github and reverting commits but I'm a little confused so far as a new user. 

Thanks,
Blake

bbf3, 2016-04-21:
After further reading up on this, I removed the current library (ver. 1.2.1) and installed the 627c0eb commit (the whole library at that commit, not just the file in question. 

The commit did not resolve the issue. LEDs still do not work when controller is plugged in, except when controller is unplugged, then replugged, and only at that time does LED1 illuminate. Pressing D-pad buttons makes the LED go out and none turn on. 

Thanks,
Blake

Lauszus, 2016-04-25:
@bbf3 you need to try the code before that commit. I.e. please try the following version: https://github.com/felis/USB_Host_Shield_2.0/tree/0fce0106ed4005a6f9611e7990b3b471168cc8db.

bbf3, 2016-04-26:
Got it. Here are the commits I've now tried (downloading the whole library and replacing into the arduino library folder each time). You can see which one's worked and which did not. The .... means there are skipped commits, ones that are next to each other are consecutive in the master branch. Seems like they LEDs worked, then didn't, then worked again, and now do not. 

627c0eb - jan 19, 2016 - not working 
0fce010 - jan 18, 2016 - not working (the commit you wanted me to try)
969eabb - jan 16, 2016 - not working
....
fb723fc- nov 17, 2015 - not working
....
fe12f0e - nov 7, 2015 - not working
.....
7e449d2 - oct 12, 2015 - not working
acfcba6 - oct 12, 2015 - not working
523e66e - sept 3, 2015 - not working
....
6d88254 - jul 23, 2015 - not working
....
20596ee - may 27, 2015 - not working
....
1f3072b - apr 20, 2015 - not working
....
4e80db2 - mar 19, 2015 - not working
....
d881e51 - mar 19, 2015 - not working
....
31e1c25 - mar 19, 2015 - not working
a2265ea - mar 19, 2015 - LEDS Work (triple-checked)
.....
46dc4e6 - mar 3, 2015 - LEDs work
......
bde1103 - sep 2, 2014 - LEDs work
....
d35b83e - aug 20, 2014 - LEDs work
....
900af96 - aug 4, 2014 - LEDs work
....
26cb553 - aug 4, 2014 - LEDs work
963c157 - aug 3, 2014 - not working
e132b5f - aug 3, 2014 - not working
....
7d3154b - aug 2, 2014 - not working
....
45df706 - aug 2, 2014 - not working
f349d61 - jul 26, 2014 - LEDS Work (triple checked)
.....
35bb596 - jul 1, 2014 - LEDs work
.....
983a7d6 - feb 25, 2014 - LEDs work
....
37d59a6 - feb 17, 2014 - LEDs work
....
f1af9b0 - dec 25, 2013 - LEDs work

Please advise.

Thanks,
Blake