felis/USB_Host_Shield_2.0 #230: Merged PS3BT and PS3USB example program

bbf3, 2016-04-21:
I don't know if this will help anyone, but I merged the PS3BT and PS3USB examples so that one program can run the controller wired and wirelessly. It's a bit long (I'm pretty new to Arduino), but it works (other than the USB-only LED control issue previously mentioned). This is hot swappable by just unplugging your bluetooth dongle and plugging in the controller with a usb cord. It also has commented sections for code to execute when there's no controller and when either controller is first plugged in. I removed the Move controller sections for clarity.

I hope this will help someone, and I'm sure someone knows a better way to do it than I did.

Thanks,
Blake

 [PS3-merged_example.zip](https://github.com/felis/USB_Host_Shield_2.0/files/230179/PS3-merged_example.zip)

Lauszus, 2016-04-21:
A smarter way would be to create a common PS3 controller class and then let both the PS3USB and PS3BT inherit that, then one could use the same API for both classes directly.

The two drivers could then be stored in a array like it is done in the [PS3Multi](https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Bluetooth/PS3Multi/PS3Multi.ino#L21) example.

Anyway that is just a suggestion. You can look into it if you want to.

In fact a common "controller" class that has the basic functionality like getButtonPress, getButtonClick etc would be useful for all the different game controller drivers.

closed 2016-04-21 by Lauszus

d33pend, 2016-04-26:
Hi, me too need to merge them, but i'm also still at the beginning, please can you give us some more help in detail ?

Lauszus, 2016-04-28:
@d33pend you can do something like I did for the PS4 driver: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS4Parser.h, https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS4Parser.cpp, https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS4USB.h and https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS4BT.h.

I was planning to do that for all drivers, but I simply don't have the time.

d33pend, 2016-04-28:
@Lauszus I tried to read all the four files that you kindy provide me and i try to understand something but unfortunately without much success, it was like reading Japanese as i'm just at the start with Arduino :(

> I was planning to do that for all drivers, but I simply don't have the time.

Yeah, because i think it's the best thing for all to have both possibility on the same file to use the controller plugged and just remove the usb cable and use it with Bluetooth instead to have to edit/upload the files just to use one of them, so please when you'll have a little time take a look at it, thank you very much for now!

Lauszus, 2016-04-28:
@d33pend if you are just getting started with programming, then it might be easier to just use @bbf3 approach.

Unfortunately I won't have time to look into it in the nearest future.

d33pend, 2016-04-28:
Yeah, from what i have understand, it seems that i'll need just to duplicate all the code of PS3BT (the loop part with comands) to PS3USB.. It is not exactly the best, especially if you have other code in the program, but since i have not alternatives for now, i'll go with that, thanks anyway!

xxxajk, 2016-04-28:
Thankfully uhs3 will avoid this sort of problem, as it will have a unified
joystick driver, much like I did for serial devices. I'll possibly have
some play time tomorrow to add joystick stuff, and I have some devices to
test. Once I do an initial commit, I'll start a thread there, where people
can request features, etc.

d33pend, 2016-05-04:
@xxxajk Thanks Andrew, in meantime, is there a way to copy those files into the current version 2.0 until the new version comes out?

xxxajk, 2016-05-04:
No, the API is not compatible.