felis/USB_Host_Shield_2.0 #234: Problem with Wii-IRCamera

Ravi0li, 2016-04-28:
Hallo Team,

I have tested the last weeks our libarary for the Wii-Controller. On much points it works fine.
But with the IR-Camera I have a problem. I use the example "WIIIRCamera". After the initialization i become for aprx. 500ms right data from the remote. And than frozen the Values of the IR-Points. All other reports of bottoms and so works fine. I use the standard parameter for the camera initialization.
Have anybody I Idea or the same problem?

Thank you.

Ravi0li, 2016-06-05:
Today, I have debug a little bit. When I use Wii.IRinitialize() in my programm, I become only 11-13 datareports with 0x33. In this half Second are the data correct. After this time, I become only 0x31 reports with no IR-Bytes. Why switch my Wii-Controller the IR-Mode off?

Ravi0li, 2016-06-05:
I have a Solution. When I comment Line332 in Wii.cpp out, work it.

Lauszus, 2016-07-02:
Thanks for the report. Please confirm that the following commit solves the issue: https://github.com/felis/USB_Host_Shield_2.0/commit/ce7f62e8d974b514261fdcce9df9ac3c36f65c59.

You can check it by applying in manually or simply downloading the code from the following branch: https://github.com/felis/USB_Host_Shield_2.0/tree/issue234.

Lauszus, 2016-07-26:
@Ravi0li have you had time to test the code yet?

Ravi0li, 2016-07-27:
Sry, Lauszus.
Yes, it's works! Thanks!

commit ce7f62e8 (2016-07-02): Keep reading the IR camera as well if an extension disconnects

Fixes #234

Also added some debugging code, so this could be detected faster in the future
PR #251 (2016-07-31): Keep reading the IR camera as well if an extension disconnects
Fixes #234

Also added some debugging code, so this could be detected faster in the future
closed 2016-07-31 by Lauszus