felis/USB_Host_Shield_2.0 #235: Use UHS2.0 to connect to remote BlueSmirf Gold (RN-41 BT Module)

bbf3, 2016-04-28:
Hey,

I've spent the last few days trying to connect to a BlueSmirf module from SparkFun that uses the Roving Networks' RN-41 module to the UHS2.0. 

I am trying to to connect from an Arduino Mega 2560 with a UHS2.0 through a bluetooth dongle (a Trendnet UBW-106UB, which supports SPP and which I've successfully used with PS3 connections), across bluetooth to a BlueSmirf Gold. The BlueSmirf will be connected to a serial enabled LED display. I'm trying to send data to the remote LED display over bluetooth from the local arduino.
(Local Arduino and UHS2.0 --> local Trendnet dongle --> remote BlueSmirf --> remote LED display)

I've been able to configure the BlueSmirf and get it to pair with the same Trendnet dongle plugged directly into my laptop, and send information and alter settings on it through Arduino's serial terminal, and print out information to the LED display. (Arduino IDE serial terminal on laptop -->local trendnet dongle --> remote BlueSmirf --> remote LED Display)

However, I have not been able to implement this with the UHS2.0's SPP examples. Does the BlueSmirf need to be in slave or master (or pair mode)? I've tried them all. Do I set the name and pin in the SPP example for the Smirf or for the local trendnet or neither (I've tried them all)? When I set the Smirf to pair mode, it actively tries to connect with the trendnet's stored mac address. I can see the lights on the dongle change when it's trying to connect, but it never establishes a connection (the Smirf never indicates it is connected, only trying). 

Any ideas on what to try next? BlueSmirf to LED display works well, and Arduino to UHS2.0 to trendnet dongle works well, but I'm not sure why the dongle won't connect to the Smirf. It feels so close to working!

Thanks,
Blake

bbf3, 2016-04-28:
Solved this issue by setting the BlueSmirf to pair mode (command mode: SM,6), and authentication mode to open (SA,0). Seems to be working. I will update further with settings and process when up and running.

bbf3, 2016-05-02:
The UHS2.0 works with the BlueSmirf Gold for connecting serial devices over SPP bluetooth. Here's the steps involved to get a basic connection and some explanation. Such a connection can be used alongside other bluetooth connections (like using a PS3 controller) with the same bluetooth dongle. 

Hardware/software used:

    

- TrendNet TBW-106UB bluetooth dongle ([Trendnet](https://www.trendnet.com/products/proddetail?prod=170_TBW-106UB))
- BlueSmirf Gold ([BlueSmirf Gold](https://www.sparkfun.com/products/12582))
- OpenSegment Serial LED Seven Segment display ([OpenSegment](https://www.sparkfun.com/products/11646)) [could be any serial device]
- Power source ([such as this thing](https://www.adafruit.com/products/2465))
-  Laptop running windows 10 with Arduino IDE installed (ver. 1.6.8 ), and the BlueSoleil package (included with the Trendnet dongle) or just figure it out with Windows.

To get a connection up and running:

1-- The Trendnet dongle comes with a "BlueSoleil Space" software package. This helps to interface the dongle with a computer. It is probably not needed, and you could do this through windows. I will describe these steps with BlueSoleil instead of Windows' interface since it's how I did it. (When I was first trying to figure this out, I did connect directly to an XP laptop without BlueSoleil.) Plug in your dongle and make sure it works with your computer. You could setup a connection to a bluetooth speaker or phone to test it out.

2-- Solder wires to Vcc, GND, Rx and Tx on the BlueSmirf. Connect Rx and Tx together temporarily. Connect Vss to 5v and Gnd to GND. I use an Adafruit PowerBoost 1000 Charger and a Lipo to power my setup via battery or USB (not required, could be any power source). The Red LED on the BlueSmirf should blink twice a second.

Note: Sparkfun/Roving Networks says the BlueSmirf is only in Config mode for 60 seconds on startup. I did not find this to be the case, and found I could enter command mode at any time.

3-- Find the BlueSmirf in the BlueSoleil software and pair to it. Then, using the "Serial" icon along the top (third from left in my version), connect to the bluetooth serial port (first you pair with it, then you connect to its serial port). In my case it says this is COM5. Your BlueSmirf should show a green LED when connected.

4-- Open Arduino IDE. Go to Tools/ and select the same port as the bluetooth serial port (in my case, again, COM5). Then open the Serial Monitor.

5-- Set the Baud to 115200 (the default of the BlueSmirf), and the line ending type (next to baud) to "no line ending".

6-- Type "$$$" (all commands are entered with no quotes) into the command line and press enter. The red LED on the BlueSmirf will start to blink fast, indicating it is in command mode now.

7-- From command mode you change settings of the BlueSmirf using commands from the manual sparkfun provides. Make sure to change line ending type back to "newline" before entering commands. You can enter a "D" to see a list of the current settings. Type "E" to see even more.

8-- From here, I set baud to 9600 with the command "SU, 96", because my seven-segment display operates best at 9600. The BlueSmirf will print "AOK" when a setting has been changed. Other settings allow you to change the name of your device ("SN, MYDEVICE"), restore to factory defaults ("SF, 1"), etc., among other things. If you change the baud, you will need to reconnect in Arduino in the new baud after power is cycled (commands take effect after power cycling). If you mess things up and can't reconnect, there is a way to hard reset the BlueSmirf tying certain pins to ground. Its discussed in the manual. ([Manual](https://cdn.sparkfun.com/assets/1/e/e/5/d/5217b297757b7fd3748b4567.pdf))

Note:  Setting the config timer to zero ("ST, 0") actually makes it so you can't enter command mode. Sparkfun implies in their writeup that "ST, 0" makes it so that there is no timer and you can enter command mode at any time ("Another handy command, if you're lazy like me, is ST,0, which turns the config timer off"). This is not true. You will be locked out of config mode and need to do a hard reset to factory defaults if you do this (i did this). What you really want is "ST, 255" for continuous config mode. ([Sparkfun writeup](https://learn.sparkfun.com/tutorials/using-the-bluesmirf/all))

9-- When you're done with settings, power everything off. Disconnect Rx and Tx from each other and plug Tx into your display. Turn everything back on, and connect back up to the BlueSmirf using BlueSoleil or Windows (or OSX). Once a connection is established, back in the Arduino IDE, you should be able to type commands for your display directly into the command line for the display to process. For example, my display (the Sparkfun OpenSegment LED Seven Segment display) requires the line ending type to be set to "no line ending" (because it looks for line ending characters when it parses instructions). For example, to clear my screen, I type "v" and hit enter, and to print a 1, I type "1" and press enter, etc.

10-- To use with the UHS2.0, I found that by putting the BlueSmirf in pair mode ("SM, 6"), setting authentication to open mode ("SA, 0"), and by setting the remote bluetooth address to that of the dongle ("SR, bluetooth hex address"), I could force the BlueSmirf to look for and connect to the USB Host Shield on startup instead of the other way around. This allowed a connection to be established and everything to just work. I u [... cut ...]
  Serial.print(F("\r\nPS3 Bluetooth Library Started"));
}

void loop() {
  Usb.Task(); //PS3 get info command

//....main PS3BT and PS3USB code

if (SerialBT.connected) {
        SerialBT.print("v"); //write to BlueSmirf/clear LED screen
        SerialBT.print("1234"); // write to BlueSmirf/write to LED screen
  }
}
```

Hope this helps someone. I originally wrote this up in response to [this forum thread](http://forum.arduino.cc/index.php?topic=74768.0). Sorry if it is a little simplistic/off topic. 

Blake

Lauszus, 2017-10-12:
Thanks for sharing and sorry for the lack of response.

closed 2017-10-12 by Lauszus