felis/USB_Host_Shield_2.0 #236: Usb android open accessory connection not stable

GivenMojapelo, 2016-04-29:
Hi.

I have been trying to connect to android in an open accessory mode to run the BlinkLed example provided. Sometimes I am able to connect and control the led but sometimes I connect and my blinkLed example only communicates with the Serial monitor. There are also times when I just can't connect and my android activates media device connection mode.  I changed the delay times in adk.Init() and I still got nothing. I tried different android os versions as well

Any ideas what the problem can be and how I can solve it will be appreciated. 

Thank you!

Lauszus, 2016-05-01:
@GivenMojapelo can you test if this issue solves your problem: https://github.com/felis/USB_Host_Shield_2.0/issues/238?

GivenMojapelo, 2016-05-03:
Hi @Lauszus. I tried that and I am not getting anything. The android device still goes straight to media device mode.

GivenMojapelo, 2016-05-03:
I have turned debugging on my serial monitor looks like this :

[image]

YuuichiAkagawa, 2016-05-03:
@GivenMojapelo 
I think that delay time is not enough.
It take time to accessory mode switching in the some of phone/tablets.

https://github.com/felis/USB_Host_Shield_2.0/blob/master/adk.cpp#L256
Please try to set a large value. (More than 500)
In my phone it is necessary to set the 1000.

@Lauszus
I found other issue.
ADK ID Strings is cleared after calling sendStr() function.
From the second time on, it worn't be valid stirng.
(When USB cable disconnect and reconnect)

I think this is better.
```C++
/* send ADK string */
uint8_t ADK::sendStr(uint8_t index, const char* str) {
        char tmpstr[256];  //ID string is upto 256 bytes.
        strncpy(tmpstr, str, 255);
        tmpstr[255] = '\0';
        return ( pUsb->ctrlReq(bAddress, 0, bmREQ_ADK_SEND, ADK_SENDSTR, 0, 0, index, strlen((const char *)tmpstr) + 1, strlen((const char *)tmpstr) + 1, (uint8_t*)tmpstr, NULL));
};
```

### USB packet capture
* First time
[image]

* Second time
[image]

### Debug log
* Code
https://gist.github.com/YuuichiAkagawa/50308c58dd81d89606784971f0bea2d9

* Log
```
HelloADK Started
ADK Init
ADK protocol rev. 0002

Begin sendStr()
ACCESSORY_STRING_MANUFACTURER: ammlab.org, 0x1b4
ACCESSORY_STRING_MODEL: HelloADK, 0x1ab
ACCESSORY_STRING_DESCRIPTION: HelloADK for Arduino, 0x196
ACCESSORY_STRING_VERSION: 1.0, 0x192
ACCESSORY_STRING_URI: https://play.google.com/store/apps/details?id=org.ammlab.android.helloadk, 0x148
ACCESSORY_STRING_SERIAL: 0000000012345678, 0x137

End sendStr()
ACCESSORY_STRING_MANUFACTURER: , 0x1b4
ACCESSORY_STRING_MODEL: , 0x1ab
ACCESSORY_STRING_DESCRIPTION: , 0x196
ACCESSORY_STRING_VERSION: , 0x192
ACCESSORY_STRING_URI: , 0x148
ACCESSORY_STRING_SERIAL: , 0x137

Accessory mode switch attempt
ADK Init
Acc.mode device detected
Configuration successful
Connected to accessory
Data rcv: D
Disconnected from accessory
```

GivenMojapelo, 2016-05-04:
Thank you @YuuichiAkagawa and @Lauszus. Both your answers helped. I also tried @YuuichiAkagawa 's reconnection method but it's also not working. I have to reset the board everytime  I reconnect.

Lauszus, 2017-10-12:
Reopening this issue, as we should investigate https://github.com/felis/USB_Host_Shield_2.0/issues/236#issuecomment-216597302 at some point.

horitahiroshi, 2018-04-06:
Hi!

I got the same problem.
It's happening the same thing as described by Freezerohmatic on issue #245.

I've tried all the above suggestions, but still didn't solve.
Have anyone gotten a solution for this issue?

xxxajk, 2018-04-06:
Educated guess here... could be the device has a bug.
Here is how to find out more about what is going on. Note that I use Linux, so whatever you are using may be slightly different.

Get the Android Development Kit (ADK) installed, and find where the adb tool is.
Connect tablet to PC using USB, and issue the following command `adb -d tcpip 5555` and disconnect the USB cable.
On the tablet go into settings -> wifi and tap the access point you are connected to and note the IP address
on the PC issue the following command `adb connect IP.FROM.THE.TABLET:5555` ,obviously replacing `IP.FROM.THE.TABLET` with the IP address from settings
on the pc issue the command `adb shell logcat`

 Now you will be able to watch what happens during connections, and a whole lot of other stuff.

Lauszus, 2018-04-09:
To be honest the ADK driver has very low priority, as it is pretty much abandoned by Google at this point.

horitahiroshi, 2018-04-10:
Thanks, @xxxajk and @Lauszus for the answer!

Actually I've already done what you suggested, @xxxajk and still have not found the problem out.
And I did some more tests with other devices here (devices with conventional USB-B micro and USB-C micro). The result is that all devices with USB-C had this problem.
So, my guess is that the problem must be the type of connector.

And as the @Lauszus said,ADK is pretty much abandoned by Google, I will find another alternative to communicate Arduino and Android.
Thank you.

thegoodhen, 2023-12-03:
I am not sure how relevant this is, but I did run into a problem where the term_test example failed to connect to the phone using an usb a to usb c cable, but connected just fine using an usb a to usb c reduction, followed by an usb c to usb c cable.