felis/USB_Host_Shield_2.0 #239: Example code can not print devices

LSChyi, 2016-05-21:
Hello, I have used an Arduino pro mini usb host shield, and an Arduino pro mini with this library.
As I ran the example codes (I ran USB_Desc, USBHID_Desc, USBHIDBootKbd, BTHID), in the Serial monitor, it print "Start", but not further information, no matter I replugged without reboot, or plugged the usb device on the shield and reboot. Devices I have tried includes: Logitech keyboard K120, a joystick, bluetooth mouse.

The Arduino pro mini I use is not an unofficial one, but I the pin is used by the shield are the same(http://www.auctionzealot.com/members/rkellyp/promini.png). Also, I am sure the function Usb.Init() succeeded because the Serial monitor didn't print "OSC did not start".

Can you give some ideas? Thanks.

Lauszus, 2016-05-21:
@LSChyi have you enabled serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging?

LSChyi, 2016-05-21:
Thanks for your advises @Lauszus, I enabled it but Serial monitor still print "Start" without any other messages. =

Lauszus, 2016-05-21:
Please try to connect an external power supply: https://github.com/felis/USB_Host_Shield_2.0#faq.

LSChyi, 2016-05-21:
I used an y cable as and draw powers from my computer, but still the same

Lauszus, 2016-05-21:
Please try the following example: https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/board_qc.

LSChyi, 2016-05-21:
Thanks a lot!

Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 02
SPI long test. Transfers 1MB of data. Each dot is 64K................ SPI long test passed
GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
Test failed. Value written: 00 Value read: FF 
Press any key to continue...

So it fails at testing GPIO?

Lauszus, 2016-05-21:
Just send any character to skip the test.

LSChyi, 2016-05-21:
Now it sucks at "Waiting for device..."

Lauszus, 2016-05-21:
Try to plug in your device.

LSChyi, 2016-05-21:
Yes I did, but nothing happened(even I use Y cable as external power).

Lauszus, 2016-05-21:
Hmm okay. What shield are you using?

LSChyi, 2016-05-21:
I use Arduino pro mini usb host shield.

Lauszus, 2016-05-21:
Where did you buy the shield?

LSChyi, 2016-05-21:
I bought it from Taobao, and it says it is made in China. So it is very possible the chip has something goes wrong. Thanks for your help!

Lauszus, 2016-05-21:
I would contact the dealer then, as there is clearly something wrong. You're welcome ;)

closed 2016-05-21 by Lauszus