felis/USB_Host_Shield_2.0 #240: ps4 over usb shield to arduino question

arpeggio1234, 2016-05-26:
Hello, i am using the usb shield 2 library to interface the Ps4 cotroller to the arduino board,
all works fine, using the basic sketch i see the readings from the sensor on the arduino serial screen. 
My question, i see the same values for lets say the lefthatjoy, and when i move the stick, 
the valurs stay the same for 7 to 9 readings and then they change to the value where the stick actualy is, how can this happen, what is happening, and the major question, is there a workaround to this so the actual values will be read, or is this a Ps4 issue?
Thanks for thinkening about..........
Dian from the Netherlands

Lauszus, 2016-05-26:
So you are basically experiencing a delay? What code are you using? Be sure to not have any delays in your code, as you will have to call Usb.Task() frequently.

arpeggio1234, 2016-05-30:
Hello, thanks for the reply, i understand the delay problem and that you have to call USB.task
frequently, i use the basic    Example sketch for the PS4 USB library - developed by Kristian Lauszus
and i see data related to a position where the lefthatjoy was in when the serial print apears, and not the actual values from that moment, so that keeps puzzeling me,
When i use this sketch for controling a speakjet chip via an arduino board and a Ps4 controller via the USB-host shield and offcourse the library, 
i use `/*
 Example sketch for the PS4 USB library - developed by Kristian Lauszus
 For more information visit my blog: http://blog.tkjelectronics.dk/ or
 send me an e-mail:  kristianl@tkjelectronics.com
 */
#include <SoftwareSerial.h>
#define txPin 2
#define rxPin 6
#define busyPin 4
SoftwareSerial speakJet =  SoftwareSerial(rxPin, txPin);
#include <SimpleMessageSystem.h> 
#include <PS4USB.h>
///#include <PS4Parser.h>
// Satisfy IDE, which only needs to see the include statment in the ino.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif

USB Usb;
PS4USB PS4(&Usb);

///boolean printAngle, printAngle1, printTouch,  diansUP, diansDOWN ;
int joyX, joyY;
int L2Value, R2Value;
int accX, accY, accZ;

int sjetArray[] = {44, 52, 58, 65, 78, 88, 104, 116, 130, 156, 175};
int note = 128;
////long accx,accy,accz,joyx,joyy,zbut,cbut;

void setup() {
     pinMode(txPin, OUTPUT);
 pinMode(busyPin, INPUT);
 Serial.begin(115200);
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
  if (Usb.Init() == -1) {
    Serial.print(F("\r\nOSC did not start"));
    while (1); // Halt
  }
     speakJet.begin(9600);
      speakJet.write (28);
      speakJet.write(5); 
 Serial.print(F("\r\nPS4 USB Library Started"));
}
 
void SJBusy(){
  delay(13); // wait 12ms minimum before checking SpeakJet busy pin
  while(digitalRead(busyPin))
  {
    delay(1); // wait here while SpeakJet is busy (pin 4 is true)
  }
//  delay(2); // a bit more delay is needed for some reason
}
void loop() {
  ps4();  
  ps4();  
    ps4();  
  ps4();  
   ps4();  
   ps4();  
  ps4();  
  ps4();  
   ps4();  
  ps4();  
  ps4();  
   ps4();  
  ps4();  
 int accyyy = constrain(accY, 60, 200);
        int range3 = map(accyyy, 60, 200, 10, 0);
              int range1 = map(accX, 0, 360, 0, 127 );
     speakJet.write(21); 
    speakJet.write(range1);
          speakJet.write(22); 
    speakJet.write(sjetArray[range3]);
    speakJet.write(23); 
    speakJet.write(15 - (joyY / 16));
     Serial.println(15 - (joyY / 16));
  // speakJet.print(129); // repeat
   speakJet.write(note++ );
 ///  speakJet.write(151);  // each sound three times
 SJBusy();
if (note > 234)
note = 128;
}
  void ps4()  {
     Usb.Task();
   delayMicroseconds(700);
   if (PS4.connected()) {
       accX  = PS4.getAngle(Roll); // ranges from approx 70 - 182
        accY  = PS4.getAngle(Pitch); // ranges from approx 65 - 173
         accZ  = PS4.getSensor(aZ);
        joyX = PS4.getAnalogHat(LeftHatX);
        joyY = PS4.getAnalogHat(LeftHatY);
       int Zbut = PS4.getButtonClick(L1);
       int Cbut = PS4.getButtonClick(R1); 
   }
  }`   this SKEtch
And than it works great, strange couse i use delays but when i move the controller to a position
the speakjet chip will get the speed and blend and pitch that actuactely match the settings of the 
Ps4 knobs and position, And when i move away the ps4() calls or the delay 700 microseconds
Then i will get a gap in the data flow, so the data that has been is used for now, and thats too late.
This same behavior, data being too late, i saw too with a nunchuck and the nunchuck library,
I solved that by calling get the data from the nunchuck twice, and then it worked fine.
I realize i do not call USB.task frequently, i call it when i need it, thats a different thing, but 
WHEN i call it i get old data. And calling it over and over agian gives me fresh data,
But i will make a Sketch calling USB.task frequently and when the speakjet chip is ready then use the data (already gathered from the Ps4 by the library)for the speakjet,
I will let you know my progress, and thanks again for the great library, midi over the USB-shield 
seems great too.
Dian

Lauszus, 2016-05-31:
You have to call Usb.Task() approximately every 1 ms or the data sent from the controller will simply be put into a buffer which will eventually get full. Thus you need to remove all your delays and instead use something like millis() to keep track of the time: https://learn.adafruit.com/multi-tasking-the-arduino-part-1.

closed 2016-05-31 by Lauszus

arpeggio1234, 2016-05-31:
I got things working the way i wanted, the main goal was no old data, 
And with the sketch i showed you it worked, and when calling USB.task frequently
It worked great, thanks again.
Dian from the Netherlands keeos on thinkening