felis/USB_Host_Shield_2.0 #241: BT Dongle has significantly less range with Host Shield

bbf3, 2016-06-04:
Hello, 

I've noticed that my Bluetooth dongle (TrendNet TBW-106UB), a class 1 device, plugged into a usb port on my laptop, is able to transmit information successfully from an Arduino Serial Monitor to another class 1 device (BlueSmirf Gold) at line-of-sight ranges over 100 meters. 

However, the same dongle plugged into the USB Host Shield, transmitting the same information to the same remote device can only seem to keep a connection at line-of-sight ranges of 8-10 meters. I'vd switched the hardware back and forth between the laptop and Arduino several times, always with the same results.

Any ideas as to why? Any ideas on what could be changed to increase range?

Thanks.

Lauszus, 2016-06-04:
@bbf3 try to connect an external 5V power supply to the Arduino, as the Bluetooth dongle might draw more current than the USB cable can supply.

bbf3, 2016-06-05:
@Lauszus I tried a 12V 1amp power source at the Arduino power input and it made no difference. I also tried directly hardwiring a 5V, 6amp power supply to the USB host shield itself by unsoldering SJ12 (5V from arduino) and soldering a jumper to I think JP24 (5V input). This also did not increase the range. 

Max power consumption of the TBW-106UB is 136mA at TX max, 95mA typical.

Could this be a bluetooth class issue (both devices are class 1), or somehow software related?

Thank you.

Lauszus, 2016-06-05:
Hmm it might seem like you have to implement the LMP protocol: https://developer.bluetooth.org/TechnologyOverview/Pages/LMP.aspx, as it can set the power level.

Here are some other useful links: http://ecee.colorado.edu/~ecen4242/marko/Bluetooth/Bluetooth/SPECIFICATION/LMP.htm, http://www.amd.e-technik.uni-rostock.de/ma/gol/lectures/wirlec/bluetooth_info/lmp.html#Power Control and https://developer.bluetooth.org/gatt/services/Pages/ServiceViewer.aspx?u=org.bluetooth.service.tx_power.xml.

bbf3, 2016-06-06:
So does the USB Host Shield not implement LMP? 

Can one implement bluetooth without using the LMP protocol?

If so, what is the default transmit power and class without LMP?