felis/USB_Host_Shield_2.0 #243: Infinite While Loop: Locked Up Code

stumpja, 2016-06-07:
First off, thanks for this library. It is really amazing!

Now onto the issue. I have a program that uses an Arduino (DUE in my case), to communicate through the USB HOST SHIELD to FTDI devices. I have basically modified the FTDIloopback sketch to my needs. It works well. However, I've noticed that if I leave my project unattended for a while, it locks up and won't work. I don't have any debugger so I drilled back with Serial.print statements to see where I am getting stuck. Turns I get stuck in the OutTransfer function in Usb.cpp. Specifically, I get stuck here:

https://github.com/felis/USB_Host_Shield_2.0/blob/master/Usb.cpp#L327

The odd thing is it works fine for about 1 hr 15 minutes or so. If you send a value every hour, it doesn't seem to fail. But if it goes longer than 1 hr 15 minutes since the last sent value, it consistently hangs.

Does anybody have any ideas? 

I think the next thing I'll try is to simplify my code so that the core can run on an uno and see if it still happens. Could this be an issue with the DUE. I welcome any other troubleshooting.

Thanks,

Josh

xxxajk, 2016-06-08:
I will try to test this problem this weekend, as I have a DUE. 
Post a ping here to remind me. I'm usually very busy and forget.

You may want to wait on your project though...

UHS3 will have FTDI support very soon, as soon as @Lauszus is done with his finals... IIRC about 2-3 weeks, he will be available to get going on this long overdue task, but hey, education first. I have left this driver for him so he can learn how to port all the other stuff that he is responsible for, after I guide him on a better structure too, because that stuff is all a big mess in UHS1 and UHS2.

If you have any regular CDC-ACM or XR21B1411 or PROLIFIC based boards, these are already supported in UHS3. It also comes with a way better example, which shows you how to support all three at the same time in a simpler way. If you don't mind, and have one of these other devices, I would not mind some testing, to see what you think and offer improvements.

stumpja, 2016-06-09:
@xxxajk 

Thanks for the help. I will be sure to ping you this weekend. I realized in my original description, however, I left out a telling piece of information. It only gets locked up after you try to send a value after waiter over 1hr 15mins or more. I have stripped down my code, sorta, and it still fails. This new code should work on an uno, so I will give that a shot today/tomorrow hopefully. I will update my results.

Anyways, it locks up, but you have to try to send something first. My "stripped-down" code is here
[troubleshooting_code.txt](https://github.com/felis/USB_Host_Shield_2.0/files/306998/troubleshooting_code.txt)

I'm excited to hear that UHS3 is getting ready for prime time. I would love to be involved. As you will undoubtedly see if/when you open my code, I'm not sure you'll want my opinion on anything. I'm certainly not advanced. Picking more skills up everyday. I however, would be happy to test, but I don't believe I have any of the boards you mentioned. I have a teensy 3.2, is that CDC-ACM? I apologize for my lack of knowledge.

Thanks again. I will update my results for the weekend.

xxxajk, 2016-06-09:
Yes, all PJRC teensy boards are CDC-ACM :-)

stumpja, 2016-06-10:
@xxxajk Great! I will try to get the Teensy set up with USH3 this month. I'll let you know my thoughts. If you have any particular testing in mind, let me know. Most of my projects focus on using Hubs to access other devices. I haven't done anything with reading USB stored data, but I would like too. Maybe I will start there.

I have an update on my issue, it DOES is fact still get locked with an UNO. This isn't DUE dependent. It gets stuck in the while loop mentioned above.

If you have any other ideas, I would be happy to try. I would love to shift to USH3 when it is ready for me, but as with many projects, I need to get this going now. PCB's will be shipping soon, hopefully, and I want to verify it will works as expected.

Thanks for all your help,

Josh

PS. @Lauszus congrats in advance to finishing school!

xxxajk, 2016-06-10:
If the project is going to be for Due, then no circuit changes will be required at all.
Modifications are only needed on non-ARM/MIPS (AVR) because it has severe limitations.
That said, the way the UHS3 API is, you will note that it is much easier to use.
For example, you don't have to run the Task method, which runs in an ISR.
Hooks are provided if you want to do things during certain events.
During enumeration, when the setup hook is ran, you have full access to the enumeration data if you need to sort by serial number, etc.
Overall, it is a lot more friendly.

stumpja, 2016-06-10:
@xxxajk That's also great! So, my hardware should work for either. The issue however, is that it looks like similar code for the issue that I have presently. In USH3, the same line can be found here

https://github.com/felis/UHS30/blob/fab9313ff3ee9415da1cd5a76bf73398572fc985/libraries/UHS_host/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h#L563

I'm worried that I'll switch to UHS3, but still have the same issue. I like the idea of not needing to run the task method. As with anything new, it will take some getting used too, but it seems like it will be much better (and cleaner, something I could really use).

Anyways, I guess my point is, I will try to switch to USH3 as soon as @Lauszus is able to port over the FTDI stuff. Until then, however, I think I need to focus and figuring out the issue. 

So question for you @xxxajk , now that I have proved it isn't DUE related, is there anything you can try. Do you have an FTDI device handy to see if you could verify my issue. I would be happy to send one your way.

As always, thanks for your help,

Josh

xxxajk, 2016-06-10:
Yeah, I have a Due and FTDI devices.
As far as that line ending up in a loop lock, the newer design should avert that issue, since it will always talk to the device.
My guess is that the FTDI chip somehow may be going into a sleep mode, or has a bug that you are hitting. I'll see what I can find out.

stumpja, 2016-06-10:
@xxxajk That makes sense. I didn't think about the FTDI sleeping. I can look into this as well. UHS3 is looking better and better all the time.

If you get a chance this weekend to look into it that would be great. Remember, it doesn't seem to be DUE dependent either. However, make sure to enjoy the weekend as well, because I certainly will. This could always wait till next week.

Thank you,

Josh

Lauszus, 2017-10-12:
Closing this issue, as further development of FTDI is moved to UHS30.

closed 2017-10-12 by Lauszus