felis/USB_Host_Shield_2.0 #244: BT Dongle Addition and Issue with that dongle

bbf3, 2016-06-08:
The [SENA Technologies UD100](http://www.senanetworks.com/products/industrial_bluetooth/ud100.php) bluetooth dongle works with the USB Host Shield 2.0. The dongle appears to have a hub inside and only works when the hub code is uncommented:
`USBHub Hub1(&Usb);`

I have used this with a PS3 controller and a BlueSmirf Gold device (SPP). One great thing about this dongle is the antenna with RP-SMA connector. After attaching a (Sparkfun WRL-00145) 2.4GHz antenna, my range from Arduino/UHS to both the PS3 and BlueSmirf Gold was greater than 120meters line-of-sight (I ran out of room to keep walking).

Issue: This dongle prevents the Arduino/UHS from properly resetting after a power cycle. I switched back and forth several times between this dongle and another working TrendNet dongle, and the UD100 will disable the Arduino after a power cycling until the reset button is pushed. I am supplying 5V directly from a large power supply to JP24 (with SJ12 closed, and arduino drawing power from same source), but it doesn't seem to make a difference. I'm not exactly sure how this is possible, but if I come up with anything definitive I'll add it here. FYI, this reset issue caused problems with programming a PS3 controller with the correct MAC address until I turned on the debug settings and could see when to plug/unplug/reset the controller and Arduino.

xxxajk, 2016-06-08:
This addition to your UHS might help you:
[https://www.circuitsathome.com/mcu/vbus-power-control-on-usb-host-shield](url)
I found it to be very useful, since it will power-cycle the device plugged into the UHS, which forces the device to be reset to a known state. Honestly though, there is likely a nasty bug preventing a proper reset, due to the UHS2 code.

bbf3, 2016-06-08:
Thanks for the tip. This is actually just the same as removing the SJ19 (5V to VBUS), and then manually connecting it back together to simulate a dongle power cycling, right? I did this with wires soldered to JP27 (VBUS) and JP24 (5V). 

I reset the entire system to working condition. Then turned off power to Arduino and UHS, and then re-powered. I then cut the VBUS connection, powering off only the dongle, waited five seconds, and re-connected VBUS. 

Result: Did not work, and dongle still disables Arduino after power cycle.

xxxajk, 2016-06-08:
After you press reset, then did you simulate the VBUS power off/on?

bbf3, 2016-06-08:
Yeah. From known working condition (i.e. PS3 controller successfully connects), then, power cycle the Arduino/UHS system, wait a couple seconds while its on, power off VBUS only (by disconnecting from 5V), wait five seconds, power VBUS back on again. 

The SENA UD100 dongle has a blue led on it which goes out during the VBUS cycling. No luck. 

I also tried this before mods anyway, by just pulling the dongle out of the USB port and reconnecting it while the system stayed on. Interesting note about this: From a known successful start (using the Serial terminal to see debugging and reset to a good startup), pulling the dongle and re-inserting will let the dongle start back up again. From this point, pulling the dongle from the USB port a second time, and re-inserting, doing nothing else, will not restart the dongle, and make the system require a reset.

xxxajk, 2016-06-08:
when you do this, ground that vbus line to drain. I bet you are seeing parasitic effects, and it is still powered

xxxajk, 2016-06-08:
oh and the other point here is that the uhs2 enumeration is crummy, and could cause unknown side-effects that may not "reboot" the dongle properly.
I can't wait 'till @Lauszus is done with his studies and finals at the uni so he can move all of this stuff to UHS3, where this kind of thing won't happen.

bbf3, 2016-06-09:
Retried the test just now, and grounding the VBUS line after removing five volts does nothing different. Wouldn't everything concerning VBUS also be accomplished by physically removing and replugging the dongle?

Tested as follows:
1. Power on Arduino/UHS, wait a few seconds, press Arduino's reset button (to find known, good condition), and start and connect PS3 controller (successfully). 
2. Power off controller (with PS button) and Arduino/UHS. Wait a few seconds.
3. Power on Arduino/UHS.
4. Wait a couple seconds, remove 5V from VBUS, connect VBUS to ground, wait a couple seconds, and reconnect VBUS to 5V. 
5. Try to power on and connect PS3 controller -- will not connect. 

Also tried removing dongle when power was off, and plugging dongle back in after Arduino/UHS were already powered on. I don't see how this can be dongle related, even though it is. It seems like there is an issue with the UHS board itself not properly starting, or the interface between it and the Arduino. 

A temporary fix for this issue is to use a timer circuit and relay to physically reset the Arduino/UHS a few seconds after startup. But this is quite a workaround.

bbf3, 2016-06-09:
Also tried bringing MRST (MAX2431E reset pin 12) low instead of VBUS at step 4 above to reset the MAX chip. No difference in behavior.

bbf3, 2016-06-09:
An interesting situation: 

In my Arduino code there is an instruction to software reset the Arduino if the peripherals (PS3 and BlueSmirf) go out of range. Going out of range here is distinct from being turned off. Both the PS3 (with "PS3.disconnect();") and the BlueSmirf (with it's physical GPIO6 input pin) can perform a "shutdown" which will not cause the dongle to hang. After a shutdown, the peripherals can be reconnected successfully. Walking out of range, however, will cause an improper shutdown which, in my experience, the dongle can't recover from (both the SENA UD100 or a TrendNet TBW-106UB, the two I've tried so far). So I've added code to the Arduino to reset it in case a controller is no longer connected. 

My code looks something like this (tested and works every time):

```
if(PS3.PS3Connected == 0 && PS3usb.PS3Connected == 0){ //skip loop if controller found
    if(connectOnce == 1){ //if controller was once connected
      Serial.print(" Resetting Arduino sketch! ");
      delay(1000);
      asm volatile ("  jmp 0"); //reset sketch
    }
}else if(PS3usb.PS3Connected == 0){ //BTPS3 commands start here
    connectOnce = 1; //controller has been connected
    if (PS3.getButtonClick(PS)) {       
      //Serial.print(F("\r\n PS Disconnect "));
      PS3.disconnect();
      connectOnce=0;
    }   
    // A lot more instructions here

} else if(PS3.PS3Connected == 0){ //USBPS3 commands start here
    connectOnce = 1;
    if (PS3usb.getButtonClick(PS)) {
      //Serial.print(F("\r\n PS Disconnect "));
      PS3usb.Release();
      connectOnce = 0;
    }
    // Even more instructions here

}
```

This will reset the Arduino If a controller is no longer connected to it, except not when the controllers are intentionally turned off. This was all coded before getting the SENA UD100 dongle.

Now the interesting part: This software resetting also allows the dongle to recover and work again after a system power cycle (where previously power cycling rendered the dongle inoperable without pressing the physical reset button or opening the serial monitor). 

So maybe this narrows the issue to the Arduino/UHS interface, or something strange happening with the Arduino sketch. What this also says is that this situation could probably be patched in the Arduino's sketch (maybe by setting the EEPROM to record power cycling? I haven't thought it through completely).

Thanks for the continued support. This is a great product.

Lauszus, 2017-10-12:
This sounds like an issue a VBUS power switch would solve: https://www.circuitsathome.com/mcu/vbus-power-control-on-usb-host-shield, so closing this issue.

closed 2017-10-12 by Lauszus