felis/USB_Host_Shield_2.0 #245: ADK failing when reconnecting to a phone

Freezerohmatic, 2016-06-15:
Hi All,

I'm having issues with the ADK connecting to a Android phone after been disconnected by the USB cord once. The first time the phone is connected everything works fine, then if it is disconnected then reconnected, the phone does not see the arduino and it is unable to connect.

If the reset button is pressed quickly then "data is received" with the value of 0x0D and 0x04 (04 is very rare) even when no data is being sent by the phone. A long press of the reset button or a power cycle is needed to get the communication working again. This happens with all the example code.

In debug mode I sometimes get this 

```
ADK Init
Acc.mode device detected
Got 1st bad code for config: 0D

getConf 0D

ADK Init
getDevDescr 0D

ADK Init
getDevDescr 0D

ADK Init
getDevDescr 0D
```

or 

```
ADK Init
Acc.mode device detected
Configuration successful
ADK Init
Address in useData rcv. :0D
Data rcv. :0D
Data rcv. :0D
Data rcv. :0D
...
```
I also receive the bellow but only if the cable has been disconnected and reconnected

```
ADK Init
ADK protocol rev. 0002

Accessory mode switch attempt
ADK Init
ADK protocol rev. 0002

Accessory mode switch attempt
```

Any ideas?

Thanks in advance.

Lauszus, 2016-07-17:
@Freezerohmatic please test if the following comment fixes your issue: https://github.com/felis/USB_Host_Shield_2.0/issues/236#issuecomment-216597302.

Freezerohmatic, 2016-07-21:
@Lauszus, Thanks for for showing me that comment. While it does help (I can now reconnect sometimes) I still get the same errors, it now happens about 50% of them time.

Lauszus, 2017-10-12:
Did you ever get this working 100 % of the time?

I believe a VBUS power switch should solve the issue: https://www.circuitsathome.com/mcu/vbus-power-control-on-usb-host-shield/.

I'm closing this issue. Feel free to re-open.

closed 2017-10-12 by Lauszus

Freezerohmatic, 2017-10-13:
No I didn't get it working. The vbus wasn't the issue.

https://www.circuitsathome.com/mcu/vbus-power-control-on-usb-host-shield/.