felis/USB_Host_Shield_2.0 #246: UHS2 and Xbee Shield

d33pend, 2016-06-18:
Hi there,
I've tested the PS3BT sketch and all worked great with a ps3 controller over bluetooth, but I've encoured in a problem when i tried to add an Xbee Shield, the controller doesn't seem to want to connect and just continue to blink, all back working again when i remove the Xbee Shield, you think there is a conflict?
the Xbee shield that i'm using is like that:
http://res.cloudinary.com/kartrocket/image/fetch/w_550,q_80,h_582,c_pad,f_auto,fl_progressive/http%3A%2F%2Fkartrocket-mtp.s3.amazonaws.com%2Fall-stores%2Fimage_rohan01%2Fdata%2FPIN1012.jpg

The Xbee Shield work also very good when i remove the UHS infact i receive all the data from the other module Xbee, and think it use just the TX-RX just to send them wireless, so what can be the problem?

d33pend, 2016-06-20:
thanks @Lauszus just made those 2 changes, i cut the 2 pad (SS and INT) and routed these pins to others, but the problem still persist

anyway i've this UHS shield: http://fizzible.com/wp-content/uploads/2015/01/IMG_20150117_141236.jpg
maybe there are something that i can try?

Lauszus, 2016-06-21:
@d33pend I can't see how connecting the reset pin to the UHS shield should prevent the Xbee shield from working? The reset pin is used to reset the UHS shield when the Arduino is also reset.

d33pend, 2016-06-21:
Problem solved. No, it wasn't a problem with the UHS shield but by the Xbee shield side, infact after investigating on why the problem occurs after i connect the pin RESET, i found the solution [here](http://randomstuff-ole.blogspot.it/2009/09/xbee-woes.html):

that suggest to disable this setting "CTS Flow Control" on the Xbee: 

> D7=0 (CTS on XBee is hardwired to reset pin of Arduino. When XBee sleeps, Arduino resets.)

> simply set IOD7=0 (or use ATD7=0, same thing). The XBee will no longer reset your Arduino simply because it wants to sleep.

So anyway @Lauszus thank you very much for your ready support!

Lauszus, 2016-06-21:
@d33pend you're welcome and thanks for sharing the solution :)

closed 2016-06-21 by Lauszus