felis/USB_Host_Shield_2.0 #247: Shield not working with the XBOXRECV example

FlawedBliss, 2016-06-28:
Hey,

I just acquired one of these shields and have trouble getting things to work.

When connecting my Xbox360 Receiver, it will only blink when entering discovery mode and else do nothing. If I get a controller to match is the blinking stops aswell. Serial output reads "OSC did not start".

I have tried it on two different Arduino Uno devices with nothing else connected and ran the board_qc example. One of them is an Uno R3. Outputs for each device are as follows:

First device:

> Circuits At Home 2011
> USB Host Shield Quality Control Routine
> Reading REVISION register... Die revision 03
> SPI long test. Transfers 1MB of data. Each dot is 64K
> Test failed.  Value written: A1 read: F6
> Unrecoverable error - test halted!!
> 0x55 pattern is transmitted via SPI
> Press RESET to restart test

Second device (R3):

> Circuits At Home 2011
> USB Host Shield Quality Control Routine
> Reading REVISION register... Die revision invalid. Value returned: 00
> Unrecoverable error - test halted!!
> 0x55 pattern is transmitted via SPI
> Press RESET to restart test

Connecting different devices (Tried USB-Sticks and my graphic tablet) does not trigger a reaction either (no status lights on the device or shield, nothing).

Anything I'm missing? 

Greets

xxxajk, 2016-06-28:
check for ripple on the 3.3v supply... i've seen this happen before with a
bad regulator on the mcu board. If that is fine, you may have to try
external power, instead of just usb. Failing that, your board is defective.

FlawedBliss, 2016-06-28:
Hey,

boards seem fine, already tried both usb, external and both at the same time.
Any way I can check what's breaking? I've enabled the debugging flag in settings.h but didn't get any additional info on my serial monitor. What's weirding me out is that the test passes the first step one one board and fails on the other.

If not I guess I'm replacing it. Thanks for the fast response.

felis, 2016-06-28:
the test shows intermittent spi connection. It it either bad contact or bad
chip. If you bought it from me, send it back - I'll check it and send you a
replacement.

FlawedBliss, 2016-06-28:
Hey,

thanks for the offer but I got it from a local shop. I'll send it back there and probably get a new one directly from you guys then.

Thanks for the quick help tho!

closed 2016-08-09 by Lauszus