felis/USB_Host_Shield_2.0 #248: How to change pin assigments.

nanthony21, 2016-07-18:
I am particularly interested in how I can change which pin is used for the interrupt on a Teensy 3.1

xxxajk, 2016-07-18:
There is no way to do this dynamically, however UHS30 does support doing so. Depending on the application a driver may or may not be available for it though. What are you attaching?

nanthony21, 2016-07-18:
Ok, Thanks. I'm currently using a USB XBox 360 for windows controller. But in the future I am also hoping to use a Thrustmastet flightx Hotas joystick

xxxajk, 2016-07-18:
HID is not available yet on UHS3.0, but you can modify it in the header file for UHS2.0 that defines it here:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/UsbCore.h#L43
replace P10 and P9 to the pins that you want to use.

You also need to rewire the board to match this change.

nanthony21, 2016-07-18:
Thanks, that's exactly the kind of solution I'm looking for.

closed 2016-07-18 by Lauszus