felis/USB_Host_Shield_2.0 #249: What is interrupt for?

nanthony21, 2016-07-18:
I had assumed that the interrupt pin was neccessary but I just realized that my setup works fine with it unplugged. I'm just wondering when it is necessary and what it's used for. Can I safely ignore it?

xxxajk, 2016-07-18:
Sounds to me like a code regression. This needs to be fixed because it is
less efficient if it is not used. I will look into it tonight

Lauszus, 2016-07-19:
The interrupt pin is checked in the Task function: https://github.com/felis/USB_Host_Shield_2.0/blob/master/usbhost.h#L500-L518, so maybe something else is going on?

nanthony21, 2016-07-19:
Weird. I know that the interrupt was necessary when I first started. But now it isn't. I haven't changed any code in the library, all I've changed is my .ino file.

I have one idea for what might be causing it. I only am interested in collecting information from the xbox controller once every 10ms. Initially i tried this:

`void loop(){
  while (elapsedmillis<=10){}
  elapsedmillis=0;
  usb.task();
  processXboxInput();
}`

This didn't work, presumably because usb.task needs to be called more frequently. So i changed to this:

`void loop(){
  while (elapsedmillis<=10){
    usb.task();
  }
  elapsedmillis=0;
  processXboxInput();
}`

Could it be that repeatedly calling usb.task without doing anything else is messing up the need for an interrupt?

nanthony21, 2016-07-19:
Also, since I only need 100hz, does it matter if I am skipping the interrupt?

felis, 2016-07-19:
try to connect/disconnect the 360 while running the sketch and see if there
is any difference with and without an interrupt.

nanthony21, 2016-07-19:
Thanks, It looks like the interrupt is necessary in order for a hotplug to be detected. this actually reminds me of a separate issue that I was having occasionally: Sometimes if I start the teensy w/ usb host and then plug the xbox controller in later, it will tie ground to VCC, e.g. everything will shut off and the teensy's regulator will start to get hot.

nanthony21, 2016-07-20:
Actually I just checked again and now I'm not seeing any difference between having the interrupt pin plugged in or not. Sometimes it detects the controller being plugged in and sometimes it doesn't

closed 2016-12-05 by Lauszus