felis/USB_Host_Shield_2.0 #252: Support for Bluetooth Connectivity with the new XBOX One S Controller

vickylance, 2016-08-02:
Please Add support for the new Xbox one S controller to connect to Arduino using Bluetooth Shield.

Lauszus, 2016-08-03:
@vickylance if you donate a controller, I would be happy to do so.

vickylance, 2016-08-03:
If I had the controller I would add the support by myself :P

Lauszus, 2016-11-29:
No, as I do not own the controller.

TheRob87, 2017-06-21:
Hi, own both is it implemented know? If not may someone can help me to develop?

Lauszus, 2017-06-21:
@TheRob87 no I haven't worked on it since I don't own the controller.

TheRob87, 2017-06-22:
@Lauszus is it correct that Xbox one s Controller over USB are supported? So Only problem is the connection between Bluetooth and the Seriell in?

Lauszus, 2017-06-22:
I'm actually not sure. Can anybody with a Xbox One S controller test #298?

Lauszus, 2017-06-23:
@TheRob87 let me clarify that: I'm not sure if it works via USB or not, Bluetooth is currently not supported.

nathanRamaNoodles, 2018-12-14:
Any update on this?

Lauszus, 2018-12-14:
@nathanRamaNoodles no. If you want me to work on it, then please donate a controller.

Lauszus, 2018-12-16:
@nathanRamaNoodles yes, that would be great! Could you please contact me at lauszus@gmail.com so we can work out the details?

Lauszus, 2019-01-06:
@nathanRamaNoodles ping?

TheRob87, 2019-04-30:
Hi, any updates? Did you get the money? Otherwise I also would donate a Controller.

nathanRamaNoodles, 2019-04-30:
@TheRob87 dude, just get a [raspberry pi zero w](https://www.microcenter.com/product/486575/zero-w).  It's cheaper($5) and has way more USB host support than an Arduino.

Lauszus, 2019-05-28:
@TheRob87 no I never got any donations.

alostsandwich, 2019-06-10:
A controller has now been sent to @Lauszus .

Lauszus, 2019-06-10:
@alostsandwich thank you! I'll let you know once I receive it.

alostsandwich, 2019-06-12:
@Lauszus Delivery was missed for some reason by UPS and it's available for pickup at the local collection point.  I sent a few emails to you but haven't heard back.  Is there any issue with picking it up?  It'll have to be picked up before June 21 otherwise they'll send the package back.

Lauszus, 2019-06-13:
@alostsandwich sorry I have been super busy at work. I'll pick up the package today.

Lauszus, 2019-06-30:
I finally took the time to look into the code. It turns out the controller only works with "simple paring" mode. I'm lacking a dongle that supports it, so I can get any further. I hope to buy a new dongle this week and continue the work.

In the meantime anyone that wants to test it can download the code here: https://github.com/felis/USB_Host_Shield_2.0/archive/XBOXONES.zip. Please post the output from the serial monitor in this thread.

Lauszus, 2019-06-30:
Btw please run the [BTHID.ino](https://github.com/felis/USB_Host_Shield_2.0/blob/XBOXONES/examples/Bluetooth/BTHID/BTHID.ino) example.

Mixinu, 2019-07-01:
Wow thanks! 

Unfortunately the Xbox USB wireless adapter doesn't seem to have built-in Bluetooth so I can only test with an old Bluetooth 2.0 dongle that I have around.

I don't expect it to work, but this is what I got just in case it helps:

___________________________

HID Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:80:5A:46:B5:48
HCI version: 03
Page number: 00
Maximum page number: 00
Extended LMP features: FF FF 8F FE 9B F9 00 80
Local does NOT support secure simple paring (controller support)
Please enable discovery of your device
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Remote Name: Xbox Wireless Controller
Connecting to HID device
Connection established
Connected to HID device
Page number: 00
Maximum page number: 03
Extended LMP features: BF 3E 8D FE DB FD 7B 87
Remote supports secure simple paring (controller support)
Received Key Request
Bluetooth pin is set too: 0000
Pairing Failed: 06
HCI Disconnected from Device
Wait For Incoming Connection Request

Lauszus, 2019-07-02:
@Mixinu yes, that is the expected output with a Bluetooth 2.0 dongle. Any change you can test at least a 2.1 dongle?

alostsandwich, 2019-07-02:
Here's the output using an Xbox One S white controller and an IOGEAR GBU521.

> HID Bluetooth Library Started
> BTD ConfigureDevice
> BTD Init
> Addr: 01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	81
> Attributes:	03
> MaxPktSize:	0010
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	82
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	02
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	01
> Bluetooth Dongle Initialized
> HCI Reset complete
> Write class of device
> Local Bluetooth Address: 5C:F3:70 -snipping the rest of the MAC-
> HCI version: 06
> Page number: 00
> Maximum page number: 01
> Extended LMP features: BF FE CF FE DB FF 7B 87
> Local supports secure simple paring (controller support)
> Please enable discovery of your device
> Gamepad found
> HID device found
> Now just create the instance like so:
> BTHID bthid(&Btd);
> And then press any button on the device
> Remote Name: Xbox Wireless Controller
> Connecting to HID device
> Connection established
> Connected to HID device
> Page number: 00
> Maximum page number: 03
> Extended LMP features: BF 3E 8D FE DB FD 7B 87
> Remote supports secure simple paring (controller support)
> Received Key Request
> Bluetooth pin is set too: 0000
> Pairing Failed: 06
> HCI Disconnected from Device
> Wait For Incoming Connection Request

Lauszus, 2019-07-02:
@alostsandwich interesting you dongle actually supports simple paring, but the controller still ask for a pin.

I have logged what my laptop does and in that case the controller does not ask for a pin, but instead it sends out a "User Confirmation Request", thus running the simple paring procedure. I'll look into it.

Lauszus, 2019-07-02:
@alostsandwich I just pushed a changed. Could download the code again and see if it made any difference?

alostsandwich, 2019-07-02:
@Lauszus Sure.

> HID Bluetooth Library Started
> BTD ConfigureDevice
> BTD Init
> Addr: 01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	81
> Attributes:	03
> MaxPktSize:	0010
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	82
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	02
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	01
> Bluetooth Dongle Initialized
> HCI Reset complete
> Write class of device
> Local Bluetooth Address: 5C:F3:70 -snip-
> HCI version: 06
> Page number: 00
> Maximum page number: 01
> Extended LMP features: BF FE CF FE DB FF 7B 87
> Dongle supports secure simple pairing (controller support)
> Please enable discovery of your device
> Gamepad found
> HID device found
> Now just create the instance like so:
> BTHID bthid(&Btd);
> And then press any button on the device
> Remote Name: Xbox Wireless Controller
> Connecting to HID device
> Connection established
> Connected to HID device
> Unsupported L2CAP Data - Channel ID: 00 01
> Data: 
> 02 01 04 00 01 00 40 00 
> Page number: 00
> Maximum page number: 03
> Extended LMP features: BF 3E 8D FE DB FD 7B 87
> Remote supports secure simple pairing (controller support)
> Received Key Request
> Pairing Failed: 05
> HCI Disconnected from Device
> HCI Command Failed: 12
> Wait For Incoming Connection Request

Lauszus, 2019-07-02:
@alostsandwich could you try one more time? I just want to confirm that simple paring gets enabled in your dongle. If that is the case, then I probably need to setup a L2CAP connection.

alostsandwich, 2019-07-02:
> 
> 
> @alostsandwich could you try one more time? I just want to confirm that simple paring gets enabled in your dongle. If that is the case, then I probably need to setup a L2CAP connection.

If I'm reading it right then it looks like simple pairing is enabled.

> HID Bluetooth Library Started
> BTD ConfigureDevice
> BTD Init
> Addr: 01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	81
> Attributes:	03
> MaxPktSize:	0010
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	82
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	02
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	01
> Bluetooth Dongle Initialized
> HCI Reset complete
> Write class of device
> Local Bluetooth Address: 5C:F3:70 -snip-
> HCI version: 06
> Page number: 00
> Maximum page number: 01
> Extended LMP features: BF FE CF FE DB FF 7B 87
> Dongle supports secure simple pairing (controller support)
> Simple pairing was enabled
> Please enable discovery of your device
> Gamepad found
> HID device found
> Now just create the instance like so:
> BTHID bthid(&Btd);
> And then press any button on the device
> Remote Name: Xbox Wireless Controller
> Connecting to HID device
> Connection established
> Connected to HID device
> L2CAP Data - Channel ID: 00 01
> Data: 
> 02 01 04 00 01 00 40 00 
> Page number: 00
> Maximum page number: 03
> Extended LMP features: BF 3E 8D FE DB FD 7B 87
> Remote supports secure simple pairing (controller support)
> Received Key Request
> Pairing Failed: 05
> HCI Disconnected from Device
> HCI Command Failed: 12
> Wait For Incoming Connection Request

Lauszus, 2019-07-02:
@alostsandwich I wrote some initial code for the SDP communication. Could you try it and post the output again?

alostsandwich, 2019-07-02:
> 
> 
> @alostsandwich I wrote some initial code for the SDP communication. Could you try it and post the output again?

> HID Bluetooth Library Started
> BTD ConfigureDevice
> BTD Init
> Addr: 01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	81
> Attributes:	03
> MaxPktSize:	0010
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	82
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	02
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	01
> Bluetooth Dongle Initialized
> HCI Reset complete
> Write class of device
> Local Bluetooth Address: 5C:F3:70 -snip-
> HCI version: 06
> Page number: 00
> Maximum page number: 01
> Extended LMP features: BF FE CF FE DB FF 7B 87
> Dongle supports secure simple pairing (controller support)
> Simple pairing was enabled
> Please enable discovery of your device
> Number of responses: 1
> Class of device: 28 04 24
> Number of responses: 1
> Class of device: 28 04 24
> -- Repeats the above 2 lines in a loop until controller is trying to sync --
> Gamepad found
> HID device found
> Now just create the instance like so:
> BTHID bthid(&Btd);
> And then press any button on the device
> Remote Name: Xbox Wireless Controller
> Connecting to HID device
> Connection established
> Connected to HID device
> L2CAP Data - Channel ID: 00 01, data: 02 01 04 00 01 00 40 00 
> L2CAP Connection Request - PSM: 00 01 SCID: 00 40 Identifier: 01
> SDP Incoming Connection Request
> L2CAP Data - Channel ID: 00 01, data: 04 02 08 00 00 00 00 00 01 02 C8 05 
> Page number: 00
> Maximum page number: 03
> Extended LMP features: BF 3E 8D FE DB FD 7B 87
> Remote supports secure simple pairing (controller support)
> L2CAP Data - Channel ID: 00 01, data: 05 02 06 00 00 00 00 00 00 00 
> SDP Successfully Configured
> L2CAP Data - Channel ID: 00 00, data: 06 00 00 00 0D 35 03 19 12 00 00 64 35 03 09 00 01 00 
> SDP_SERVICE_SEARCH_ATTRIBUTE_REQUEST_PDU
> UUID: 1200
> Length: 000D
> Data: 35 03 19 12 00 00 64 35 03 09 00 01 00 
> Received Key Request
> L2CAP Data - Channel ID: 00 00, data: 02 00 01 00 08 35 03 19 12 00 00 01 00 
> SDP_SERVICE_SEARCH_REQUEST
> Pairing Failed: 05
> HCI Disconnected from Device
> Wait For Incoming Connection Request

Lauszus, 2019-07-02:
Does it make any difference if you change:

```C++
hci_read_remote_extended_features(0); // "Requests the normal LMP features as returned by the HCI_Read_Remote_Supported_Features command"
//hci_read_remote_extended_features(1); // Read page 1
```

to:

```C++
//hci_read_remote_extended_features(0); // "Requests the normal LMP features as returned by the HCI_Read_Remote_Supported_Features command"
hci_read_remote_extended_features(1); // Read page 1
```
in BTD.cpp?

alostsandwich, 2019-07-02:
Same result.
> HID Bluetooth Library Started
> BTD ConfigureDevice
> BTD Init
> Addr: 01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	81
> Attributes:	03
> MaxPktSize:	0010
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	82
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	02
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	01
> Bluetooth Dongle Initialized
> HCI Reset complete
> Write class of device
> Local Bluetooth Address: 5C:F3:70
> HCI version: 06
> Page number: 00
> Maximum page number: 01
> Extended LMP features: BF FE CF FE DB FF 7B 87
> Dongle supports secure simple pairing (controller support)
> Simple pairing was enabled
> Please enable discovery of your device
> Number of responses: 1
> Class of device: 28 04 24

> Gamepad found
> HID device found
> Now just create the instance like so:
> BTHID bthid(&Btd);
> And then press any button on the device
> Remote Name: Xbox Wireless Controller
> Connecting to HID device
> Connection established
> Connected to HID device
> L2CAP Data - Channel ID: 00 01, data: 02 01 04 00 01 00 40 00 
> L2CAP Connection Request - PSM: 00 01 SCID: 00 40 Identifier: 01
> SDP Incoming Connection Request
> Page number: 01
> Maximum page number: 03
> Extended LMP features: 01 00 00 00 00 00 00 00
> Remote supports secure simple pairing (host support)
> L2CAP Data - Channel ID: 00 01, data: 04 02 08 00 00 00 00 00 01 02 C8 05 
> L2CAP Data - Channel ID: 00 01, data: 05 02 06 00 00 00 00 00 00 00 
> SDP Successfully Configured
> L2CAP Data - Channel ID: 00 00, data: 06 00 00 00 0D 35 03 19 12 00 00 64 35 03 09 00 01 00 
> SDP_SERVICE_SEARCH_ATTRIBUTE_REQUEST_PDU
> UUID: 1200
> Length: 000D
> Data: 35 03 19 12 00 00 64 35 03 09 00 01 00 
> Received Key Request
> L2CAP Data - Channel ID: 00 00, data: 02 00 01 00 08 35 03 19 12 00 00 01 00 
> SDP_SERVICE_SEARCH_REQUEST
> Pairing Failed: 05
> HCI Disconnected from Device
> Wait For Incoming Connection Request

Mixinu, 2019-07-04:
I just tried with a Bluetooth 4.0 dongle but it failed to upload BTHID.ino because "Sketch uses 31940 bytes (111%) of program storage space. Maximum is 28672 bytes" (I'm using an Arduino Leonardo).

It was ok when I disabled serial debugging on settings.h, but apparently the dongle didn't get to connect to the controller.

Do you think I could disable some of the included libraries, just to save some space to enable debugging and see what's going on?

Lauszus, 2019-08-31:
@alostsandwich I just updated the PR, could you run it again and post the output?

My current symptoms are:

It just connects and opens a SDP channel it then disconnects the SDP channel and I get a "Paring Faild: 05 (Authentication Failure)"

For some reason I never receive the "IO Capability Request" event

Here's the full log:

```
HID Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 34:C9:F0:8B:C9:3B
HCI version: 06
Page number: 00
Maximum page number: 01
Extended LMP features: BF FE CF FE DB FF 7B 87
Dongle supports secure simple pairing (controller support)
Simple pairing was enabled
Please enable discovery of your device
Number of responses: 1
Class of device: 00 05 08
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Remote Name: Xbox Wireless Controller
Connecting to HID device
Connection established
Connected to HID device
L2CAP Connection Request - PSM: 00 01 SCID: 00 40 Identifier: 01
SDP Incoming Connection Request
Page number: 00
Maximum page number: 03
Extended LMP features: BF 3E 8D FE DB FD 7B 87
Remote supports secure simple pairing (controller support)
SDP Configuration Request
SDP Configuration Complete
SDP Successfully Configured
SDP_SERVICE_SEARCH_ATTRIBUTE_REQUEST
UUID: 1200
Length: 000D
Data: 35 03 19 12 00 00 64 35 03 09 00 01 00 
Received Key Request
SDP_SERVICE_SEARCH_REQUEST
Disconnect Request: SDP Channel
Disconnected SDP Channel
Pairing Failed: 05
HCI Disconnected from Device
HCI Command Failed: 12
Num HCI Command Packets: 01
Command Opcode: 06 04
Wait For Incoming Connection Request
```

alostsandwich, 2019-08-31:
@Lauszus here you go.
```
HID Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: -snip-
HCI version: 06
Page number: 00
Maximum page number: 01
Extended LMP features: BF FE CF FE DB FF 7B 87
Dongle supports secure simple pairing (controller support)
Simple pairing was enabled
Please enable discovery of your device
...
...
Number of responses: 1
Class of device: 28 04 24
Number of responses: 1
Class of device: 00 05 08
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Remote Name: Xbox Wireless Controller
Connecting to HID device
Connection established
Connected to HID device
L2CAP Connection Request - PSM: 00 01 SCID: 00 40 Identifier: 01
SDP Incoming Connection Request
Page number: 00
Maximum page number: 03
Extended LMP features: BF 3E 8D FE DB FD 7B 87
Remote supports secure simple pairing (controller support)
SDP Configuration Request
SDP Configuration Complete
SDP Successfully Configured
SDP_SERVICE_SEARCH_ATTRIBUTE_REQUEST
UUID: 1200
Length: 000D
Data: 35 03 19 12 00 00 64 35 03 09 00 01 00 
Received Key Request
SDP_SERVICE_SEARCH_REQUEST
Disconnect Request: SDP Channel
Disconnected SDP Channel
Pairing Failed: 05
HCI Disconnected from Device
HCI Command Failed: 12
Num HCI Command Packets: 01
Command Opcode: 06 04
Wait For Incoming Connection Request
```

alostsandwich, 2019-08-31:
Not sure that it matters but the controller firmware that I'm using is 3.1.1221.0 which is the latest from Microsoft.

Lauszus, 2019-08-31:
@alostsandwich thanks. It's the exact same as I am getting. For some reason it never receives the "IO Capability Request" event...

MX682X, 2019-12-21:
There is no IO capability probably because it's filtered out. Clearing the event filter and setting an event mask did the trick on my own project.
HID Commands logged with Wireshark on linux:
01 05 0c 01 00 (Event filter)
01 01 0c 08 ff ff fb ff 07 f8 bf 3d (Event mask)

gabrielgbs97, 2020-04-21:
I do not have the controller, could someone perform following command "sdptool browse [controller_hw_addr]" in order to retrieve sdp record database?

HisashiKato, 2020-07-19:
I modified the library to make it work.
Is it useful?
[https://github.com/HisashiKato/USB_Host_Shield_Library_2.0_BTXBOX](https://github.com/HisashiKato/USB_Host_Shield_Library_2.0_BTXBOX)

Mixinu, 2020-07-29:
Tested and works flawless. Thanks HisashiKato!

MX682X, 2020-08-23:
@Jc7tee looking into the datasheet of your adapter it seems that only BLE is supported up to version 4.0. Classic Bluetooth is only supported up to v2.1, but you need classic bluetooth v4.0 or higher for the controller as far as I know. 
I may be wrong though. I don't know the library, but I do know that the controller establishes the connection through a (Secure) Simple Pairing method, where no pin is needed. Look into the library, maybe there is some option about it. (You need to enable SSP throuh a command first)

HisashiKato, 2020-08-24:
@Jc7tee This is an example log when I run the sketch example "XBOXONE_BT_Test.ino" of the library I modified for the Xbox One S controller.
```Start
XBOX Bluetooth Library Started

Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Simple pairing was enabled
Set Event Mask
HCI_CHECK_DEVICE_SERVICE
pairWithHIDDevice = 01
Please enable discovery of your device.
Gamepad found
BD_ADDR: 98:7A:14:30:4D:5C
HID device found
Remote Name: Xbox Wireless Controller
Connecting to HID device
EV_CONNECTION_COMPLETE:Connection established
Connected to HID device
EV_LINK_KEY_REQUEST 
pairWithHIDDevice = 01
hci_link_key_request_negative_reply
Start Simple Pairing
Received IO Capability Request: 
User confirmation Request: 
Link Key Notification
EV_AUTHENTICATION_COMPLETE
Pairing successful with HID device
pairWithHIDDevice:01
connectToHIDDevice:01
Send HID Control Connection Request
EV_Encryption Changed
EV_Encryption Changed
EV_Encryption Changed
Send HID Control Config Request
Set protocol mode: 01
Send HID Interrupt Connection Request
Send HID Interrupt Config Request
HID Channels Established   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 0000

HCI_DONE_STATE: >>> HCI_DISCONNECT_STATE:Dn: 1
Up: 1
Dn: 2
Up: 2
Hat Switch: 01
Hat Switch: 00
Hat Switch: 05
Hat Switch: 00
Hat Switch: 01
Hat Switch: 00
   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 0029
   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 015B
   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 024D
   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 025F
   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 02C8
   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 02C2
   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 021C
   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 0119
   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 0033
   X: 89E1  Y: 8A1D   Z1: 81C2  Z2: 8226   LT: 0000   RT: 0000
```
Xbox One S controller uses SSP for pairing, so my library uses "BTDSSP.cpp" which I modified "BTD.cpp" for SSP only.

Jc7tee, 2020-08-24:
@HisashiKato I uncommented the debugging/reporting definitions in BTDSSP.cpp, BTXBOX.cpp and settings.h but still no output on my serial monitor except It only says
```
Start
XBOX Bluetooth Library Started
```

That's why I went back to the original files. Your code didn't trigger my dongle to search for devices (the LED is not blinking).  I'm not sure if this related but I'm using atmega2560 with MegaCore bootloader installed.

xxxajk, 2020-08-24:
The bootloader should not affect it. The Bluetooth plugged into the USB has
to be smart all on its own and not require any firmware uploading. if it
requires firmware uploading then you need to get a different one.

Mixinu, 2020-08-24:
This is my Bluetooth dongle, just in case it helps:
[https://www.amazon.co.uk/dp/B01F75JR42/](url)

Working perfectly. I just had to update the controller firmware.

HisashiKato, 2020-10-27:
The following is the process by which the BTD.cpp function I modified to connect to the Xbox One S Controller via bluetooth establishes an SSP connection.

Bluetooth USB Adapter <-> USB Host Shield

<- HCI_Reset_Command
<- HCI_Write_Class_Of_Device_Command
<- HCI_Write_Simple_Pairing_Mode_Command(enabled)
<- HCI_Set_Event_Mask_Command

Start scanning the device.
:
:
Perform the process of connecting to the device.
:
:
Connected to the device.
-> HCI_Connection_Complete_Event

<- HCI_Authentication_Requested_Command
-> HCI_Link Key_Request_Event
<- HCI_Link Key_Request_Negative_Reply_Command  //Start Simple Pairing.

-> HCI_IO_Capability_Request_Event
<- HCI_IO_Capability_Request_Reply_Command(NoInputNoOutput)
-> HCI_IO_Capability_Response_Event

:
:
(the public keys are exchanged between the two devices.)
:
:

-> HCI_User_Confirmation_Request_Event(Value)
<- HCI_User_Confirmation_Request_Reply_Command(Succeeded) //Skip the step asking for the user confirmation.

-> HCI_Simple_Pairing_Complete_Event

-> HCI_Link_Key_Notification_Event(new Link Key) //Notification of generated new link key.

:
:
(Store the paired BD ADDRESS and Link Key somewhere.)
(The stored link key will be used for future connections with the paired device.)
:
:

-> HCI_Authentication_Complete_Event

<- HCI_Set_Connection_Encryption_Command(on)
-> HCI_Encryption_Change_Event(on)

The pairing process is complete.

HisashiKato, 2020-10-27:
@Lauszus
>For some reason it never receives the "IO Capability Request" event...

If you send a command to set such an HCI_Set_Event_Mask, you can receive HCI_IO_Capability_Request event.

```
void BTD::hci_set_event_mask() {
        hcibuf[0] = 0x01; // HCI OCF = 01
        hcibuf[1] = 0x03 << 2; // HCI OGF = 3
        hcibuf[2] = 0x08;
        hcibuf[3] = 0xFF;
        hcibuf[4] = 0xFF;
        hcibuf[5] = 0xFF;
        hcibuf[6] = 0xFF;
        hcibuf[7] = 0xFF;
        hcibuf[8] = 0x1F;
        hcibuf[9] = 0xFF;
        hcibuf[10] = 0x00;

        HCI_Command(hcibuf, 11);
}
```

HisashiKato, 2020-10-27:
Here is the repository of the library that I modified based on USB_Host_Shield_Library_2.0.

https://github.com/HisashiKato/USB_Host_Shield_Library_2.0_BTXBOX

Questions about my modified library should be in my repository.

Lauszus, 2020-11-16:
@HisashiKato thanks a lot! I've now added support for the controller: https://github.com/felis/USB_Host_Shield_2.0/pull/471. I'll merge the PR into master once the Travis build has finished.

closed 2020-11-16 by Lauszus

Thammaraj, 2022-03-29:
support Xbox Series X controller (Model 1914) and CSR bluetooth 5.0...?

HisashiKato, 2022-03-30:
@Thammaraj
I do not have an Xbox Series X/S Controller MODEL 1914.
I do not know how the Xbox Series X/S Controller MODEL 1914 communicates via USB cable.
The Xbox Series X/S Controller MODEL 1914 has two types of wireless: Xbox proprietary and Bluetooth LE.
The Xbox One S Controller MODEL 1708 has two types of wireless: Xbox proprietary and Bluetooth Classic.
The BTD.cpp in the USB Host Library 2.0 is only for Bluetooth Classic. Bluetooth LE is not supported.
Adding Bluetooth LE support is a very large task.  I cannot do it.
Bluetooth 5.0 is backward compatible with Bluetooth 4.0. However, the extent of compatibility varies from product to product.
To be honest, you won't know until you buy it and try it out.

(Translated from Japanese to English using a translation site)

Thammaraj, 2022-03-31:
> @ธรรมราช ฉันไม่มี Xbox Series X/S Controller MODEL 1914 ฉันไม่รู้ว่า Xbox Series X/S Controller MODEL 1914 สื่อสารผ่านสาย USB ได้อย่างไร Xbox Series X/S Controller MODEL 1914 มีระบบไร้สายสองประเภท: กรรมสิทธิ์ของ Xbox และ Bluetooth LE Xbox One S Controller MODEL 1708 มีระบบไร้สายสองประเภท: กรรมสิทธิ์ของ Xbox และ Bluetooth Classic BTD.cpp ใน USB Host Library 2.0 ใช้สำหรับ Bluetooth Classic เท่านั้น ไม่รองรับบลูทูธ LE การเพิ่มการรองรับ Bluetooth LE เป็นงานที่ใหญ่มาก ฉันไม่สามารถทำมันได้. มาตรฐาน Bluetooth ทุกรุ่นรองรับความเข้ากันได้ที่ลดลง อย่างไรก็ตาม ขอบเขตของความเข้ากันได้แตกต่างกันไปในแต่ละผลิตภัณฑ์ บอกตามตรงว่าคุณจะไม่มีวันรู้จนกว่าคุณจะซื้อมันมาลองใช้ดู
> 
> (แปลจากภาษาญี่ปุ่นเป็นภาษาอังกฤษโดยใช้ไซต์แปล)

Ok, I understand why my new device won't connect. I also have an Xbox Model 1708 joystick used for testing. thanks for the answer

HisashiKato, 2023-03-14:
@GGKTL 
NO.
MODEL 1914 is use Blutooth LE (BLE) .
My library does not support Bluetooth LE (BLE).
It is very, very difficult for me.

GGKTL, 2023-03-20:
@HisashiKato 
Sir, can use Arduino GIGA R1 WiFi , yes or no , Thanks you
New update yours >> USB_Host_Shield_Library_2.0_BTSSP
https://github.com/HisashiKato/USB_Host_Shield_Library_2.0_BTSSP

PR #471 (2020-11-16): Xbox One S controller support
Fixes #252

Also fixed some issues so the Travis build succeeds.