felis/USB_Host_Shield_2.0 #255: Pre-Pairing PS3 controller with troublesome Bluetooth Adapters

bbf3, 2016-08-25:
Hey,

Just wanted to provide a tip to anyone struggling to pair a PS3 controller with a Bluetooth dongle and the UHS2.0. I have found that some dongles are easier than others to pair. For example, I have struggled with pairing a PS3 controller to the SENA UD100, an otherwise great long-distance BT dongle with antenna.

An easier way:
1. Search google for "Sixaxis pair tool" and download the small application (version 0.3.0 worked for me on Windows 10, available [here](http://dancingpixelstudios.com/sixaxiscontroller/SixaxisPairToolSetup-0.3.0.exe)). 
2. Open the program, and then plug in your PS3 controller to your computer via USB. The pair tool will display the current stored BTA for the controller (it may display 00:00:00:00:00:00 if the controller is new). 
3. Find your BT dongle's BT address and write it into the program (with separating colons). Click update.
 
The controller will now be paired to the dongle without using the UHS2.0.

Hope this helps someone. With other dongles (i.e. TrendNET TBW-106UB, etc.) programming through the UHS2.0 has been no problem.  

Blake

Lauszus, 2016-08-25:
Or you can just enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging. Copy the BT address and copy that into the sketch.

bbf3, 2016-08-25:
Yeah, you're right @Lauszus, if you have access to the Arduino and want to reprogram it every time you need to switch/replace BT dongles. My Arduino is built into equipment and accessing and programming it is not as handy as having a laptop nearby. 

The method outlined above provides (in my opinion) a more fool-proof procedure for PS3 controller/dongle pairing than plugging and unplugging the dongle and controller into the UHS a bunch of times (which even with serial debugging does not always work with certain dongles, in my experience).

Lauszus, 2016-08-25:
Ahh okay makes sense then. You could easily modify the code to blink an LED or similar though :)

closed 2016-08-25 by Lauszus