felis/USB_Host_Shield_2.0 #256: GPX PIN #6

didix21, 2016-09-12:
Where exactly is defined the GPX PIN and can I changed it? For example, I would like to change D8 by D3.

Lauszus, 2016-09-15:
The GPX pin is not used by the library. Please read: https://github.com/felis/USB_Host_Shield_2.0#interface-modifications.

closed 2016-09-15 by Lauszus

xxxajk, 2016-09-15:
Actually it is used, for power control.

Lauszus, 2016-09-15:
Oh yeah you are right :) But that is only if it is added, the default shield does not have a power switch.

Lauszus, 2016-09-15:
Anyway it is handled in this file: https://github.com/felis/USB_Host_Shield_2.0/blob/master/usbhost.h. And here are the commits that added the support: https://github.com/felis/USB_Host_Shield_2.0/commit/b30c56a86e096ef3147cebc66e761f6ad2b1acfd and https://github.com/felis/USB_Host_Shield_2.0/commit/a57215fb727e481c75b6b4625c4c2f257a357e95.

xxxajk, 2016-09-15:
It will try to use it on start up/init, after that, if it is unused, the user is free to reprogram it. UHS3 may not end up that way, though. There is no way to detect if the switch is actually present, which is a shame.