felis/USB_Host_Shield_2.0 #257: What Arduino Mega 2560 pins are being utilized?

irbigturtle, 2016-09-22:
I'm using the depreciated ArduinoUSBHostShield and everything is working ok, but I am experiencing conflicts with another Shield/software.

While there are technically no pin conflicts between the two shields after I re-routed INT (Pin 9) to P45, I am still experiencing issues. 

```
//typedef MAX3421e<P10, P9> MAX3421E; // old
typedef MAX3421e<P10, P45> MAX3421E; // new
```

Is the software utilizing additional pins? I am only aware of these:

```
ICSP (SCK, MISO, MOSI)
P10 - SS
P7 - GPX?
P8 - RES?
P9 - INT (re-routed to P45)
```

xxxajk, 2016-09-23:
RES == /RESET line
GPX ==USB Vbus switch on 2.0.1 shield
Neither are connected to any pin.

irbigturtle, 2016-09-23:
@xxxajk So the only pins in use should be ICSP, P10 and P9?

Any ideas come to mind if I say that USB host shield works fine, but a CAN BUS shield stacked with it will not function until the host shield is removed? They have no pins in common. It FEELS like either the USB shield/software is not releasing the SPI bus appropriately, but I'm unsure of how to test that.

xxxajk, 2016-09-23:
1: Does the other shield use 3.3v? You may not have enough there to support both.
2: Try using the extrnal power plug instead of plain USB for power. USB will only supply 500mA, whereas if you use the extrenal power option, you can source 800mA. Use at least a 1A power supply @ 7.5VDC, but no higher than 12VDC.

irbigturtle, 2016-09-23:
@xxxajk 
1. I'm not sure if it uses 3.3v, but it does have it listed in the pinout. I'm assuming the usb host shield probably uses it as well? 
2. External power is also something that I can try.

Thanks for your input!

xxxajk, 2016-09-23:
The MAX3421E uses 3.3v, yes. 
Over the course of seeing over 1000 Arduino Megas go by (yes really, they are used in a product) I have also found a rare case where something is wrong with the 3.3v regulator on the MCU board, which causes a nasty ripple on the 3.3v line. 

This can also happen when using external power on the 5v supply. 
I have never seen this happen though, because the product powers differently. The 5v regulator is removed, as well as the mosfet for switching in 5v from USB. It uses 5v supplied directly.

Lauszus, 2016-09-23:
@irbigturtle my guess would be that the other library you are not using is not using the Arduino SPI transaction API, as this library is already going: https://github.com/felis/USB_Host_Shield_2.0/pull/137. This will automatically release the SPI bus.

What other library are you using exactly? Please send the link.

irbigturtle, 2016-09-23:
@xxxajk That's a pretty definitive sample size. I would assume that if a ripple was present, it would affect the shields individually as well. Not always true, but it makes my odds a little better :)

@Lauszus I've pretty new to the platform, so I'm not 100% sure, but it looks like it's using the native API: https://github.com/Seeed-Studio/CAN_BUS_Shield/blob/master/mcp_can.cpp

xxxajk, 2016-09-23:
Which version of seeed's shield are you using? Are you sure that the ICSP solder pads are correct on the bottom?

irbigturtle, 2016-09-23:
@xxxajk I'm using the latest (V1.2)

irbigturtle, 2016-09-23:
@xxxajk it should default to the ICSP header with pin 9 as the SS. Maybe I should further inspect the hardware...

Lauszus, 2016-09-24:
@irbigturtle the USB Host shield is using the SPI pins and pin 9 and 10 on the Arduino Mega, but pin 9 and 10 can easily be rerouted. Please see: https://github.com/felis/USB_Host_Shield_2.0#interface-modifications.

irbigturtle, 2016-09-24:
@Lauszus Yep! I already tried re-routing both of them (changed in usbcore) and it still didn't work. I can init the usb shield but it kills the CAN shield. If I physically remove the usb shield then the CAN shield starts working again.

When I get a chance, my next test will be physically re-routing/cutting any pin used by CAN (besides SPI). Maybe the usb shield has a component that is messing with it. A resistor on P2 perhaps? Something like that.

irbigturtle, 2016-10-01:
I never got around to playing with other connections. Still doesn't work, and ATM I suspect the CAN shield is to blame.

didix21, 2016-10-04:
As I understand you are trying to use a CAN port and Arduino Host Shield both at the same time with the same pins SPI ? isn't it ?

Lauszus, 2017-02-04:
Closing this, as it sounds like an issue with the CAN shield.

closed 2017-02-04 by Lauszus