felis/USB_Host_Shield_2.0 #262: ESP8266-Arduino support?

huming2207, 2016-10-24:
Hello there,

I've got a compile error for targeting to ESP8266 boards using ESP8266 Arduino SDK (a.k.a ESPduino, see here: [https://github.com/esp8266/Arduino](https://github.com/esp8266/Arduino)) 

Here are some information for my environment: 
Arduino IDE: 1.6.11
Host OS: Mac OSX 10.12
USB Host Shield library version: 1.2.1
**Target: ESP8266 NodeMCU 1.0 Board, 4MB Flash**

...and here is the compilation log:
```
In file included from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/Usb.h:34:0,
                 from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/hid.h:20,
                 from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/hidboot.h:20,
                 from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino:1:
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/avrpins.h:1246:2: error: #error "Please define board in avrpins.h"
 #error "Please define board in avrpins.h"
  ^
In file included from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/Usb.h:36:0,
                 from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/hid.h:20,
                 from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/hidboot.h:20,
                 from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino:1:
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:42:26: error: #elif with no expression
 #elif SPI_HAS_TRANSACTION
                          ^
In file included from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/Usb.h:36:0,
                 from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/hid.h:20,
                 from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/hidboot.h:20,
                 from /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino:1:
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:101:2: error: #error "No SPI entry in usbhost.h"
 #error "No SPI entry in usbhost.h"
  ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:155:24: error: #if with no expression
 #if SPI_HAS_TRANSACTION
                        ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:165:26: error: #elif with no expression
 #elif SPI_HAS_TRANSACTION
                          ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:186:24: error: #if with no expression
 #if SPI_HAS_TRANSACTION
                        ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:198:24: error: #if with no expression
 #if SPI_HAS_TRANSACTION
                        ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:207:26: error: #elif with no expression
 #elif SPI_HAS_TRANSACTION
                          ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:239:24: error: #if with no expression
 #if SPI_HAS_TRANSACTION
                        ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:261:24: error: #if with no expression
 #if SPI_HAS_TRANSACTION
                        ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:275:44: error: operator '||' has no right operand
 #elif !defined(SPDR) || SPI_HAS_TRANSACTION
                                            ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:288:24: error: #if with no expression
 #if SPI_HAS_TRANSACTION
                        ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:300:24: error: #if with no expression
 #if SPI_HAS_TRANSACTION
                        ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:309:26: error: #elif with no expression
 #elif SPI_HAS_TRANSACTION
                          ^
/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/usbhost.h:351:24: error: #if with no expression
 #if SPI_HAS_TRANSACTION
                        ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

```

Hope you can fix this, thank you!

Regards,
Jackson

xxxajk, 2016-10-24:
Planned for UHS 3.0

huming2207, 2016-10-25:
@xxxajk  Okay great! Thanks for your reply!

BTW I the HID stuff haven't been implemented yet, right? I didn't quite understand what's that means in the HID example.  (This file: [https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_HID/examples/USB_HOST_SHIELD/UHS_HID/UHS_HID.ino](https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_HID/examples/USB_HOST_SHIELD/UHS_HID/UHS_HID.ino)) Is it still has the same API as the old one, the UHS2.0?

Regards,
Jackson

xxxajk, 2016-10-25:
Correct, right now there is no HID. 
The API has not been discussed yet by anyone... You are the first to request it.
The time is now to break anything annoying in UHS2's API, and move to something better, unless you like the old way. Briefly, that means it is open for comment/discussion, and ofcourse any help.

xxxajk, 2016-10-25:
That said, I'll now post what I have in mind.

HID-RAW interface driver, will be the root for all interfaces afterwards.
This gives us some avenues to explore.

- If a HID interface is not supported, you can still fiddle with it!
- Possibly use HID raw as the interface driver bridge. -- no need to know deeper USB details.
- Interface drivers that do exist can take precedence over HID raw for special circumstances.

xxxajk, 2016-10-25:
Oh yeah, please move the discussion to UHS30 :-)

Lauszus, 2016-10-25:
This commit: https://github.com/felis/USB_Host_Shield_2.0/commit/0b3f4d925bde0034a416752e9b1d81d3914b6427 should add support for the ESP8266.

Currently it just uses pin 10 and 9 for SS and INT respectively: https://github.com/felis/USB_Host_Shield_2.0/blob/master/UsbCore.h#L43. Since I do not have much experience with the ESP8266 I just used the current default pins. @huming2207  you have a better proposal to what the default pins should be?

Lauszus, 2016-10-25:
@huming2207 please test the code in the following branch: https://github.com/felis/USB_Host_Shield_2.0/tree/ESP8266 and report back if it is working or not, as I don't have a physical board at hand.

Lauszus, 2016-10-25:
See: https://github.com/felis/USB_Host_Shield_2.0/pull/263

huming2207, 2016-10-25:
@Lauszus I'm so sorry mate...I'm not at home now, so I cannot test the ESP8266 USB shield for you. But I've got a NodeMCU board (ESP12E module with CP2102 serial to UART converter) now on my hand, and the compilation did pass successfully anyway. 

There is a strange thing that I haven't connect the USB shield (and anything else) to my NodeMCU. It only prints the words **"Start"** and nothing follows on. If I don't have the USB shield board connected to Arduino, will it also prints **"OSC did not start"**?

Here is the successful compile output:

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/Jackson/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Jackson/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Jackson/Documents/Arduino/libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10611 -build-path /var/folders/fn/q_bqjgkd3y741hxbk3z6hpnm0000gn/T/build91a5071480d2520b0daefb4fea08aea3.tmp -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/Jackson/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/Users/Jackson/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -prefs=runtime.tools.esptool.path=/Users/Jackson/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9 -verbose /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_2.0-ESP8266/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/Jackson/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Jackson/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Jackson/Documents/Arduino/libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10611 -build-path /var/folders/fn/q_bqjgkd3y741hxbk3z6hpnm0000gn/T/build91a5071480d2520b0daefb4fea08aea3.tmp -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/Jackson/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/Users/Jackson/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -prefs=runtime.tools.esptool.path=/Users/Jackson/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9 -verbose /Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_2.0-ESP8266/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino
Using board 'nodemcuv2' from platform in folder: /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0
Using core 'esp8266' from platform in folder: /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0
Detecting libraries used...
"/Users/Jackson/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/var/folders/fn/q_bqjgkd3y741hxbk3z6hpnm0000gn/T/build91a5071480d2520b0daefb4fea08aea3.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10611 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-I/Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-I/Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/nodemcu" "/var/folders/fn/q_bqjgkd3y741hxbk3z6hpnm0000gn/T/build91a5071480d2520b0daefb4fea08aea3.tmp/sketch/USBHIDBootKbd.ino.cpp" -o "/dev/null"
"/Users/Jackson/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/var/folders/fn/q_bqjgkd3y741hxbk3z6hpnm0000gn/T/build91a5071480d2520b0daefb4fea08aea3.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10611 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-I/Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-I/Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/nodemcu" "-I/Users/Jackson/Documents/Arduino/libraries/USB_Host_Shield_2.0-ESP8266" "/var/folders/fn/q_bqjgkd3y741hxbk3z6hpnm0000gn/T/build91a5071480d2520b0daefb4fea08aea3.tmp/ske [... cut ...]
................................................................................ [ 34% ]
................................................................................ [ 68% ]
.........................................................................        [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
```

Anyway thanks for your works and patches!

Regards,
Jackson.

xxxajk, 2016-10-25:
Does it work? My 8266 doesn't have pins 9 and 10....

xxxajk, 2016-10-25:
@Lauszus See hangouts chat please :-)

huming2207, 2016-10-25:
@Lauszus I was just to start to use ESP8266 stuff these days.  For the NodeMCU (ESP12/ESP12E modules) pinout, I guess it should be better if we can use the pinout defined by ESP8266-Arduino project themselves?

See here: [https://github.com/esp8266/Arduino/blob/master/variants/generic/common.h](https://github.com/esp8266/Arduino/blob/master/variants/generic/common.h)

They've set the pins like this:

```
static const uint8_t SS    = 15;
static const uint8_t MOSI  = 13;
static const uint8_t MISO  = 12;
static const uint8_t SCK   = 14;
```
Maybe you should specify these stuff in the library? 

Jackson

huming2207, 2016-10-25:
@xxxajk Some ESP modules removed some of the pins. For ESP12E they have those pins.

xxxajk, 2016-10-25:
Then we need a patch that can work on all of them, what pins do you suggest?

Lauszus, 2016-10-25:
Maybe we could use pin 15 for SS and pin 16 for INT?

xxxajk, 2016-10-25:
*seems* to be the standard.

Lauszus, 2016-10-25:
Yes all variants include common.h except the wifio variant: https://github.com/esp8266/Arduino/blob/master/variants/wifio/pins_arduino.h, so I might just make a specific check for that.

Lauszus, 2016-10-25:
It does not seem like it is supported anyway: https://github.com/esp8266/Arduino/blob/master/boards.txt.

xxxajk, 2016-10-25:
cool :-)

Lauszus, 2016-10-25:
Please see: https://github.com/felis/USB_Host_Shield_2.0/pull/263/commits/238c26908f116ee21ea5f5b320aa46e703eee3ec.

Lauszus, 2016-10-25:
Note that we can not use the pins defined in common.h, but at least this has been fixed in newer version of the Arduino core: https://github.com/arduino/Arduino/pull/4814. I will send a PR to the ESP8266 repository tomorrow.

huming2207, 2016-10-25:
@Lauszus Got it, thanks mate. I'll test it later when I return back home. 

By the way, ESP8266/8285 platforms have a lot of variants. I'm not quite familiar with the C/C++ and those hardware communication stuff, but I guess you may need to find more details before you type your code.

Lauszus, 2016-10-25:
Yes I know, but all modules should have at least those pins: https://github.com/esp8266/Arduino/blob/master/variants/generic/common.h#L30 and https://github.com/esp8266/Arduino/blob/master/variants/generic/common.h#L34. If the SPI pins are not broken out, then there is much we can do anyway.

Lauszus, 2016-10-25:
Btw this commit: https://github.com/felis/USB_Host_Shield_2.0/pull/263/commits/231fb542a8f42d5ada6325bdb5a3a2e1d8d37348 ensures that it will just use the pin defines in the future.

huming2207, 2016-10-25:
@Lauszus Nice, thanks for your patches!

Lauszus, 2016-10-26:
You're welcome. Let me know when you have had time to test it ;)

huming2207, 2016-10-26:
@Lauszus Yep but still have about 20+ days to go, so it will takes a while...sorry about that...

Lauszus, 2016-10-26:
No problem ;)

Lauszus, 2016-10-26:
Opened PR's upstream as well: https://github.com/esp8266/Arduino/pull/2639 and https://github.com/esp8266/Arduino/pull/2640.

xxxajk, 2016-10-26:
@Lauszus :+1:  Go get 'em :-)

huming2207, 2016-10-26:
I guess the developers in ESP8266-Arduino project are busy coding for ESP32 platform? They seemed haven't check the pull request for several week...

xxxajk, 2016-10-26:
That is a possibility.

huming2207, 2016-10-26:
If it's important, we can just call the ESP8266 SDK functions **(the one that ESP8266-Arduino based on)** instead. Here is the SDK v2.0.0 's reference: [https://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf](https://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf)

huming2207, 2016-10-26:
I mean, instead of submitting patches to ESP8266-Arduino, we should call the platform SDK's functions. Sometimes it's even better if someone need this library (UHS2.0) in some non-Arduino ESP8266 projects.

xxxajk, 2016-10-26:
that causes code complexity, though, and UHS2.0 really isn't divided well enough so that things like this don't cause confusion.

Lauszus, 2016-10-26:
I agree in limiting it to the Arduino core for now. Anyway the code should work as is, I just sent those PR's, so it can be fixed in the future.

huming2207, 2016-10-26:
Yep hope so. Espressif themselves and those fanboys won't give up ESP8266 too soon.

huming2207, 2016-11-02:
@Lauszus Hello again, I've found an issue today, I suggest that we should use some other GPIOs for INT instead of GPIO16, some platforms will use this pin as "awake" pin and connect the pin to RESET pin to wake the ESP8266 itself from sleeping. By the way using GPIO0 and 2 probably is not a good idea. Users may use it as debug serial pin (GPIO2) and bootloader enter pin (GPIO0).

huming2207, 2016-11-02:
@Lauszus  Maybe GPIO works, but I'm not sure, need to evaluate later.

huming2207, 2016-11-02:
@Lauszus Maybe GPIO 0 works, but I'm not sure, need to be evaluated later when I'm back.

huming2207, 2016-11-02:
@Lauszus It was proved that GPIO16 WILL NOT work for attaching interrupt, see here: [http://esp8266.github.io/Arduino/versions/2.3.0/doc/reference.html#digital-io](http://esp8266.github.io/Arduino/versions/2.3.0/doc/reference.html#digital-io)

They've mentioned that : 

> Pin interrupts are supported through attachInterrupt, detachInterrupt functions. Interrupts may be attached to any GPIO pin, **except GPIO16**.

Lauszus, 2016-11-02:
@huming2207 INT is not used as a interrupt anyway. It is simply polled.

huming2207, 2016-11-02:
@Lauszus Ahh...okay...so is that means I've made a mistake that "INT" is not a interrupt pin? 😂

Lauszus, 2016-11-03:
No, in this version of the USB Host Shield library it is not connected to an interrupt pin. However this is the case for 3.0.

huming2207, 2016-11-03:
Okay got it... If I do not connect this pin, what will happens? Will it works?

xxxajk, 2016-11-03:
Pin 10 is polled in uhs1 and 2. So... you need it connected to some pin for
it to decide to run a shorter code path.

huming2207, 2016-11-03:
Okay now I understand, thanks mates!

Now I've just started making a PCB for my project. It's a module for converting USB keyboard to bluetooth via a special HC-05 bluetooth module. It uses ESP-12E module and I've reserved GPIO0 as INT pin for USB shield. So far I don't have much free time to start coding, but I've done a Eagle schematic and PCB project. See here for the schematic: [Github - HyperaBt-Hardware](https://github.com/huming2207/HyperaBt-Hardware)

commit f087a578 (2016-10-25): Added support for ESP8266

Fixes #262
closed 2017-06-14 by Lauszus