felis/USB_Host_Shield_2.0 #264: Bluetooth 4.0 dongle adapter

oivinds, 2016-10-25:
Will it work? Or is it only Bluetooth 2.0 that will work with the library.

xxxajk, 2016-10-25:
Good question. Do you have one to try? If not, I'll see if I have one here.

oivinds, 2016-10-25:
Thanks!, No I have not, but the local shop has only 4.0. I have though tried with a dongle based on Logitech unified receiver that (seemingly) did not work.

xxxajk, 2016-10-25:
ok, I found one here. I'll let you know shortly.

Lauszus, 2016-10-25:
Yes Bluetooth 4.0 dongles work with the library.

xxxajk, 2016-10-25:
Mine doesn't, actually...
Vendor  ID:             19FF
Product ID:             0239
Revision ID:            0112

xxxajk, 2016-10-25:
All interfaces on it are Vendor Specific Class, except the last one which is used for updating....
Even Linux hates it :8ball:

xxxajk, 2016-10-25:
Apparently this particular dongle wants firmware sent to it.... lovely :-)
Just be aware of this.

oivinds, 2016-10-25:
Thanks guys!

closed 2016-10-25 by oivinds