felis/USB_Host_Shield_2.0 #265: AfterGlow PS3 Controller

drsmall, 2016-10-28:
So I'm using a wireless AfterGLow PS3 controller. It is wireless via a 2.4ghz connection.
After setting the correct VID and PID, I was able to connect successfully, but the button mapping is all messed up. Moving left stick causes PS, Triangle, Circle, L1, R1 etc to all fire randomly. Pressing Left triggers Select and Start. Things like that. I'm using PS3USBCustom. Is there an easy fix to this? Or do I need to remap everything?

Lauszus, 2016-10-29:
Yeah you will have to write a new driver for it if the mapping is different.

drsmall, 2016-10-29:
Does the library have a function that can give me the hex value of the button being pressed?

drsmall, 2016-10-29:
The more I think about it, the more I feel something else is afoot. I mean a ps3 controller is a ps3 controller. So it should use the same data mappings. Since it's a wireless controller, could it be adding an additional piece of data when it transmits. Maybe something like that is the issue?

Lauszus, 2016-10-29:
Just use ```Serial.print(val, HEX);``` or you can also use ```PrintHex<uint8_t > (val, 0x80);```.

It might be using the Bluetooth mapping. Please see the following comment: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3Enums.h#L90. Try to remove it all the places where is says ```-9``` i.e: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3USB.cpp#L328, https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3USB.cpp#L336, and https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3USB.cpp#L361.

xxxajk, 2016-10-29:
Well, yes, and no...
Yes a USB joystick class has a mapping list in the device.
No, UHS2 doesn't use it. IMHO this is a design flaw, BUT it was the only way to get it to work on RAM starved microcontrollers.

drsmall, 2016-10-29:
@Lauszus that didn't change anything, which is strange. I figured that make something break or at least change the behavior.

Lauszus, 2016-10-31:
You should uncomment this line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3USB.cpp#L21 then. It allows you to see the raw incoming data, then you should be able to map the data yourself.

Lauszus, 2017-10-12:
Closing this, as I believe the issue is solved.

closed 2017-10-12 by Lauszus