felis/USB_Host_Shield_2.0 #268: USB_desc not connecting to FTDI based Leica Microscope

fcollman, 2016-11-04:
Hi, I am trying to use a USB host shield to connect to a Leica DMI-8 microscope, which uses a FTDI based usb-serial chip, based upon the VID and PID i see when i connect it to my PC.  However, when connected to the USB-host shield, the USB_dev program fails to detect the device, although the same shield connected to the same Arduino Uno can detect other USB device (i.e. another arduino or a mouse) just fine.

I tried to dive into the code a bit, and put some serial debug statements in USB.Task and discovered that the Leica scope caused usb_task_state to oscillate between USB_DETACHED_SUBSTATE_INITIALIZE and USB_ATTACHED_SUBSTATE_SETTLE, but never triggered the USB_ATTACHED_SUBSTATE_RESET_DEVICE.

I would love to debug further what is going wrong, and figure out how to fix it, but i'm a bit stuck.

fcollman, 2016-11-04:
FWIW the VID and PID are
usb\vid_0403&pid_6001

and the compatilbe Ids are
USB\Class_ff&SubClass_ff&Prot_ff

xxxajk, 2016-11-05:
Just curious, have you tried it under the most recent UHS 3.0?
While far from perfect (can be hit and miss there too) it would be helpful to know if it can work with the CDC-ACM demo, which should automatically be able to use FTDI.

fcollman, 2016-11-06:
no i have not.. but i will give it a shot, if i am understanding the documentation correctly and i simply can use the same hardware, with pins 9 and 3 connected. I'll let you know what i find.

xxxajk, 2016-11-06:
That is correct you are using a mega 2560 right?

fcollman, 2016-11-06:
No.. i'm using a an arduino uno, which has a ATmega328P.  Would you
recommend something else?

xxxajk, 2016-11-06:
Try it. Mega2560 will fit better, though... lack of flash is the main
problem. USB tends to be a little heavy at this point. The extra RAM is
helpful too.

fcollman, 2016-11-06:
So i was able to get it to compile and upload onto my Uno, and got the  descriptor_parser program to work with the USB3.0 library, though i had to change a few things such as adding some const to lines  like this
const char * const usage_pages [] PROGMEM =
and changing Spi.h to SPI.h

but it did compile and upload, and did work when i plugged in another arduino into the USB port, but the Leica is still not giving me a device descriptor from the same program.

xxxajk, 2016-11-06:
Descriptor parser..... well, that's built-in to UHS3.0 and works totally different :-)
I do need to have a proper example for that I guess, if you think it would be helpful to have.
Let's move this discussion to the UHS30. Post there and we'll go on from that point.