felis/USB_Host_Shield_2.0 #270: Support for Teensy 3.6 is missing

BlueGene00, 2016-11-26:
Hello,

for the Teensy 3.6 there are some new definitions necessery. I get this error:

```
In file included from D:\Programme\Arduino\libraries\USB_Host_Shield_2.0-master/Usb.h:34:0,

                 from D:\Programme\Arduino\libraries\USB_Host_Shield_2.0-master/XBOXRECV.h:23,

                 from D:\Programme\Arduino\libraries\USB_Host_Shield_2.0-master\examples\Xbox\XBOXRECV\XBOXRECV.ino:8:

D:\Programme\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:1134:2: error: #error "Please define board in avrpins.h"

 #error "Please define board in avrpins.h"

  ^

In file included from D:\Programme\Arduino\libraries\USB_Host_Shield_2.0-master/Usb.h:36:0,

                 from D:\Programme\Arduino\libraries\USB_Host_Shield_2.0-master/XBOXRECV.h:23,

                 from D:\Programme\Arduino\libraries\USB_Host_Shield_2.0-master\examples\Xbox\XBOXRECV\XBOXRECV.ino:8:

D:\Programme\Arduino\libraries\USB_Host_Shield_2.0-master/usbhost.h:101:2: error: #error "No SPI entry in usbhost.h"

 #error "No SPI entry in usbhost.h"

  ^
```

Lauszus, 2016-11-27:
You need to add support for it similar to what was done for the Teensy LC: https://github.com/felis/USB_Host_Shield_2.0/pull/170/files. A PR is more than welcome.

Lauszus, 2016-11-27:
Please test #271 and confirm that it works.

xxxajk, 2016-11-27:
I'll test this asap. I have 3.6's here.

xxxajk, 2016-11-27:
@BlueGene00  BTW, UHS3 already supports 3.6's full speed port already. I'm a bit busy at the moment, so the high speed port is on the back burner currently. We could use some help porting drivers, even if they are only skeletal. You seem to have some skills that could help out. Have you taken a look at it?

BlueGene00, 2016-11-27:
Like Lauszus posted here:
https://github.com/felis/USB_Host_Shield_2.0/pull/271/commits/6ec044c23045678d5734ca89458930916d097d12

Before that, I have modified the library by myself. It seems, that I have done everything right. I added  `defined(__MK66FX1M0__)` (Teensy 3.6) to avrpins.h and to the usbhost.h. To support one pin which I needed, I also added:

`+MAKE_PIN(P37, CORE_PIN37_PORTREG, CORE_PIN37_BIT, CORE_PIN37_CONFIG);`

This seems to work, because I now can control my XBOX-controller with the Teensy 3.6 and the mini USB-host shield: https://www.youtube.com/watch?v=2KfGNg0H07Y

I will add the other changes also into the library and test it again, but I don't expect any issues then.

BlueGene00, 2016-11-29:
Another question: The Teensy 3.6 has a build-in USB-Host-port (pins). Would it be somehow possible to get this working with the USB-Host-library? It would save an additional shield, which could make projects more cheat and compact. Why not use something, when it is already there ;)

xxxajk, 2016-11-29:
UHS 3.0 will be supporting this. 2.0 simply can't. 
UHS 3.0 already supports the full/low speed port (the one that you use to upload your sketches) on all teensy boards that use the freescale arm chip.

closed 2016-12-26 by Lauszus