felis/USB_Host_Shield_2.0 #272: teensy3.6 and Mini USB Host Shield

leoneckert, 2016-12-15:
Hello! Could you give me a hint into the right direction for a project I am working on? 

I am using a teensy 3.6 and this [Mini USB Host Shield](https://www.circuitsathome.com/wp/wp-content/uploads/2011/02/uhm11_manual.jpg). I want the teensy to register keystrokes from a keyboard that's connected to the Mini USB Host Shield. Then, the teensy will send those keystrokes on to the computer. 
I connected shield and teensy as described in [https://www.circuitsathome.com/mcu/teensy-3-0-now-supported-by-the-usb-host-library/](https://www.circuitsathome.com/mcu/teensy-3-0-now-supported-by-the-usb-host-library/ ) (I did also experiment changing a few pins as I found them hardcoded in some files of this library). the board_qc seems to not throw any errors, but I am stuck which example/code to run to actually get the keystrokes in. I tried USBHIDBootKbd, but don't get any prints after the initial "Start". 

Any advice regarding wiring and/or code would be hugely appreciated. Thanks!

this doesn't change a lot but here is a photo of my current connection, though as I said, I changed the order of ss miso mosi sck when I saw them differently hard coded in here as in the link above. 
[image]

xxxajk, 2016-12-15:
> com/mcu/teensy-3-0-now-supported-by-the-usb-host-library/ (I did also

Make sure the bus is 5volts not 3.3 volts this will probably solve your
problem. Let us know if that works for you or not. Please know that right
now work is beginning on Native support for the 3.6

leoneckert, 2016-12-15:
Thanks @xxxajk ! I assume that would refer to the "USB Host Power Voltage" section in [here](http://www.pjrc.com/teensy/td_libs_USBHostShield.html). To make sure the port gets the full 5 volt from the laptop while teensy and shield run in 3.3?

[image]

xxxajk, 2016-12-15:
Yes exactly that is what you need to do.

<https://cloud.githubusercontent.com/assets/14130931/21241479/ceda47e0-c2dd-11e6-9888-5c309e4db537.png>

leoneckert, 2017-01-22:
Hey, sorry I have to ask this. I am sure I have all wired up correctly. 

I am connecting:

|keyboard| ----> |USB Host Shield Mini| ----> |teensy 3.6| ------> |computer| 

I am jumping the power from teensy's usb port to the Shield's to make sure there is 5V on both, the keyboard lights up when plugging in making me think it has enough power. All seems right, but what is the simplest example script to use, all I want is to see the keystrokes I take printed into the teensy console (which I have open on my computer). 

I tried a bunch of the example scripts, [this one](https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino) sounded most promising, but all I get is "OSC did not start.". Can you point me to anything that is more likely to work? Am I missing something? 

Thanks so much for you help, I thought I could figure it out from here but hitting a wall...

Lauszus, 2017-01-23:
@leoneckert it sounds like an connection issue. Have you connected the Teensy to the shield, as described here: https://www.circuitsathome.com/mcu/teensy-3-0-now-supported-by-the-usb-host-library? Also did you remember to cut the VBUS jumper in order not to damage the MAX3421E chip? Please also see: https://github.com/felis/USB_Host_Shield_2.0/pull/263#issuecomment-266190628.

mjs513, 2017-01-24:
I just connected a 3.5 to the same host shield.  The only thing that gave me an issue at first is that I forgot to connect the reset to 3.3v.  Once I did that I got a DS4 to connect via the PS4USB.ino sketch no problem.

Lauszus, 2017-02-12:
I'll close this issue now, as it has been confirmed by several people that the Teensy 3.5 and 3.6 works.

closed 2017-02-12 by Lauszus