felis/USB_Host_Shield_2.0 #273: PS4 slim controller does not connect via USB

kleindiek, 2016-12-19:
changed PID to 0x09cc (old PS4 controller has PID 0x05c4) in PS4USB.h

PS4 controller still does not connect.
PS4 controller works fine with BT.

#define ENABLE_UHS_DEBUGGING 1 in settings.h
Output:
PS4 USB Library StartedHU Init
Addr:01
NC:01

[image]

Lauszus, 2016-12-20:
Could you run the following example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USB_desc/USB_desc.ino?

kleindiek, 2016-12-20:
Here is the output:

Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		054C
Product ID:		09CC
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		00E1
Num.intf:		04
Conf.value:		01
Conf.string:		00
Attr.:			C0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		0A
Type:		24
Contents:	01000147000201020C24
Unknown descriptor:
Length:		0C
Type:		24
Contents:	020101010602030000000A24
Unknown descriptor:
Length:		0A
Type:		24
Contents:	06020101030000000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030302040402000C24
Unknown descriptor:
Length:		0C
Type:		24
Contents:	020402040301000000000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	060504010300000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030601010105000904

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		07
Type:		24
Contents:	01010101000B24
Unknown descriptor:
Length:		0B
Type:		24
Contents:	020102021001007D000905

Endpoint descriptor:
Endpoint address:	01
Attr.:			09
Max.pkt size:		0084
Polling interval:	01
Unknown descriptor:
Length:		07
Type:		25
Contents:	01000000000904

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00

Interface descriptor:
Intf.number:		02
Alt.:			01
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		07
Type:		24
Contents:	01060101000B24
Unknown descriptor:
Length:		0B
Type:		24
Contents:	020101021001803E000905

Endpoint descriptor:
Endpoint address:	82
Attr.:			05
Max.pkt size:		0022
Polling interval:	01
Unknown descriptor:
Length:		07
Type:		25
Contents:	01000000000904

Interface descriptor:
Intf.number:		03
Alt.:			00
Endpoints:		02
Intf. Class:		03
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		21
Contents:	1101000122FB010705

Endpoint descriptor:
Endpoint address:	84
Attr.:			03
Max.pkt size:		0040
Polling interval:	05

Endpoint descriptor:
Endpoint address:	03
Attr.:			03
Max.pkt size:		0040
Polling interval:	05

Addr:1(0.0.1)

kleindiek, 2016-12-26:
Just verified that everything works with a borrowed old PS4 Controller.

Still would like to get the new SLIM PS4 Controller to work.

Lauszus, 2016-12-27:
Thanks. Could you run the following example as well: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHID_desc/USBHID_desc.ino. Also please upload the PS4USB example again while the following line is uncommented: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS4Parser.cpp#L33, so I can see if it is sending data or not.

I will look more into this issue after the holidays.

kleindiek, 2016-12-30:
Here is the output of https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHID_desc/USBHID_desc.ino

"Addr:1(0.0.1)
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		054C
Product ID:		09CC
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		00E1
Num.intf:		04
Conf.value:		01
Conf.string:		00
Attr.:			C0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		0A
Type:		24
Contents:	01000147000201020C24
Unknown descriptor:
Length:		0C
Type:		24
Contents:	020101010602030000000A24
Unknown descriptor:
Length:		0A
Type:		24
Contents:	06020101030000000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030302040402000C24
Unknown descriptor:
Length:		0C
Type:		24
Contents:	020402040301000000000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	060504010300000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030601010105000904

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		07
Type:		24
Contents:	01010101000B24
Unknown descriptor:
Length:		0B
Type:		24
Contents:	020102021001007D000905

Endpoint descriptor:
Endpoint address:	01
Attr.:			09
Max.pkt size:		0084
Polling interval:	01
Unknown descriptor:
Length:		07
Type:		25
Contents:	01000000000904

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00

Interface descriptor:
Intf.number:		02
Alt.:			01
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		07
Type:		24
Contents:	01060101000B24
Unknown descriptor:
Length:		0B
Type:		24
Contents:	020101021001803E000905

Endpoint descriptor:
Endpoint address:	82
Attr.:			05
Max.pkt size:		0022
Polling interval:	01
Unknown descriptor:
Length:		07
Type:		25
Contents:	01000000000904

Interface descriptor:
Intf.number:		03
Alt.:			00
Endpoints:		02
Intf. Class:		03
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		21
Contents:	1101000122FB010705

Endpoint descriptor:
Endpoint address:	84
Attr.:			03
Max.pkt size:		0040
Polling interval:	05

Endpoint descriptor:
Endpoint address:	03
Attr.:			03
Max.pkt size:		0040
Polling interval:	05

Addr:1(0.0.1)

kleindiek, 2016-12-30:
uploaded the PS4USB example again while the following line is uncommented: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS4Parser.cpp#L33.

Here is the output:
PS4 USB Library Started
HU Init
Addr:01
NC:01

The last three lines 

HU Init
Addr:01
NC:01

are replicated when the PS4 slim controller is unplugged and plugged in again.

kleindiek, 2017-01-10:
Were you able to look into this issue?

kleindiek, 2017-01-16:
Were you able to look into this? Would it make sense if I send you the new SLIM PS4 controller?

Lauszus, 2017-01-16:
@kleindiek I am really sorry for not looking into this issue. Yes that would be better, can you send me an email at: lauszus@gmail.com, so you can get my address?

clement360, 2017-01-22:
any updates on this? My PS4 PRO controllers are also not connecting. Thanks

Lauszus, 2017-01-23:
@clement360 still waiting for the controller to arrive. When it does I will look into this issue.

clement360, 2017-01-23:
Sounds Good! I'm going to try and find an older ps4 controller to test. to try and verify my system is not the issue

Lauszus, 2017-01-30:
@kleindiek and @clement360 please test #281 and report back.

clement360, 2017-01-31:
@Lauszus it works flawlessly, thank you!

clement360, 2017-01-31:
quick question though. I loaded up ps4bt and ran it, I was able to connect almost instantly when I hit ps4 and share at the same time. then I hit the ps4 button after playing around for a while and can no longer get it to reconnect. i've tried using the share and ps4 button as well as ps4 button alone. all after resetting the arduino and serial monitor

Lauszus, 2017-01-31:
Hmm I tried the same thing yesterday and everthing was fine. Can you try to print the angles (press OPTIONS) and see if the values change? Btw you only need to press the PS button in order to connect once they are paired.

clement360, 2017-01-31:
I haven't made any changes to the code at all. I can no longer connect at all. btw I'm testing two controllers

Lauszus, 2017-01-31:
Hmm it sounds like an issue with the Bluetooth dongle. I have one that does not work and another one that does...

I would recommend just buying a lot from a different sellers from eBay and then just keep the ones that work. They are really cheap though.

Also please make sure it is not a power issue. Please see: https://github.com/felis/USB_Host_Shield_2.0#faq.

xxxajk, 2017-01-31:
Some USB dongles require a firmware upload and that is why they do not
work. It totally depends on the chipset inside the dongle. Do a little
research before you buy it will go a long way to avoid headaches.

Lauszus, 2017-01-31:
@xxxajk another problem is that often Chinese  sellers will just sell old dongles that do not support 2.0+EDR. Normally dongles listed as Bluetooth 4 dongles work, but they are more expensive.

clement360, 2017-01-31:
just tried a different dongle and it seems to be working, however I have noticed that it works very well if the ps4 is in pairing mode before I start the monitor. does that make sense? is it possible to just have the arduino constantly search for connections? you guys have been very helpful thank you

xxxajk, 2017-01-31:
Yes that makes sense what you may want to do is constantly check if it is
paired and if not try to repair. That should fix your problem.

clement360, 2017-01-31:
I assume that would be using the PS4.pair() function or something like PS4(&Btd, PAIR);? where could I find the docs for the ps4 BT API?

Lauszus, 2017-01-31:
@clement360 please see: https://github.com/felis/USB_Host_Shield_2.0/issues/282#issuecomment-276489942.

A way to check if it is just waiting for an incoming connection is by checking the following variable: http://felis.github.io/USB_Host_Shield_2.0/class_b_t_d.html#aa7735da01865bab01b569ee836173737 like so:
```C++
if (Btd. watingForConnection) {
    Btd. watingForConnection = false;
    PS4.pair();
}
```

Lauszus, 2017-01-31:
@clement360 some information documenting the code is available here: http://felis.github.io/USB_Host_Shield_2.0/class_p_s4_b_t.html.

kleindiek, 2017-02-02:
I changed the two files according to https://github.com/felis/USB_Host_Shield_2.0/pull/281/files but get the folloging error message:

In file included from /Users/stephankleindiek/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/examples/PS4USB/PS4USB.ino:7:0:
/Users/stephankleindiek/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0/PS4USB.h:68:35: error: 'USBHID' has not been declared
virtual void ParseHIDData(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf) {
^
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Uno.

Lauszus, 2017-02-02:
@kleindiek please take a look at the following section of the README: https://github.com/felis/USB_Host_Shield_2.0#manual-installation, as it sounds like you have not included the library correctly.

kleindiek, 2017-02-02:
thanks. Did so, no errors.
enebled debugging, when I plug in the PS4 controller the following output is generated: 
HU Init
Addr:01
NC:01

Do I have to press a button on the PS4 controller?

Lauszus, 2017-02-02:
And you did apply the changes from the PR?

kleindiek, 2017-02-06:
Now I applied the changes to the two files PS4Parser.cpp and confdescparser.h.

Still no success.

I noticed however that the charging LED (white breathing light) showes up one time when I plug in the USB cable to the controller. I did not see that with earlier tries.

Lauszus, 2017-02-06:
@kleindiek can you try to uncomment these lines as well: https://github.com/felis/USB_Host_Shield_2.0/blob/e7a01565cdaed7b033dafcdd7742e82fcbafc069/confdescparser.h#L164-L168, so it prints the HID descriptor.

Also could you try to run the following example again: https://github.com/felis/USB_Host_Shield_2.0/blob/e7a01565cdaed7b033dafcdd7742e82fcbafc069/examples/USB_desc/USB_desc.ino and paste the output?

kleindiek, 2017-02-06:
This ist the output after uncommenting the five lines in confdescparser.h#L164-L168:
PS4 USB Library StartedHU Init
Addr:01
NC:01

HID Descriptor:
bDescLength: 09
bDescriptorType: 21
bcdHID: 0111
bCountryCode: 00
bNumDescriptors: 01
bDescrType: 22
wDescriptorLength: 01FB
Cnf:01
HU configured

kleindiek, 2017-02-06:
Output after running https://github.com/felis/USB_Host_Shield_2.0/blob/e7a01565cdaed7b033dafcdd7742e82fcbafc069/examples/USB_desc/USB_desc.ino:

var/folders/k0/c0wwtshd3_s_l3dct6bj88780000gn/T/arduino_modified_sketch_159756/sketch_feb06a.ino:3:24: fatal error: pgmstrings.h: No such file or directory
#include "pgmstrings.h"
^
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Uno.

Lauszus, 2017-02-06:
@kleindiek the file "pgmstrings.h" needs to be put next to the .ino file. Please see: https://github.com/felis/USB_Host_Shield_2.0/tree/e7a01565cdaed7b033dafcdd7742e82fcbafc069/examples/USB_desc.

kleindiek, 2017-02-06:
Here is the output of https://github.com/felis/USB_Host_Shield_2.0/blob/e7a01565cdaed7b033dafcdd7742e82fcbafc069/examples/USB_desc/USB_desc.ino:

Start

01
--

Device descriptor: 
Descriptor Length: 12
Descriptor type: 01
USB version: 0200
Device class: 00
Device Subclass: 00
Device Protocol: 00
Max.packet size: 40
Vendor ID: 054C
Product ID: 09CC
Revision ID: 0100
Mfg.string index: 01
Prod.string index: 02
Serial number index: 00
Number of conf.: 01

Configuration descriptor:
Total length: 00E1
Num.intf: 04
Conf.value: 01
Conf.string: 00
Attr.: C0
Max.pwr: FA

Interface descriptor:
Intf.number: 00
Alt.: 00
Endpoints: 00
Intf. Class: 01
Intf. Subclass: 01
Intf. Protocol: 00
Intf.string: 00
Unknown descriptor:
Length: 0A
Type: 24
Contents: 01000147000201020C24
Unknown descriptor:
Length: 0C
Type: 24
Contents: 020101010602030000000A24
Unknown descriptor:
Length: 0A
Type: 24
Contents: 06020101030000000924
Unknown descriptor:
Length: 09
Type: 24
Contents: 030302040402000C24
Unknown descriptor:
Length: 0C
Type: 24
Contents: 020402040301000000000924
Unknown descriptor:
Length: 09
Type: 24
Contents: 060504010300000924
Unknown descriptor:
Length: 09
Type: 24
Contents: 030601010105000904

Interface descriptor:
Intf.number: 01
Alt.: 00
Endpoints: 00
Intf. Class: 01
Intf. Subclass: 02
Intf. Protocol: 00
Intf.string: 00

Interface descriptor:
Intf.number: 01
Alt.: 01
Endpoints: 01
Intf. Class: 01
Intf. Subclass: 02
Intf. Protocol: 00
Intf.string: 00
Unknown descriptor:
Length: 07
Type: 24
Contents: 01010101000B24
Unknown descriptor:
Length: 0B
Type: 24
Contents: 020102021001007D000905

Endpoint descriptor:
Endpoint address: 01
Attr.: 09
Max.pkt size: 0084
Polling interval: 01
Unknown descriptor:
Length: 07
Type: 25
Contents: 01000000000904

Interface descriptor:
Intf.number: 02
Alt.: 00
Endpoints: 00
Intf. Class: 01
Intf. Subclass: 02
Intf. Protocol: 00
Intf.string: 00

Interface descriptor:
Intf.number: 02
Alt.: 01
Endpoints: 01
Intf. Class: 01
Intf. Subclass: 02
Intf. Protocol: 00
Intf.string: 00
Unknown descriptor:
Length: 07
Type: 24
Contents: 01060101000B24
Unknown descriptor:
Length: 0B
Type: 24
Contents: 020101021001803E000905

Endpoint descriptor:
Endpoint address: 82
Attr.: 05
Max.pkt size: 0022
Polling interval: 01
Unknown descriptor:
Length: 07
Type: 25
Contents: 01000000000904

Interface descriptor:
Intf.number: 03
Alt.: 00
Endpoints: 02
Intf. Class: 03
Intf. Subclass: 00
Intf. Protocol: 00
Intf.string: 00
Unknown descriptor:
Length: 09
Type: 21
Contents: 1101000122FB010705

Endpoint descriptor:
Endpoint address: 84
Attr.: 03
Max.pkt size: 0040
Polling interval: 05

Endpoint descriptor:
Endpoint address: 03
Attr.: 03
Max.pkt size: 0040
Polling interval: 05

Addr:1(0.0.1)

Lauszus, 2017-02-06:
@kleindiek it all looks alright. Have you tried pressing a button after the "HU configured" message?

Also can you try to connect an external power supply to your Arduino or use a powered hub: https://github.com/felis/USB_Host_Shield_2.0#faq?

kleindiek, 2017-02-06:
When I plug the USB cable from the controller to the Host Shield the output is:

HID Descriptor:
bDescLength:		09
bDescriptorType:	21
bcdHID:			0111
bCountryCode:		00
bNumDescriptors:	01
bDescrType:		22
wDescriptorLength:	01FB
Cnf:01
HU configured

I have tried an external power supply, same message as above.

And yes, I tried pressing all buttons after the „HU configured“ message appeared.

Again, the LED „breathes“ when I plug the cable in.

Lauszus, 2017-02-06:
Can you set this line to 1: https://github.com/felis/USB_Host_Shield_2.0/blob/master/hiduniversal.cpp#L403. Does it print anything?

kleindiek, 2017-02-07:
Yes, it prints a lot:
E 00 E1 FF FA FF 4D 00 32 20 F2 03 00 00 00 00 001 D9 8 00 00 80 00 

Buf: 01 7D 7F 7A 7D 08 00 78 00 00 98 20 F9 0D 00 DF FF F9 FF 78 00 40 20 E7 03 00 00 00 00 00 08 00 00 01 D9 81 � 00 80 00 

Buf: 01 7D 7F 7A 7D 08 00 78 00 00 98 20 F9 0D 00 DF FF F9 FF 78 00 40 20 E7 03 00 00 00 00 00 08 00 00 01 D9 81 � 00 80 00 

Buf: 01 7D 7F 7A 7D 08 00 78 00 00 98 20 F9 0D 00 DF FF F9 FF 78 00 40 20 E7 03 00 00 00 00 00 08 00 00 01 D9 81 �
PS4 USB Library StartedHU Init
Addr:01
NC:01

HID Descriptor:
bDescLength: 09
bDescriptorType: 21
bcdHID: 0111
bCountryCode: 00
bNumDescriptors: 01
bDescrType: 22
wDescriptorLength: 01FB
Cnf:01
HU configured

Buf: 01 7D 7F 7A 7D 08 00 84 00 00 1A 46 F9 0E 00 E1 FF F8 FF 57 00 35 20 0B 04 00 00 00 00 00 08 00 00 01 D9 81 3D A2 09 80 00 00 00 00 80 00 00 00 80 00 00 00 00 80 00 00 00 80 00 00 00 00 80 00 

Buf: 01 7D 7F 7A 7D 08 00 88 00 00 1A 46 F9 0E 00 E1 FF F8 FF 57 00 35 20 0B 04 00 00 00 00 00 08 00 00 01 D9 81 3D A2 09 80 00 00 00 00 80 00 00 00 80 00 00 00 00 80 00 00 00 80 00 00 00 00 80 00 

and so on...

Lauszus, 2017-02-07:
Okay. The data looks right. Now try to uncomment this line: https://github.com/felis/USB_Host_Shield_2.0/blob/e7a01565cdaed7b033dafcdd7742e82fcbafc069/PS4Parser.cpp#L33. What does it print?

kleindiek, 2017-02-07:
This line is already uncommented.

When commented the output is :

PS4 USB Library StartedHU Init
Addr:01
NC:01

HID Descriptor:
bDescLength: 09
bDescriptorType: 21
bcdHID: 0111
bCountryCode: 00
bNumDescriptors: 01
bDescrType: 22
wDescriptorLength: 01FB
Cnf:01
HU configured

Buf: 01 7D 80 7A 82 08 00 0C 00 00 45 91 04 0D 00 E2 FF F9 FF A2 00 54 20 0A 04 00 00 00 00 00 07 00 00 01 D9 81 3D A2 09 80 00 00 00 00 80 00 00 00 80 00 00 00 00 80 00 00 00 80 00 00 00 00 80 00

and so on.

When uncommented the output is:

PS4 USB Library StartedHU Init
Addr:01
NC:01

HID Descriptor:
bDescLength: 09
bDescriptorType: 21
bcdHID: 0111
bCountryCode: 00
bNumDescriptors: 01
bDescrType: 22
wDescriptorLength: 01FB
Cnf:01
HU configured

Buf: 01 7F 7D 7A 82 08 00 94 00 00 61 B6 06 0E 00 E0 FF F9 FF 93 00 4A 20 7C 04 00 00 00 00 00 08 00 00 00 00 80 00 00 00 80 00 00 00 00 80 00 00 00 80 00 00 00 00 80 00 00 00 80 00 00 00 00 80 00 

and so on.

There seems to be no difference.

Lauszus, 2017-02-07:
@kleindiek it really sounds like you have not added this line: https://github.com/felis/USB_Host_Shield_2.0/pull/281/files#diff-3f3376c03973f2c116f6423ff7d2a733R69.

Please download the code from that branch: https://github.com/felis/USB_Host_Shield_2.0/archive/PS4Slim.zip and it should work.

kleindiek, 2017-02-07:
Success!

You are correct, I didn't update PS4USB.h but PS4BT.h instead with the code of PS4USB.h. Sorry for the inconvenience.

Lauszus, 2017-02-07:
@kleindiek no problem. I'll merge the PR now then :)

closed 2017-02-07 by Lauszus