felis/USB_Host_Shield_2.0 #274: To make it work with Arduino Zero ?

Synless, 2017-01-02:
Hello everyone ! 

I've noticed that the [USB_Host 3.0](https://github.com/felis/UHS30) has some support for the Arduino Zero (the included examples compiled just fine), but that the 2.0 lakes it ( `#error "Please define board in avrpins.h"` when compiling). 
This is very unfortunate for such an awesome library as the Arduino Zero is a very nice ARM chip and one of the few Arduino compatible microcontroller with HID.

So I've played around with avrpins.h and tried to compared 2.0 and 3.0 but I ended up kind of lost. I'm not good enough in Arduino and C++ to get think working by myself. 

I someone could provide me any kind of info concerning the support of the Arduino Zero that would be great :)

Lauszus, 2017-01-02:
Please see: https://github.com/felis/USB_Host_Shield_2.0/pull/202#issuecomment-216075846.

Synless, 2017-01-03:
Thanks for your answer Lauszus.

closed 2017-01-03 by Synless