felis/USB_Host_Shield_2.0 #275: Question about usb hub

ernesto-xload, 2017-01-05:
Hello, just a question:

To handle a keyboard connected to a usb hub only is needed include usbhub.h library and create the instance with: "USBHub Hub1(&Usb)" after "USB     Usb"?

Best regards and a lot of thanks in advance.

xxxajk, 2017-01-05:
Simple answer is yes.

<https://github.com/felis/USB_Host_Shield_2.0/issues/275>, or mute the
thread

Lauszus, 2017-01-05:
This is also documented in the README: https://github.com/felis/USB_Host_Shield_2.0#faq.

closed 2017-01-05 by Lauszus

sudha100, 2019-05-02:
SIR IN THE USB HUB THE USB HUB  DEVICE DETAILS ARE NOT DISPLAYED

sudha100, 2019-05-02:
how to use USB HUB demo for connecting two or more devices with arduino.  and find the details of 

usb hub

and connected devices