felis/USB_Host_Shield_2.0 #276: ThrustMaster HOTAS Throttle only recieving NAK from Usb.Task()

j3nks, 2017-01-11:
Hello,

I am having trouble with the Thrustmaster HOTAS Throttle sending data back to the parse function. As in it doesn't.  So I drilled down and it appears that from the Poll() function located in the Usb.Task() I am only ever NAK for the rcode.  I can successfully send HID report to the Throttle to set LEDs and I successfully get the HID Report back which can be seen bellow.  Any help or suggestions would be greatly appreciated.

Thanks,
Matt

I attached the USB desc and HID desc reports.
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	08
Vendor  ID:		044F
Product ID:		0404
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0029
Num.intf:		01
Conf.value:		01
Conf.string:		00
Attr.:			80
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		03
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		21
Contents:	1101000122AB000705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0040
Polling interval:	08

Endpoint descriptor:
Endpoint address:	02
Attr.:			03
Max.pkt size:		0040
Polling interval:	08

Addr:1(0.0.1)

-----------------------------------------------------------------------------------------------------

Opening port
Port open
Start
HU Init
Addr:01
NC:01
Cnf:01
HU configured
0000: 05 01 09 04 A1 01 85 01 05 09 15 00 25 01 35 00 
0010: 45 01 19 01 29 20 75 01 95 20 81 02 05 01 09 39 
0020: 15 00 25 07 35 00 46 3B 01 65 14 75 04 95 01 81 
0030: 42 81 01 65 00 09 30 26 FF 03 46 FF 03 75 0A 95 
0040: 01 81 02 75 06 81 03 09 31 75 0A 81 02 75 06 81 
0050: 03 09 36 75 0A 81 02 75 06 81 03 09 32 26 FF 3F 
0060: 46 FF 3F 75 0E 81 02 75 02 81 03 09 35 75 0E 81 
0070: 02 75 02 81 03 75 10 27 FF FF 00 00 47 FF FF 00 
0080: 00 06 FF 00 09 01 09 02 09 03 09 04 09 05 95 05 
0090: 81 02 26 FF 00 46 FF 00 75 08 95 0A 81 03 09 BB 
00A0: 95 23 91 02 85 02 09 06 81 02 C0 
Usage Page Gen Desktop Ctrls(01)
Usage Game Pad
Collection Application
Report Id(01)
Usage Page Button(09)
Logical Min(00)
Logical Max(01)
Physical Min(00)
Physical Max(01)
Usage Min(01)
Usage Max(20)
Report Size(01)
Report Count(20)
Input(00000010)
Usage Page Gen Desktop Ctrls(01)
Usage Hat Switch
Logical Min(00)
Logical Max(07)
Physical Min(00)
Physical Max(3B01)
Unit(14)
Report Size(04)
Report Count(01)
Input(01000010)
Input(00000001)
Unit(00)
Usage X
Logical Max(FF03)
Physical Max(FF03)
Report Size(0A)
Report Count(01)
Input(00000010)
Report Size(06)
Input(00000011)
Usage Y
Report Size(0A)
Input(00000010)
Report Size(06)
Input(00000011)
Usage Slider
Report Size(0A)
Input(00000010)
Report Size(06)
Input(00000011)
Usage Z
Logical Max(FF3F)
Physical Max(FF3F)
Report Size(0E)
Input(00000010)
Report Size(02)
Input(00000011)
Usage Rz
Report Size(0E)
Input(00000010)
Report Size(02)
Input(00000011)
Report Size(10)
Logical Max(FFFF0000)
Physical Max(FFFF0000)
Usage Page Undef(FF)
Usage
Usage
Usage
Usage
Usage
Report Count(05)
Input(00000010)
Logical Max(FF00)
Physical Max(FF00)
Report Size(08)
Report Count(0A)
Input(00000011)
Usage
Report Count(23)
Output(00000010)
Report Id(02)
Usage
Input(00000010)
End Collection

j3nks, 2017-01-13:
Never mind this fixed it: https://github.com/felis/USB_Host_Shield_2.0/pull/174

Might want to consider pulling the fix into the USB.cpp in the master?

Lauszus, 2017-01-23:
@j3nks thanks for the feedback.

@xxxajk is there any reason why #174 can't be merged? I will have to test it on a bunch devices before it is merged though. Also this line might also be a bug: https://github.com/teamcarma/USB_Host_Shield_2.0/blob/b47c912d1a9ae02adf4ec98254c908813337de9a/Usb.cpp#L173, as also noted by the PR comment: https://github.com/felis/USB_Host_Shield_2.0/pull/174#issue-105813663.

xxxajk, 2017-01-23:
@Lauszus The solution is to flip it in the driver. This is similar to the FTDI problem that I fixed in UHS30.
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_CDC_ACM/UHS_CDC_ACM_INLINE.h#L185-L188

Lauszus, 2017-10-12:
See: #174.

closed 2017-10-12 by Lauszus