felis/USB_Host_Shield_2.0 #277: testusbhostFAT and Teensy3.5

mjs513, 2017-01-23:
I am trying to test out the sketch on the Teensy 3.5 and am receiving errors related to missing libraries. The missing libraries are as follows: 

#include <Storage.h>
#include <PCpartition/PCPartition.h>
#include <FAT/FAT.h>

xmem2 was also missing but I found that one on line. I did just see that it is not needed.

Thanks for your help
Mike

xxxajk, 2017-01-23:
It is much easier and more reliable to use UHS30.

https://github.com/felis/UHS30

It even supports the built-in USB when you use a micro USB host adapter.

xxxajk, 2017-01-23:
Oh yeah, one other major point here... it also supports multiple USB hosts at the same time.
You can mix the on-chip host and shield, or > 1 shield.

mjs513, 2017-01-23:
thanks for the info.  I downloaded the libraries but am confused on which sketch to run to test the usb thumb drive.  Right now its connected to the USB host shield.  Is there a link for little more description of the examples.

xxxajk, 2017-01-23:
UHS2's FS stuff is pretty complicated to wrap your head around. 
The examples that are provided are what they are. 
You'll have to dig around to find out more, and worse the FS stuff isn't maintained anymore.

UHS3's is WAY easier and includes many demos that explain things better, and is also maintained now.
UHS2 is being deprecated because it has several problems, and is harder to use.
UHS3's main goals are:

1. Ease of use with a better API that makes sense.
2. Better enumeration.
3. Common abstracted classes for groups of like devices, e.g. all serial devices use a single class.
4. Allow more advanced use within the grouped classes.
5. Allow users to hook in to classes by sub-classing and adding in common hook methods.
6. Easier to maintain.
7. Give users a chance to tell us what kind of interface and features they want.
8. Break away from the limitations imposed by previos versions, even if it breaks the old API.
9. Easier to develop new drivers and easy to port over old drivers.
10. Allow a driver to be developed right inside a sketch.

mjs513, 2017-01-23:
Thanks for getting back to me so fast.  I started playing around with the samples sketches for the usb host shield, new demo and test bulk.  They seem to start ok but then all I get out of them is:

Start.
SWI_IRQ_NUM 30
USB HOST READY.
USB HOST state 01
No media. Waiting to mount /
USB HOST state 02
USB HOST state 0a
USB HOST state 03
USB HOST state 0c
USB HOST state 0d
USB HOST state 60

I am getting only one warning:
 #warning "Your SPI library installation lacks 'SPI_ATOMIC_VERSION'. Please complain to the 
maintainer."

xxxajk, 2017-01-23:
That's correct. 
Don't worry about the warning... Complain to Paul, the more complaints we get the more likely he will fix this.
You need to format your media as FAT32 and no label for the demo to work at all. (this is explained in the UHS3 demos)

mjs513, 2017-01-23:
Thanks.  It is fat32, missed the label part.  Delete the label and it is working now.  One of the reasons that I like uhs2 right now is that it has all the examples to support Bluetooth and usb controllers like the PS4 which I am planning on using for a robotics project.  Its not clear how I would do that with UHS30.

xxxajk, 2017-01-23:
Those parts will be included as time permits. 
I'm currently waiting to collaborate with @Lauszus to do these things.
Perhaps your input on it will be helpful to us during porting, and you can let us know what you think would be easy to do... and I do hope to wrap bluetooth serial right into the current singular serial class, and then, as an example it won't matter if you use bluetooth serial, or FTDI. You'll just use one class for all of it.

mjs513, 2017-01-23:
I will be certainly glad to help in anyway I can.  Oh by the way think I spoke too soon with the demo sketch.  Thought it was working until I tried running it again and compared it to what it was suppose to print.  Here is what I am getting now that I am running it again:

Start.

SWI_IRQ_NUM 30

USB HOST READY.
USB HOST state 01
No media. Waiting to mount /
USB HOST state 02
USB HOST state 0a
USB HOST state 03
USB HOST state 0c
USB HOST state 0d
USB HOST state 60
/ mounted.
Removing '/HeLlO.tXt' file... completed with 0

Starting Write test...
File opened OK, fd = 1

and then it hangs.  Did notice that the teensy led is on but very faint.  

Mike

xxxajk, 2017-01-23:
Try UHS3 to compare. As I said, it is no longer maintained, so there very
well may be issues.

mjs513, 2017-01-23:
I am currently using uhs30 library from here: [https://github.com/felis/UHS30](https://github.com/felis/UHS30).  Is this the wrong one?

xxxajk, 2017-01-23:
That is correct.
Native is more tested than the host shield. I have to do more on what is going on.
Some stuff you can try...
`
#define UHS_DEVICE_WINDOWS_USB_SPEC_VIOLATION_DESCRIPTOR_DEVICE 0
`
This may help. Let me know if it does.
You can also turn on debugging in the sketch, please see the top of the sketch for details.

mjs513, 2017-01-23:
Just got home will try a little later and let you know.  Would you prefer if i move this over to the uhs30 github page.

xxxajk, 2017-01-23:
Absolutely, that is where the discussion belongs. Close this one and open a new issue on UHS30.
I was about to suggest it anyway.

closed 2017-01-23 by mjs513