felis/USB_Host_Shield_2.0 #282: PlayStation 4 controller - Bluetooth pairing without new code upload

BlueGene00, 2017-01-31:
Hello,

I am using the PS4 controller bluetooth library and it works very fine. Thank you very much for that.

But I have one point, where I need some help:

```
/* You can create the instance of the PS4BT class in two ways */
// This will start an inquiry and then pair with the PS4 controller - you only have to do this once
// You will need to hold down the PS and Share button at the same time, the PS4 controller will then start to blink rapidly indicating that it is in pairing mode
PS4BT PS4(&Btd, PAIR);

// After that you can simply create the instance like so and then press the PS button on the device
//PS4BT PS4(&Btd);
```

The pairing works. But after have paired it once, I need to to change back the code and upload it again. The library is part of a bigger project, which is controlled over a custom Android app. 

My goal is to just send a command over the app to the board. Then a function is called, which takes care of the pairing process. Is there already some function to just trigger the pairing process for the PS4-BT?

Lauszus, 2017-01-31:
You can simply call the following function when you want to pair with the PS4 controller:
```C++
PS4.pair();
```

closed 2017-01-31 by Lauszus

BlueGene00, 2017-01-31:
Thanks! Seems to work without issues on the first try