felis/USB_Host_Shield_2.0 #283: Support/Library for WLAN-Dongle?

BlueGene00, 2017-02-03:
Hello,

would it somehow be possible to use the USB-Host-Shield with a WLAN-Dongle? I mean like on the Raspberry you can also plug in a WLAN-Dongle and set up a network connection.

This could replace a Ethernet-Shield. The best solution would be if the the standard Ethernet-Libary could be used, only with some additions to the WLAN-connection.

Have you thought about this use case or is it technically not possible/too much work?

xxxajk, 2017-02-05:
This isn't going to happen on UHS2 because it simply is not practical.
This will be possible with UHS3 on a sufficient ARM MCU. 
Why this isn't going to happen on AVR is that it really lacks enough RAM to do much of anything useful.
Yes, it can run simpler implementations, yes some things are doable, and you could add RAM to the mega2560 to be able to do this. I have actually done it, and it was picked up by [http://hackaday.com/2014/02/17/expanded-memory-for-the-teensy-2-0/](url) and I was able to run it easily. on the 1284 (1280 with USB).

BlueGene00, 2017-02-11:
What about the Teensy 3.6? It has 256K RAM instead of 8K... Could this work or is it still too small?

Here is my project:
http://puremoco.com/pages/motioncontroller.php

In the inside there is a custom PCB which connects the Teensy 3.6, some stepper drivers and the mini USB-host-shield.

Currently I use the USB-Host-Shield only for the Bluetooth-connection to a Playstation 4 controller.

But it would be a great feature if I just could plug in a WLAN-dongle and do some communication to other devices.

Lauszus, 2017-02-12:
@BlueGene00 this would be a major task, but this is not something I am personal planning on working on.

Have you considered using something like a separate ESP8266 for the WiFi connection?

BlueGene00, 2017-02-12:
Yes, the ESP8266 would be an alternative. Since the USB-Host-Port is already there, I want to check all possibilities to make the most use out of it.