felis/USB_Host_Shield_2.0 #285: USB hub problems using teensy 3.1/3.2

mattomatto, 2017-02-06:
Hi there,

I'm trying to get Teensy 3.1 as a USB host communicating with multiple USB MIDI devices.

I'm using the Mini USB Host shield with this awesome library. 

I also have the high speed SPI library installed and included as recommended: https://github.com/xxxajk/spi4teensy3

I have a single USB MIDI device working perfectly and am receiving data as intended, so I'm confident my connections are sound. 

However, when using the example of using multiple devices via a hub (Code below), I don't get anything from either device.
I have 2 MIDI devices plugged into a hub, and the hub plugged into the Host shield. I have tried powering the hub, and I have tried multiple hubs.

Running the hub example gives me information about the hub,  so it can obviously see the OK, just not anything connected to it. 

Here is the code I'm using:

[CODE] 
/*
 *******************************************************************************
 * USB-MIDI to Legacy Serial MIDI converter
 * Copyright (C) 2012-2016 Yuuichi Akagawa
 *
 * Idea from LPK25 USB-MIDI to Serial MIDI converter
 *   by Collin Cunningham - makezine.com, narbotic.com
 *
 * This is sample program. Do not expect perfect behavior.
 *******************************************************************************
 */

#include <usbh_midi.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#include <SPI.h>
#endif

//#ifdef USBCON
//#define _MIDI_SERIAL_PORT Serial1
//#else
//#define _MIDI_SERIAL_PORT Serial
//#endif
//////////////////////////
// MIDI Pin assign
// 2 : GND
// 4 : +5V(Vcc) with 220ohm
// 5 : TX
//////////////////////////

USB Usb;
USBHub  Hub1(&Usb);
USBH_MIDI  Midi1(&Usb);
USBH_MIDI  Midi2(&Usb);

void MIDI_poll();
void doDelay(unsigned long t1, unsigned long t2, unsigned long delayTime);

void setup()
{
  //_MIDI_SERIAL_PORT.begin(31250);

  if (Usb.Init() == -1) {
    while (1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );
}

void loop()
{
  unsigned long t1;

  Usb.Task();
  t1 = micros();
  if ( Usb.getUsbTaskState() == USB_STATE_RUNNING )
  {
    MIDI_poll();
  }
  //delay(1ms)
  doDelay(t1, micros(), 1000);
}

// Poll USB MIDI Controler and send to serial MIDI
void MIDI_poll()
{
  uint8_t outBuf[ 3 ];
  uint8_t size;

  do {
    if ( (size = Midi1.RecvData(outBuf)) > 0 ) {
      //MIDI Output
      Serial.println(outBuf[0]);
      //_MIDI_SERIAL_PORT.write(outBuf[0], size);
    }
  } while (size > 0);
  do {
    if ( (size = Midi2.RecvData(outBuf)) > 0 ) {
      //MIDI Output
      Serial.println(outBuf[0]);
      //_MIDI_SERIAL_PORT.write(outBuf, size);
    }
  } while (size > 0);
}

// Delay time (max 16383 us)
void doDelay(unsigned long t1, unsigned long t2, unsigned long delayTime)
{
  unsigned long t3;

  if ( t1 > t2 ) {
    t3 = (0xFFFFFFFF - t1 + t2);
  } else {
    t3 = t2 - t1;
  }

  if ( t3 < delayTime ) {
    delayMicroseconds(delayTime - t3);
  }
}

[/CODE] 

Any help would be greatly appreciated.

Thanks, 

Matt

Lauszus, 2017-02-06:
@YuuichiAkagawa can you take a look at this?

xxxajk, 2017-02-06:
Soon as I get a free day, I really need to port MIDI to UHS30.

YuuichiAkagawa, 2017-02-06:
Sorry, I don't have teensy.
Is there enough power?

mattomatto, 2017-02-07:
I'm sure there's enough power - The USB host socket power comes form the Teensy 'USB V' Pin (So it gets it's power straight form the computer's USB port), plus I've tried a couple of different powered hubs.

Thanks for getting a Teensy to try!  Can't wait to get this up and running!

YuuichiAkagawa, 2017-02-10:
@mattomatto

I bought Teensy 3.2. And I use with USB Host Mini.
I connected with the following configuration.
https://www.circuitsathome.com/mcu/teensy-3-0-now-supported-by-the-usb-host-library/
(VBUS jumper is cut)

It seems work fine.
https://pbs.twimg.com/media/C4UGyRbUcAAXqRQ.jpg
AKAI MPK mini + QuNeo

mattomatto, 2017-02-10:
@YuuichiAkagawa 

Hmmm, well that's good news I suppose! Since it all works fine for me without a hub, I must presume my wiring is fine and it must either be a hub or code issue.  I have tried 4 different powered hubs (Brands: Amazon, D-link, Tesco, & unbranded).  Did you use my code from above and see data coming in from both devices?

Many thanks for your time.

YuuichiAkagawa, 2017-02-10:
@mattomatto
> Did you use my code from above and see data coming in from both devices?
Yes. I use your code.

mattomatto, 2017-02-10:
@YuuichiAkagawa 

Ok, I guess I must be having bad luck with hubs!  I will try another brand and see how I get on, unless you can think of any other reason why it would work without a hub but not through.
Thank you very much for testing and getting back to me so quickly :)

YuuichiAkagawa, 2017-02-11:
Is there anyone who knows how to test HUB?

Lauszus, 2017-02-11:
@YuuichiAkagawa try this example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/hub_demo/hub_demo.ino.

YuuichiAkagawa, 2017-02-11:
@Lauszus
I can't retrieve each device. Only hub info. Why?
I used with Arduino Uno.

YuuichiAkagawa, 2017-02-12:
@Lauszus
I found a hub_demo issue.
https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/hub_demo/hub_demo.ino#L99
It is must be
`if ((long)(millis() - next_time) >= 0L) {`
This problem since from https://github.com/felis/USB_Host_Shield_2.0/commit/d56ed57495a9c8d01c4cd94653c5ea3d28ba60c6 , https://github.com/felis/USB_Host_Shield_2.0/commit/f6244bbe59c46a176251125f07bca1b92daf4d6a

@mattomatto
Can you run hub_demo?  Is your MIDI device listed?

mattomatto, 2017-02-16:
@YuuichiAkagawa 
Sorry for the late reply.  When I run hub_demo, the hub information appears, but the MIDI device connected to the hub doesn't. The device is powered ok.

Thanks

YuuichiAkagawa, 2017-02-16:
@mattomatto 
Did you use "rollover" branch?
https://github.com/felis/USB_Host_Shield_2.0/tree/rollover

mattomatto, 2017-02-16:
@YuuichiAkagawa 
Ah, no - I was describing the previous behaviour.  I'll test this when I've got my teensy with me and get back to you when I can.  Thanks :)

mattomatto, 2017-02-27:
@YuuichiAkagawa @Lauszus 

I have just given this a proper test - with some success and some problems, perhaps you can shed some light:

I'm using the hub_demo on the "rollover" branch.
I have a powered 7 port D-link hub plugged into the UHS.

With 5 devices plugged into the hub, the demo reports 2 hubs and 0 devices.

If I edit the hub_demo sketch to only include 2 hub objects, it picks up all 5 devices, and still sees the same 2 hubs (working correctly).

If I edit the hub_demo sketch to include 3 hub objects, it doesn't detect any devices, but still sees the same 2 hubs. 

The sketch takes about 10 seconds to find the devices, not sure if that's normal.

Thanks

YuuichiAkagawa, 2017-02-27:
@mattomatto 
May be you need multiple USBHub instances are required for your HUB.
Like this.
```
USB Usb;
USBHub  Hub1(&Usb);
USBHub  Hub2(&Usb);
USBH_MIDI  Midi1(&Usb);
USBH_MIDI  Midi2(&Usb);
```
My 7 ports HUB with 5 devices are need three instances.
(Japanese) http://www2.elecom.co.jp/game/cable/usb-hub/u2h-gm002/
But my 4 ports HUB with 4 devices are need single instance.
(Japanese) https://www.sanwa.co.jp/product/syohin.asp?code=USB-HMU403BK
I am always testing with the 4 ports HUB.

Unfortunately I do not know the detailed behavior of the USBHub code.
@Lauszus Does it correct?

mattomatto, 2017-02-27:
@YuuichiAkagawa 

This is what I was experimenting with in my above post.

Doing the following result in no devices being found:

`USB     Usb;
USBHub  Hub1(&Usb);
USBHub  Hub2(&Usb);
USBHub  Hub3(&Usb);
USBHub  Hub4(&Usb);`

Whereas as the following appears to work fine:

`USB     Usb;
USBHub  Hub1(&Usb);
USBHub  Hub2(&Usb);`

YuuichiAkagawa, 2017-03-02:
The number of instances needs to be adjusted according to the USB hub to be used.
It is case by case.

Most 4 port HUBs should work with 1 instance.
Some 7 port USB hubs are cascaded internally with two 4 port hubs. That means more than two instances are needed.

BTW, where is UHS2 USBHub documents?

Lauszus, 2017-03-30:
@YuuichiAkagawa there is not that much documentation about the USB hub code. There is only the source code: https://github.com/felis/USB_Host_Shield_2.0/blob/master/usbhub.h and https://github.com/felis/USB_Host_Shield_2.0/blob/master/usbhub.cpp.

And the documentation generated by Doxygen: http://felis.github.io/USB_Host_Shield_2.0/class_u_s_b_hub.html.

YuuichiAkagawa, 2017-03-30:
That means we have to do trial and error to determine how many hub instances it needs.

@mattomatto can we close the issue?

mattomatto, 2017-04-04:
Yes, thank you

closed 2017-04-04 by mattomatto