felis/USB_Host_Shield_2.0 #286: USB Host Shield not working with MFRC522

cyberjuputer, 2017-02-10:
As you know, both communicates through SPI and I can't get them to work. I have initialed both SS pin of the devices HIGH and set it to LOW when I want to use the device independently. I have a USB barcode scanner connected through the USB host shield.

What could be the problem and how can I fix this?

cyberjuputer, 2017-02-10:
As for reference, I include my code
`
      writeBlock(block, blockcontent);
      readBlock(block, readbackblock);
      Serial.print("read block: ");
      for (int j=0 ; j<16 ; j++)
      {
        Serial.write (readbackblock[j]);
      }
      Serial.println("");
`
One thing to note here is that the USB barcode scanner stop functioning right after putting these lines
` writeBlock(block, blockcontent);
      readBlock(block, readbackblock);
      Serial.print("read block: ");
      for (int j=0 ; j<16 ; j++)
      {
        Serial.write (readbackblock[j]);
      }
      Serial.println("");
`
I hope there will be a solution.

Thanks in advance.

Lauszus, 2017-02-10:
@cyberjuputer what library do you use for the MFRC522? Please send the link. My guess would be that it do not use SPI transactions: https://www.arduino.cc/en/Tutorial/SPITransaction.

cyberjuputer, 2017-02-10:
This is the library I use

https://github.com/miguelbalboa/rfid

I did not modify anything on the library.

Lauszus, 2017-02-10:
@cyberjuputer I found your bug. You should control the SS pins manually. This is handled by both libraries. Please remove your defines:
```C++
#define SS_USB 10
#define SS_RFID 3
```

cyberjuputer, 2017-02-10:
I wouldn't worry about the USB. But the define is needed by MFRC522 to work.

Lauszus, 2017-02-10:
Just replace the instance with:
```C++
MFRC522 mfrc522(3, 4);
```
And then please remove the defines, so I am sure that you do not manipulate the pins manually.

cyberjuputer, 2017-02-11:
Do I still need to make my SS LOW when I want to use one device only in my main sketch? And turn it off by making the SS HIGH again.

cyberjuputer, 2017-02-11:
Hi @Lauszus 

I can see that I am getting a bit of progress. After removing the defines, my barcode scanner indicates a successful reading but my serial is outputting "Start" everytime I scan an item.

cyberjuputer, 2017-02-11:
UPDATE:

Tried giving a shot by connecting an LCD. It displays the output perfectly. But why is there nothing on the serial monitor?

Lauszus, 2017-02-11:
@cyberjuputer no you should not handle the SS pins yourselves. This is handled by both libraries.

You need to enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging.

cyberjuputer, 2017-02-11:
@Lauszus Okay I am going to use the LCD to display my output as thats my ultimate intention. Now the barcode is working but after I tried to use the MFRC522 by scanning the card, there is nothing on the serial monitor. And when I tried to scan a barcode, the barcode stop functioning. 

At the moment, I am going to enable serial debugging.

cyberjuputer, 2017-02-11:
@Lauszus After enabling serial debugging, my serial is outputting "Start" almost every second. So, I set it to 0 back. Both devices work but only if I use LCD to show the output. It is kinda solved but idk it looks like something is missing. 

I will do more experiment and get back to you.

cyberjuputer, 2017-02-11:
@Lauszus I think my problem is solved although enabling serial debugging gives me more headache. However, putting an LCD since to do what I want exactly(but still nothing on serial monitor but I dont need it anyway). Thanks to your suggestion to remove the defines.

Now, my code is working as expected.

Your help was indeed very big to me.

I would love to put your name in my report as one of the contributor. Would you mind about that?

Again, thanks.

Lauszus, 2017-02-12:
@cyberjuputer okay I'll close this issue now then. Yes no problem. My full name is "Kristian Sloth Lauszus" :)

closed 2017-02-12 by Lauszus

Amir-ESP, 2018-05-03:
@cyberjupiter can i contact you directly by E-mail ?

cyberjupiter, 2018-05-04:
@Amir-ESP I prefer you discuss anything related here.

Amir-ESP, 2018-05-04:
@cyberjupiter Okay sure, sorry i'm kinda new here
I'm doing the same project as you, but i'm doing at two separate parts.
The first part is reading the barcode and post it on the LCD, the second part is to take that code and pass it on to the RC522 to write it on the tag, my probleme is how to make connection between the 2 parts, and yesterday i saw that you have done the same project but you made a lot a changing as the one you poste it.
those are my 2 codes, i wonder if you can help with it or if you can share your final code.

the first code:

`#include <usbhid.h>
#include <usbhub.h>
#include <hiduniversal.h>
#include <hidboot.h>
#include <SPI.h>
#include <LiquidCrystal.h>
#define DISPLAY_WIDTH 16

LiquidCrystal lcd(7, 6, 5, 4, 3, 2);

class MyParser : public HIDReportParser {
  public:
    MyParser();
    void Parse(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf);
  protected:
    uint8_t KeyToAscii(bool upper, uint8_t mod, uint8_t key);
    virtual void OnKeyScanned(bool upper, uint8_t mod, uint8_t key);
    virtual void OnScanFinished();
};

MyParser::MyParser() {}

void MyParser::Parse(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf) {
 
  if (buf[2] == 1 || buf[2] == 0) return;

  for (uint8_t i = 7; i >= 2; i--) {
   
    if (buf[i] == 0) continue;

   
    if (buf[i] == UHS_HID_BOOT_KEY_ENTER) {
      OnScanFinished();
    }

   
    else {
      
      OnKeyScanned(i > 2, buf, buf[i]);
    }

    return;
  }
}

uint8_t MyParser::KeyToAscii(bool upper, uint8_t mod, uint8_t key) {
  
  if (VALUE_WITHIN(key, 0x04, 0x1d)) {
    if (upper) return (key - 4 + 'A');
    else return (key - 4 + 'a');
  }

  // Numbers
  else if (VALUE_WITHIN(key, 0x1e, 0x27)) {
    return ((key == UHS_HID_BOOT_KEY_ZERO) ? '0' : key - 0x1e + '1');
  }

  return 0;
}

void MyParser::OnKeyScanned(bool upper, uint8_t mod, uint8_t key) {
  uint8_t ascii = KeyToAscii(upper, mod, key);

  static uint32_t next_time = 0;      
static uint8_t current_cursor = 0;  

   if( millis() > next_time ) {
      lcd.clear();
      current_cursor = 0;
      delay( 5 );  
      lcd.setCursor( 0,0 );
    }
 
    next_time = millis() + 200;  
 
    if( current_cursor++ == ( DISPLAY_WIDTH + 1 )) {  
      lcd.setCursor( 0,1 );
    }
  Serial.print((char)ascii);
  lcd.print((char)ascii);
}

void MyParser::OnScanFinished() {
  Serial.println(" - Finished");
}

USB          Usb;
USBHub       Hub(&Usb);
HIDUniversal Hid(&Usb);
MyParser     Parser;

void setup() {
  Serial.begin( 115200 );
  Serial.println("Start");

  if (Usb.Init() == -1) {
    Serial.println("OSC did not start.");
  }

  delay( 200 );

  Hid.SetReportParser(0, &Parser);
}

void loop() {
  Usb.Task();
}`

the second code : 

`#include <SPI.h>
#include <MFRC522.h>

#define SS_PIN 10  
#define RST_PIN 5  
MFRC522 mfrc522(SS_PIN, RST_PIN);        
MFRC522::MIFARE_Key key;

void setup() {
        Serial.begin(9600);
        SPI.begin();       
        mfrc522.PCD_Init();
        Serial.println("Scan a MIFARE Classic card");
        
        for (byte i = 0; i < 6; i++) {
                key.keyByte[i] = 0xFF;
        }

}

int block=2;

byte blockcontent[16] = {"makecourse_____"};

byte readbackblock[18];

void loop()
{

	if ( ! mfrc522.PICC_IsNewCardPresent()) {
		return;
	}

	
	if ( ! mfrc522.PICC_ReadCardSerial()) {
		return;//
	}

       
         
         Serial.println("card selected");
         

         
         writeBlock(block, blockcontent);
       
         
         readBlock(block, readbackblock);
         Serial.print("read block: ");
         for (int j=0 ; j<16 ; j++)
         {
           Serial.write (readbackblock[j]);
         }
         Serial.println("");
         
 
}

`

cyberjupiter, 2018-05-06:
@Amir-ESP 
You can't take the barcode and directly put it into the tag. You need to store the scanned barcode in an array first.

You could do something like this:
1) Put the barcode that has been scanned into an array.
2) Pass that array to the RC522 writeBlock function.

Amir-ESP, 2018-05-08:
@cyberjupiter  thx a lot Bro

cyberjupiter, 2018-05-08:
I think you should post your question on the official Arduino Forum(include your effort). This is generally programming issue and not related to USB Host Shield.

Amir-ESP, 2018-05-09:
@cyberjupiter yeah i know it is not telated to USB Host shield, i just post it here cause my project is a bit like yours.

thepitho, 2022-10-23:
Hello, I'm having the same problems with the same devices, I don't understand the sentence should you check the pins manually, can you help me?

thepitho, 2022-10-23:
@cyberjupiter @Lauszus  Please Help me