felis/USB_Host_Shield_2.0 #292: [Question] Flip Reset and GPX pin numbers

stevemk14ebr, 2017-03-08:
I've made a custom arduino board with the atmega 328p that interfaces to the max chip. In my design i accidentally flipped the GPX and RESET pins. In the USB_Host_Shield_Library_1.0 I just flipped the #defines for GPX and RESET pin port numbers my board worked as it should. In this USB_Host_Shield_Library_2.0 library however i cannot find the define statements relating to GPX and RESET. Can you point out which parts of the code i need to change so that GPX -> RESET and RESET -> GPX.

Ex:
Default pins:
GPX = 8
RESET = 7

Desired change:
RESET = 8
GPX = 7

xxxajk, 2017-03-08:
https://github.com/felis/USB_Host_Shield_2.0/blob/0f0da7c5b4131a241aa1b07c8ee6d972677b5d83/UsbCore.h#L43

Few other tid-bits...
There is no 'reset', reset is done via the hardware reset line and via soiftware.
GPX alternately is used for power control for Vbus.

stevemk14ebr, 2017-03-08:
Those are slave select and interrupt? How does that deal with GPX and reset? How is reset done via hardware and software that makes no sense to me?

Lauszus, 2017-03-08:
The GPX pin is not used my the library at all. Simply connect the reset pin to the reset pin of the ATmega328P.

stevemk14ebr, 2017-03-08:
Oh that's neat i didnt realize the pins were tied. Thank you!

closed 2017-03-08 by stevemk14ebr