felis/USB_Host_Shield_2.0 #293: Issue pairing the ps4 controller.

sawhim, 2017-04-10:
So I used the PS4 Bluetooth library to pair the dongle with my new ps4 controller. I ordered my dongle from Ali express that has already been under some tested dongle. So here is the problem that I'm having... Once I load the code to arduino uno, my dongle goes in pairing mode. When I start the pairing mode on my ps4 it fails to conncet. The only result I see is that it says Bluetooth library started and nothing else and my ps4 pairing light goes off and once I power it again i.e PS bottom, it does blink but in a slower rate and all I see in my serial monitor is Bluetooth library started.  Do you know what could be the possible problem?

Lauszus, 2017-04-10:
Please enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging.

closed 2017-04-10 by Lauszus

sawhim, 2017-04-10:
> Please enable serial debugging: https://github.com/felis/USB_
> Host_Shield_2.0#enable-debugging.

#include <PS4BT.h>
#include <usbhub.h>
#define ENABLE_UHS_DEBUGGING 1
// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#include <SPI.h>
#endif........

IS THAT WHAT YOU MEAN? if so i'm still getting the same result. Thanks for
helping out :)