felis/USB_Host_Shield_2.0 #295: Arduino Leonardo never completes Acm.isReady()

arthurv, 2017-04-15:
Setup: Arduino UNO + USB Host shield -> Arduino Leonardo (USB port)

The Arduino Leonardo is running the Caterina bootloader (with modified VID/PID), which shows as a generic CDC ACM serial port on PCs. I ran acm_terminal with enabled debug:

Start
ACM Init
Addr:01
NC:01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
Endpoint descriptor:
Length:		07
Type:		05
Address:	83
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00

But ACM.isReady() in the acm_terminal example never becomes true, so the serial connection is never made. Any suggestions?

ghostcaesar, 2017-10-02:
Hi, I came across a similar issue, the problem seems to be the cdcacm connection insisting there are at least 4 USB endpoints, while there are not enough here.

Commenting line 159 & 160 of the cmcacm.cpp file in the library seemed to fix the issue for me.

//        if(bNumEP < 4)
//                return USB_DEV_CONFIG_ERROR_DEVICE_NOT_SUPPORTED;

Im not sure if the endpoint requirement is necessary or correct, can devs look into this please?

gdsports, 2019-03-11:
I verified this with a Leonardo and Pro Micro (both 32u4) and real Uno. I changed the required number of endpoints to 3. Works fine with 3. It also works on an Adafruit Metro M4 running CircuitPython which is a composite device with CDC ACM and MSC.

 if(bNumEP < 3)

gdsports, 2019-03-28:
I found changing the required number endpoints allows the claim to succeed but 4 are required. I changed the interface (class, subclass, protocol) matching to exclude the protocol. This catches the interrupt endpoint and allows data transfer to occur with Leonardo and various other boards such as Zero, MKR, and Uno.

```
diff --git a/cdcacm.cpp b/cdcacm.cpp
index 4733890a..67dd2953 100644
--- a/cdcacm.cpp
+++ b/cdcacm.cpp
@@ -133,11 +133,9 @@ uint8_t ACM::Init(uint8_t parent, uint8_t port, bool lowspeed) {
 
         for(uint8_t i = 0; i < num_of_conf; i++) {
                 ConfigDescParser< USB_CLASS_COM_AND_CDC_CTRL,
-                        CDC_SUBCLASS_ACM,
-                        CDC_PROTOCOL_ITU_T_V_250,
+                        CDC_SUBCLASS_ACM, 0,
                         CP_MASK_COMPARE_CLASS |
-                        CP_MASK_COMPARE_SUBCLASS |
-                        CP_MASK_COMPARE_PROTOCOL > CdcControlParser(this);
+                        CP_MASK_COMPARE_SUBCLASS > CdcControlParser(this);
 
                 ConfigDescParser<USB_CLASS_CDC_DATA, 0, 0,
                         CP_MASK_COMPARE_CLASS> CdcDataParser(this);
@@ -156,6 +154,7 @@ uint8_t ACM::Init(uint8_t parent, uint8_t port, bool lowspeed) {
                         break;
         } // for
 
+         USBTRACE2("bNumEP:", bNumEP);
         if(bNumEP < 4)
                 return USB_DEV_CONFIG_ERROR_DEVICE_NOT_SUPPORTED;
```

xxxajk, 2019-03-28:
This is why UHS2 is kind of broken by design, and why I abandoned it.

BansalSac, 2020-12-09:
I am trying to communicate with CDC USB device with Arduino UNO using USB Host Shield. I am not able to get and output if i send predefined ASCII command as per manual. I am using acm_terminal.ico sketch.
Few basic info:
1. I am using USB Host Shield Library 2.0
2. Host Shield is working perfectly when another Arduino is connected as slave
3. To find more details of CDC USB device, after connecting to Host shield i uploaded USB_desc program and attached is output of that
[USB_desc_output.txt](https://github.com/felis/USB_Host_Shield_2.0/files/5667208/USB_desc_output.txt)

3. Looking at earlier suggestions in this post, in cdcacm.cpp i tried changing the required number of endpoints to 3, But no success.
4. Later i tried commenting below lines  :
// if(bNumEP < 4)
// return USB_DEV_CONFIG_ERROR_DEVICE_NOT_SUPPORTED;

But still no success

I see that Acm is not getting ready and because of that communication is not happening.

My Gut feeling is based on attached output of USB_desc some tweaks are required in library.

Looking for any support on this issue