felis/USB_Host_Shield_2.0 #297: Help with Teensy 3.5

mjs513, 2017-05-27:
Hi all.

I am using a homebrew shield for a teensy 3.5 breakout board that I made and am receiving the following error when using board_qc:
```
Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K
Test failed.  Value written: 79 read: 40
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test
```

Because it is a homebrew, I tested the shield separately on a Arduino 101 and it worked fine so I am pretty sure the shield is fine.  I tested the SPI interface on the breakout board using a Memory Card reader and the standard example for a teensy and had no problem with the tests.  I also tried using the official shield but it didn't even read the revision register.  I also checked the voltages to the board and they are ok.  I also see the light on the BT dongle.

Any suggestions would be appreciated, I am at a loss now.

Mike

mjs513, 2017-05-27:
Just by way of an update I tried it with the version of UHS3.0 on my machine (probably out of date by now) but it worked fine.  Here is the output of the board_qc  test:
```
Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K 
................
SPI long test passed
GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
Test failed. Wrote 0x00, read 0xff

Press any key to continue...
```

xxxajk, 2017-05-29:
Totally normal, send a character to finish the test.

mjs513, 2017-05-29:
@xxxajk Even though the revision level is invalid (thought it only went up to 3) ?

Mike

xxxajk, 2017-05-29:
Rev3 is fine. Looks correct...

mjs513, 2017-06-01:
Still can't get it to work with the 1.8.2 and teensyduino 1.36.  I did come across this post on the teensy forum about spi.  Not sure if it may be affecting SPI transfer or not but thought I should pass it along:
[https://forum.pjrc.com/threads/44413-SPI-transfer(bytes-count)-has-byte-order-issue-w-Teensy-3-2-and-Teensyduino-1-36](url)

which is mentioned it the new version of teensyduino coming out: [https://forum.pjrc.com/threads/44509-Arduino-1-8-3?p=144663&viewfull=1#post144663](url)

mjs513, 2017-06-02:
Me Again.  Decided to do a little more debugging via modifying the board_qc.ino file to use the read/wrt routines directly from the INO which was copied from the UHS3.0.  Lo and behold it worked. Here is the results:

```
Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K................ SPI long test passed
GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
Test failed. Value written: 00 Value read: FF 
Press any key to continue...
GPIO test passed.
PLL test. 100 chip resets will be performed
Resetting oscillator

Reset number 0 Time to stabilize - 1423 cycles
Reset number 1 Time to stabilize - 1421 cycles
Reset number 2 Time to stabilize - 1422 cycles
Reset number 3 Time to stabilize - 1424 cycles
Reset number 4 Time to stabilize - 1422 cycles
Reset number 5 Time to stabilize - 1422 cycles
Reset number 6 Time to stabilize - 1421 cycles
Reset number 7 Time to stabilize - 1423 cycles
Reset number 8 Time to stabilize - 1424 cycles
Reset number 9 Time to stabilize - 1422 cycles
Reset number 10 Time to stabilize - 1422 cycles
Reset number 11 Time to stabilize - 1422 cycles
Reset number 12 Time to stabilize - 1423 cycles
Reset number 13 Time to stabilize - 1423 cycles
Reset number 14 Time to stabilize - 1422 cycles
Reset number 15 Time to stabilize - 1422 cycles
Reset number 16 Time to stabilize - 1423 cycles
Reset number 17 Time to stabilize - 1423 cycles
Reset number 18 Time to stabilize - 1422 cycles
Reset number 19 Time to stabilize - 1422 cycles
Reset number 20 Time to stabilize - 1422 cycles
Reset number 21 Time to stabilize - 1423 cycles
Reset number 22 Time to stabilize - 1422 cycles
Reset number 23 Time to stabilize - 1422 cycles
Reset number 24 Time to stabilize - 1422 cycles
Reset number 25 Time to stabilize - 1422 cycles
Reset number 26 Time to stabilize - 1423 cycles
Reset number 27 Time to stabilize - 1422 cycles
Reset number 28 Time to stabilize - 1422 cycles
Reset number 29 Time to stabilize - 1422 cycles
Reset number 30 Time to stabilize - 1423 cycles
Reset number 31 Time to stabilize - 1422 cycles
Reset number 32 Time to stabilize - 1422 cycles
Reset number 33 Time to stabilize - 1421 cycles
Reset number 34 Time to stabilize - 1422 cycles
Reset number 35 Time to stabilize - 1423 cycles
Reset number 36 Time to stabilize - 1422 cycles
Reset number 37 Time to stabilize - 1422 cycles
Reset number 38 Time to stabilize - 1422 cycles
Reset number 39 Time to stabilize - 1423 cycles
Reset number 40 Time to stabilize - 1422 cycles
Reset number 41 Time to stabilize - 1422 cycles
Reset number 42 Time to stabilize - 1422 cycles
Reset number 43 Time to stabilize - 1422 cycles
Reset number 44 Time to stabilize - 1423 cycles
Reset number 45 Time to stabilize - 1423 cycles
Reset number 46 Time to stabilize - 1422 cycles
Reset number 47 Time to stabilize - 1422 cycles
Reset number 48 Time to stabilize - 1423 cycles
Reset number 49 Time to stabilize - 1422 cycles
Reset number 50 Time to stabilize - 1422 cycles
Reset number 51 Time to stabilize - 1422 cycles
Reset number 52 Time to stabilize - 1422 cycles
Reset number 53 Time to stabilize - 1423 cycles
Reset number 54 Time to stabilize - 1422 cycles
Reset number 55 Time to stabilize - 1422 cycles
[... 404 lines cut ...]
        gpin &= 0xf0; //clean lower nibble
        gpin |= (regRd(rIOPINS1) >> 4); //shift low bits and OR with upper from previous operation.
        return ( gpin);
}

/* reset MAX3421E. Returns number of microseconds it took for PLL to stabilize after reset
  or zero if PLL haven't stabilized in 65535 cycles */
uint16_t reset(void) {
        uint16_t i = 0;
        regWr(rUSBCTL, bmCHIPRES);
        regWr(rUSBCTL, 0x00);
        int32_t now;
        uint32_t expires = micros() + 65535;
        while((int32_t)(micros() - expires) < 0L) {
                if((regRd(rUSBIRQ) & bmOSCOKIRQ)) {
                        break;
                }
        }
        now = (int32_t)(micros() - expires);
        if(now < 0L) {
                i = 65535 + now; // Note this subtracts, as now is negative
        }
        return (i);
}
```

xxxajk, 2017-06-02:
I think I know why...don't use spi4teensy3 and see if that solves the problem.
This can be changed by setting this macro to zero:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/settings.h#L71
If that solves the problem, then it can be easily fixed in the selection code.

mjs513, 2017-06-02:
Hmm.  That looked like it was already set to zero by: 
```
// Disabled on the Teensy LC, Teensy 3.5, and Teensy 3.6 as it is incompatible for now
#if defined(__MKL26Z64__) || defined(__MK64FX512__) || defined(__MK66FX1M0__)
#undef USE_SPI4TEENSY3
#define USE_SPI4TEENSY3 0
#endif
```
I gave it a try anyway this way:
```
#ifndef USE_SPI4TEENSY3
#define USE_SPI4TEENSY3 0
#endif
```
Any still did not work, here is the output:
Circuits At Home 2011
```
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K
Test failed.  Value written: 62 read: 40
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test
```

xxxajk, 2017-06-02:
Ok, just making certain.

mjs513, 2017-06-02:
Ok.  I updated the usbhost.h file with the read/wrt stuff from above and it worked.  In other words it read the revision, passed the SPI long writes, etc.  Everything was good.  I tried running the PS4USB sketch and just hung even with the uhs_debugging enabled.   So maybe now an interrupt issue.  Am wondering is anyone else is having this problem now? 

So the million dollar question is how to get this fixed?

mjs513, 2017-06-03:
OK take my last post back.  Think I was getting crossed eyed.  Ran the usb sketch instead of the bt sketch.  THey both work now problem.

mjs513, 2017-06-03:
Here are the files I modified - think someone needs to take a look and see what needs to be changed for possible incorporation.

[USB_Host_Shield_2.0-1.3.0.zip](https://github.com/felis/USB_Host_Shield_2.0/files/1049239/USB_Host_Shield_2.0-1.3.0.zip)

mjs513, 2017-06-03:
OK.  Paul released teensyduino 1.37-beta1 today and it seems to have cleared up the problem.  No need to apply any changes to make it work now.  All is good.

closed 2017-06-03 by mjs513