felis/USB_Host_Shield_2.0 #299: XBoxOne 'Xbox' button clicks not recognized

guruthree, 2017-06-17:
Testing out the fix for https://github.com/felis/USB_Host_Shield_2.0/issues/216, I noticed that in the demo program for the XBOXONE library, that button clicks of the 'XBox' button were not recognized. I've tracked this back to https://github.com/felis/USB_Host_Shield_2.0/commit/fcab83dcb398ec904db22009d21fa2c1fa133ba1 [lines 266 to 270](https://github.com/felis/USB_Host_Shield_2.0/commit/fcab83dcb398ec904db22009d21fa2c1fa133ba1#diff-170d81b53a3e1d72de19b2bd28603c58L267). The click state update there is needed as when there's a 0x07 packet, the click state update later [line 293](https://github.com/felis/USB_Host_Shield_2.0/blob/XBOXONE_fix/XBOXONE.cpp#L293) of [XBOXONE.cpp](https://github.com/felis/USB_Host_Shield_2.0/blob/XBOXONE_fix/XBOXONE.cpp) is never reached due to [line 270](https://github.com/felis/USB_Host_Shield_2.0/blob/XBOXONE_fix/XBOXONE.cpp#L270). Sorry for the delay in testing...

Lauszus, 2017-06-22:
@guruthree did you have time to test #299 yet?

guruthree, 2017-06-22:
Sorry, I'll get to it this weekend!

Lauszus, 2017-06-23:
@guruthree no problem :)

guruthree, 2017-06-25:
Works just fine. 👍

closed 2017-06-25 by guruthree

Lauszus, 2017-06-25:
@guruthree thanks the feedback!

commit 39150a15 (2017-06-18): Also update the ButtonClickState variable when the Xbox button is pressed

Fixes #299