felis/USB_Host_Shield_2.0 #301: Interfacing Arduino to 3G USB modem ( Huawei E3131 )

imedaouidene, 2017-07-07:
I need to interface 3g modem USB dongle ( Huawei E3131)   with my Arduino UNO , To transfer data with high bit rate .
any help ?

Blaud, 2018-01-27:
For me the best way to get internet on arduino was using nodemcu + [https://www.aliexpress.com/item/Tolkien-LW3G-UIFI-WiFi-Modem-Mini-Portable-Wi-fi-Mobile-Device-3G-Wireless-Router-with-TF/32811968977.html](uri) - usb 3g wifi dongle

Blaud, 2018-07-29:
@pawisoon It wasn't straight usb connection. I've used a dongle with integrated wifi module, then just power it up with 5v and connect with nodemcu(esp8266) via wifi. It is really simplest and cheapest way to get 3g on your arduino. (this dongle cost about 12$)

MohammedZ666, 2021-02-07:
@Blaud can u explain how to power up the USB Wifi modem, does it auto-connect to the internet?

Blaud, 2021-02-10:
@MohammedZ666 yes, it auto connects on power on. I was able to power it by connecting parallel to nodemcu 5v power supply. 
_modem like [this ](https://aliexpress.ru/item/4000595683138.html)_

MohammedZ666, 2021-02-10:
@Blaud thanks for replying!!