felis/USB_Host_Shield_2.0 #302: PS4BT not working

vedangasDARKNINJA, 2017-07-09:
I tried the other arduino file, PS4USB.ino, it's working as it's expected. But when I tried to run the PS4BT.ino, i get the following output:

PS4 Bluetooth Library Started
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:0B:0D:31:75:AB
Please enable discovery of your device
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Connecting to HID device
Connection Failed: 21
Please enable discovery of your device
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Connecting to HID device
Connection Failed: 21
Please enable discovery of your device
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Connecting to HID device
Connected to HID device
Received Key Request
Bluetooth pin is set too: 0000

after this nothing happens, even my PS4 would  stop blinking after some time. Help would be greatly appreciated.......thanks in advance !

This was one problem, 
this is just to confirm that there's no need to change the dongle, 
I did run the PS3BT.ino on the same dongle and it's working great ! but not the same response on PS4

Lauszus, 2017-07-23:
I just moved, so it will take me some time before I get all my stuff. I will look into this eventually.

vedangasDARKNINJA, 2017-07-23:
Hey actually the above problem is solved but this new problem regarding the pairing between PS4 and the USB Host Shield....... If I flash the board with the PS4BT.ino, the PS4 will connect at first.......but if I disconnect the PS4 using "PS button" then it would disconnect but if I try again to connect it..........it won't.........and it's so annoying.........

xxxajk, 2017-07-23:
Clearly something is missing in the initialization.
Unfortunately I don't have time to look into it, nor do I own that
controller, else I'd try to help out.
Swamped with work here.
Hopefully we can resolve this issue in UHS2 before the driver gets ported
to UHS3.

vedangasDARKNINJA, 2017-07-23:
That problem of pairing is kind of irregular, so I find it very difficult to debug the problem as it's not letting me find the cause of the problem.

xxxajk, 2017-07-23:
Does it pair and work ok if you re-plug the bluetooth?

vedangasDARKNINJA, 2017-07-23:
If I turn off the the board and turn it on then there's no guarantee that the controller will pair again

xxxajk, 2017-07-23:
If you did the same on the PS4 (power off, power on) does it automatically reconnect?

vedangasDARKNINJA, 2017-07-23:
for PS4, that dull white light will show up otherwise it would blink (not the way it blinks while pairing) and after some time it would turn off

xxxajk, 2017-07-23:
I wonder if the PS4 is sending a message before power-off, that's why I'm asking.
PS4 doesn't just power down instantly, IIRC it takes a few moments.
I don't own one, but a friend of mine does.

vedangasDARKNINJA, 2017-07-23:
yeah around 10 sec.s

vedangasDARKNINJA, 2017-07-23:
I need to press the PS button for a long time then only it will kinda 'force shutdown'

xxxajk, 2017-07-23:
That could be why. Not saying it is, because I'm just guessing.
Try unplugging just the bluetooth dongle, then plug it back in, and see if it acts the same way (unreliable). Perhaps there is some sort of magic that could be done.

vedangasDARKNINJA, 2017-07-23:
and one more observation was even if it pairs correctly (at least it seems that it is paired correctly as the blue light can be seen in that front LED) serial monitor shows 'unknown id : 80'

vedangasDARKNINJA, 2017-07-23:
or sometimes the ps4 is not able to shutdown in the paired state

xxxajk, 2017-07-23:
That is a huge hint. I wonder if it is not able to understand the initialization if it is already initialized.
The only way I can think of figuring out exactly what happens would be to sniff the bluetooth packets.

vedangasDARKNINJA, 2017-07-23:
like the blue light still can be seen but after pressing PS button.....................nothing happens

xxxajk, 2017-07-23:
Are you able to sniff the ps4 bluetooth from a pc using wireshark in promiscuous mode?

vedangasDARKNINJA, 2017-07-23:
so I am really confused right now,cz I'm not able to figure out who's fault is it, 1. Arduino UNO
2. USB Host Shield 3. Bluetooth Dongle 4. PS4 itself

vedangasDARKNINJA, 2017-07-23:
how to do that ???
("Are you able to sniff the ps4 bluetooth from a pc using wireshark in promiscuous mode?")

xxxajk, 2017-07-23:
It may be a combination of things that happen, which the driver can't cope with, but the PS4 does.

xxxajk, 2017-07-23:
Video that can help you. Not using wireshark, but...
I'm sure you can find one using wireshark.

https://www.youtube.com/watch?v=TPxw0V42p1o

xxxajk, 2017-07-23:
https://www.youtube.com/watch?v=tk_wC434ft4

gylian, 2018-04-07:
I exactly have the same issue
I can pair the PS4 using pairing mode, but after that if I press the PS button, controller turn off, then if I press again, the controller's light blink slowly then stay white.
Debug mode say connected but no dialog with the serial monitor
Please help