felis/USB_Host_Shield_2.0 #303: PS3BT won't connect to Navigation Controller

shmorgan, 2017-07-12:
Trying to connect the Navigation Controller with Host Shield on Mega 2560. Sony Wireless Controller CECHZC2U works great with no issues. I repeated the same process of pairing as the Shock controller with the Navigation Controller (CECH-ZCS1U) but will not connect BlueTooth. Works fine with the USB sketch.

Log:

PS3 Bluetooth Library Started
Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:15
Wait For Incoming Connection Request
>> Connect Navigation Controller via USB
Navigation Controller Connected
Bluetooth Address was set to: 00:1A:7D:DA:71:15
>> Re-Install Bluetooth Dongle
Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:15
Wait For Incoming Connection Request

>>Press the PS button on the Navigation Controller, and any other button at this point and I get nothing. I have now tried 4 different Navigation Controllers from Game Stop.

Help Please......

shmorgan, 2017-07-15:
Will be happy to ship you one of the controllers

xxxajk, 2017-07-15:
I'd love to help, but...
1: This is @Lauszus department.
2: I'm in the middle of a ton of work.

So, wait for his response.

shmorgan, 2017-07-16:
Will do!

Lauszus, 2017-07-23:
I just moved, so it will take me some time before I get all my stuff. I will look into this eventually.

shmorgan, 2017-07-24:
Thanks for the Reply!! Well issue is resolved. New to the PS3 Navigation Controllers, I did not realize that the LED not coming on when you press the PS button was an issue!! Turns our that the first two I purchased from Gamestop had bad batteries and did not charge correctly. Once I charged the two replacements, they connected without issue.

closed 2017-07-24 by shmorgan