felis/USB_Host_Shield_2.0 #304: RobotDyn MEGA ADK for USB mass Storage not Behaving

shadeyinka, 2017-07-15:
I got a RobotDyn Mega ADK board with the intention of Building a Datalogger where the primary memory will be an SD Card and the Secondary memory will be through a Flash-Drive (memory Stick) which is expected to be connected to the USB Host Port of the Mega ADK once every week to harvest the data  on the SD card.

I have used your USB_Host_Shield_2.0 library with testusbHostFAT. It compiles and i could upload it but appart from that, nothing is happenig with respect to writing or reading from the Flash Drive. Of the SerialEvent(), only the  flicking of  VBUS seems to work.

I have completed the data logging aspect for an SD card using an SD card shield.

For the data harvesting part, I am using the USB Host Shield library Version 2.0 released By Oleg Mazurov. I have successfully compiled and uploaded the testusbHost example file from the  USB_Host_Shield library.

Examples abound for every other use of the USB host but NONE for simple writing of a datafile on a Flash Drive.

I will appreciate a sample code that will
1. allow me to either write data on a Flashdrive
or
2. allow me to move datafiles from an SD Card to a Flash Drive.

Ive been on this for more than 2 straight weeks with no solution in sight.

Thanks in Advance

xxxajk, 2017-07-15:
See here:
https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/testusbhostFAT

xxxajk, 2017-07-15:
> tree/master/examples/testusbhostFAT

Note that is may be better for you to use UHS3, since it uses a simpler API.

shadeyinka, 2017-08-02:
Thank you Sir. Your excellent code works. The only issue is that the code would not compile with SD.h or SdFat.h.

Is there a way out?
Have you experienced this before?

All I want to do is to be able to move data from SD card to flash drive. Any help will be appreciated

Thanks

Yinka

xxxajk, 2017-08-02:
The SD library conflicts, and would probabbly add too much extra code anyway.
Don't worry, though, there is an easy solution, and you don't need to even use the SD library anymore.

1: Place a volume label on the SD card, for example "SDCARD"
2: Attach a USB hub.
3: connect the sdcard reader and the thumb drive.
4: copy the files from "/SDCARD" to "/"

shadeyinka, 2017-08-02:
Thanks so much for the help. Looks so simple but I am sorry, its a little
cryptic to me. For example in the code,
1. How do I attach the USB hub in the code?
2. How do I copy from "/SDCARD" to "/"

There seems to be no example that shows how to interact with an SD card.

So sorry for the trouble and thanks so much for the help so far. You have
revived my hope in completing this project.

Yinka

shadeyinka, 2017-08-02:
My idea Sir
1. To attach USB hub
USB SDCARD;
USBhub *Hubs[1]; //since only one SD card is to be used

2. To copy SD card file to flash drive:
No clue!

Thanks and sorry for the trouble

Yinka

shadeyinka, 2017-08-02:
Sorry my error in the last post

My idea Sir
1. To attach USB hub
USB SDCARD;
USBhub *SDCARD[1]; //since only one SD card is to be used

2. To copy SD card file to flash drive:
No clue!

Thanks and sorry for the trouble

Yinka

xxxajk, 2017-08-03:
Notice that all you need to do is grab ANY classic C book, and just prepend "fs_" to any file-function and it means the same thing as in this library. 
This library is modeled after UNIX-like filesystems, and file APIs.

Example pseudo code. 
This is pretty much a classic case, one of the first things you should have learned in class!!

Assumes file on sdcard is called "file.txt" and the volume label is SDCARD.
Assumes there is no volume label on the target (thumb drive, or other media)
Please see examples for how to get directory entries.
You should change the fixed file names for pointers to strings. It is not done here for clarity.
This should be a function, called once per file.
As in classic C errno is instead fs_err for the global error code.
For any other details please see the sources and examples in the UHS_FS directory.
All of the API is very common. Go grab a C language book.

```
int res; // return value from functions
int in_fd; // input file descriptor number
int out_fd; // output  file descriptor number
uint8_t data;
in_fd = fs_open("/sdcard/file.txt", O_RDONLY);
if(in_fd > 0) {
        // input opened OK
        out_fd = fs_open("/sdcard/file.txt", , O_WRONLY | O_CREAT);
        if(out_fd > 0) {
                // output opened OK, copy file. NOTE: this could be made faster by using larger chunks.
                while(!fs_eof(in_fd)) {
                        res=fs_read(fd_in, &data,1);
                        while(res>0) {
                                res=fs_write(fd_out,&data,1);
                        }
                        if(res < 0) {
                                 // error. You may want to say something here ;-)
                                break;
                        }
                }
                // close output file
                fs_close(out_fd);
                fs_sync(); // IMPORTANT! Sync all caches to all medias!
        } else {
                res = out_fd; // could not open output file.
        } // fall thru
        // close input file
        fs_close(in_fd);
} else {
        res = in_fd; // no input file found.
}
return res;

```

xxxajk, 2017-08-03:
Note I updated the above code, it had an error.

shadeyinka, 2017-08-03:
Dear Sir
Thank you so so much for the detailed explanation. I will start work on
this and hopefully feed you back of my success.

You are truly a rare gem to the community for your patience and
understanding.

Thanks again.

Yinka

xxxajk, 2017-08-03:
not a problem.
There are plenty of examples available everywhere.
Use a search engine, and you'll find good answers.
this example
[https://stackoverflow.com/questions/2180079/how-can-i-copy-a-file-on-unix-using-c](url)
with slight modifications should work as well.

shadeyinka, 2017-08-03:
Thanks so much for the link. I am unto it.

Just one more question Sir.

1. Will this pseudocode work with testusbhostFAT.ino?
I have been able to upload the example and test its functionality with
Flash Drives
(UHS3 looks simpler)

2. The TEST_BULK example from
UHS_host/UHS-BULK_STORAGE/USB_HOST_SHIELD/TEST_BULK.ino

and the UHS_FSNEW_DEMO example from
UHS_FS/USB_HOST_SHIELD/UHS_FSNEW_DEMO.ino

both Compiles and Uploads (on Aduino ADK) but get gets stuck at execution at

" while(KINETIS_Usb.Init(1000) !=0); "

I dont know why? I am using Arduino 1.6.9  on Windows 7

Thanks

Yinka

-- 
Dr Olayinka Babalola
Department of Physics,
University of Ilorin'
+234 (0) 703 395 4960

xxxajk, 2017-08-03:
```
1: UHS3 looks simpler because it hides many details, and takes care of a lot for you. 
   UHS2 is kind of clunky, and not recommended because of how complicated it is.

2:  Tools -> boards ->boards manager
        check the AVR boards version.
        Arduino AVR boards by Arduino 1.6.11 <-- this one works!

```

shadeyinka, 2017-08-16:
Hello Sir,
Thanks for all your help. The final question I want to ask is this.

How do I mount the SD card. It is on digital pin 8 (cs_pin=8) of my Arduino
board.

I have used FatFs.begin(cs_pin)
and so many other variations to no avail.

Thanks for your understanding.

Note:
I still couldn't get UHS3 to work. I am using UHS2. UHS3 compiles but do
nothing on my board.

shadeyinka, 2017-08-16:
Or Sir,
Can you please help with a basic code to use UHS2 or UHS3 to read/write to
the SD card?

Sorry for the trouble

xxxajk, 2017-08-16:
https://github.com/felis/UHS30/tree/master/libraries/UHS_FS/examples/USB_HOST_SHIELD/UHS_FS_NEW_DEMO

Move the conversation to UHS30 for followup questions by starting an issue/request there.

xxxajk, 2018-05-24:
Closing due to inactivity, assumed all issues fixed/explained.

closed 2018-05-24 by xxxajk