felis/USB_Host_Shield_2.0 #305: Having an issue with the xboxone library

mojoe128, 2017-07-22:
I have been getting init failed whenever i try to connect an Xbox one controller.  I was wondering if you had any idea why.  I have tested my host shield with the ps4 and xbox libraries.  They both work.  This is what is is the serial monitor with all of the debugs on.

XBOX USB Library Started
XBOXONE Init
Addr: 01
Xbox One Controller Connected

XboxCommand, Return: 0D
Xbox One Init Failed, error code: 0D

Xbox One Controller Disconnected

XBOXONE Init
getDevDescr 
Xbox One Init Failed, error code: 0D

Xbox One Controller Disconnected

XBOXONE Init
getDevDescr 
Xbox One Init Failed, error code: 0D

Xbox One Controller Disconnected

XBOXONE Init
getDevDescr 
Xbox One Init Failed, error code: 0D

Xbox One Controller Disconnected

**********************************************
Any help would be greatly appreciated 

thank you

Lauszus, 2017-07-23:
@mojoe128 are you running the latest code here form GIthub or have you downloaded it through the Arduino library manager?

mojoe128, 2017-07-23:
Hi!  I downloaded it from here.  I then added a .zip library with sketch/include library/add .zip library.  

Here is the usb_desc for the controller i'm trying to connect.

Thank you for your time

Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	47
Device Protocol:	D0
Max.packet size:	40
Vendor  ID:		045E
Product ID:		02EA
Revision ID:		0301
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		0060
Num.intf:		03
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			03
Max.pkt size:		0040
Polling interval:	04

Endpoint descriptor:
Endpoint address:	82
Attr.:			03
Max.pkt size:		0040
Polling interval:	04

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		00E4
Polling interval:	01

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0040
Polling interval:	01

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Interface descriptor:
Intf.number:		02
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	04
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	84
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)

Lauszus, 2017-10-12:
Please download the newest version of the library (currently 1.3.1). It has a fix for Xbox One controllers: https://github.com/felis/USB_Host_Shield_2.0/pull/298.

mojoe128, 2017-10-14:
Hi!  I have tried the updated library and am still having the same issue.  Would you have any idea why?

Thanks

Lauszus, 2017-10-15:
Can you try to update the firmware on your controller?

mojoe128, 2017-10-15:
I just checked the firmware.  I was already up to date.  Any other ideas?  Thanks for your time!

Carsonsmuts, 2017-11-07:
Same here 

XBOX USB Library Started
XBOXONE Init
Addr: 01
Xbox One Controller Connected

XboxCommand, Return: 0DMade it this far 00

Xbox One Init Failed, error code: 0D

Xbox One Controller Disconnected

XBOXONE Init
getDevDescr 
Xbox One Init Failed, error code: 0D

Xbox One Controller Disconnected

XBOXONE Init
getDevDescr 
Xbox One Init Failed, error code: 0D

Xbox One Controller Disconnected

XBOXONE Init
getDevDescr 
Xbox One Init Failed, error code: 0D

Xbox One Controller Disconnected

Carsonsmuts, 2017-11-07:
[image]

Lauszus, 2017-11-07:
@Carsonsmuts could you try to change this line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXONE.cpp#L150 to:

```bash
epInfo[ XBOX_OUTPUT_PIPE ].epAddr = 0x02; // XBOX one output endpoint
```

And this line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXONE.cpp#L156 to.

```bash
epInfo[ XBOX_INPUT_PIPE ].epAddr = 0x02; // XBOX one input endpoint
```

I.e. this will use endpoint address 2 instead of 1.

Carsonsmuts, 2017-11-08:
Yep, that worked! I search everywhere for new possible protocols. Where did you find this info if I may ask?

Lauszus, 2017-11-09:
Actually you can tell it from the comment by @mojoe128: https://github.com/felis/USB_Host_Shield_2.0/issues/305#issuecomment-317261821. I will write a fix, so it works for all Xbox One controller when I have time.

Lauszus, 2017-11-09:
@mojoe128  and @Carsonsmuts please test: https://github.com/felis/USB_Host_Shield_2.0/pull/325 and report back if it works or not.

PR #325 (2017-12-10): Extract the endpoints from the Xbox One controller
This fixes #305
commit 7078ea9f (2017-12-10): Extract the endpoints from the Xbox One controller instead of hardcoding them

Also cleaned up the code a bit

Fixes #305
closed 2017-12-10 by Lauszus