felis/USB_Host_Shield_2.0 #309: Can't Buil lib in Arduino

TienHuyIoT, 2017-08-14:
I had got error message when build any sketch Example Form New lib USB_Host_Shield_2.0 today .
If i replay old lib USB_Host_Shield_2.0, It's OK.
***** Error Message:
C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2\cdcftdi.cpp: In member function 'virtual uint8_t FTDI::Init(uint8_t, uint8_t, bool)':
C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2\cdcftdi.cpp:223:26: error: qualified-id in declaration before '(' token
 void FTDI::EndpointXtract(uint8_t conf, uint8_t iface, uint8_t alt, uint8_t proto, const USB_ENDPOINT_DESCRIPTOR *pep) {
                          ^
C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2\cdcftdi.cpp:348:1: error: expected '}' at end of input
 }
 ^

xxxajk, 2017-08-14:
Looking into it. Just might be only a mismatched brace.

xxxajk, 2017-08-14:
Please report if it is fixed. :-)

TienHuyIoT, 2017-08-14:
It's OK. Tks so much!

I installed full lib:
 * https://github.com/xxxajk/xmem2 Install as 'xmem', provides memory services.
 * https://github.com/xxxajk/generic_storage provides access to FAT file system.
 * https://github.com/xxxajk/RTClib provides access to DS1307, or fake clock.
But when i Buil Sketch "testusbhostFAT". I got this error message. Help me fix this error please. Tks!

C:\Program Files (x86)\Arduino\libraries\generic_storage/FAT/FatFS/src/ff.c: In function 'f_write':
C:\Program Files (x86)\Arduino\libraries\generic_storage/FAT/FatFS/src/ff.c:2880:1: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
C:\Program Files (x86)\Arduino\libraries\generic_storage/FAT/FatFS/src/ff.c:2880:1: error: this is the insn:
(insn 216 215 217 33 (set (reg:HI 26 r26)
        (reg/f:HI 194)) C:\Program Files (x86)\Arduino\libraries\generic_storage/FAT/FatFS/src/ff.c:2837 83 {*movhi}
     (nil))
C:\Program Files (x86)\Arduino\libraries\generic_storage/FAT/FatFS/src/ff.c:2880: confused by earlier errors, bailing out
lto-wrapper: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
Using library xmem2 in folder: C:\Program Files (x86)\Arduino\libraries\xmem2 (legacy)
Using library USB_Host_Shield_2 at version 1.3.0 in folder: C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
Using library RTClib in folder: C:\Program Files (x86)\Arduino\libraries\RTClib (legacy)
Using library generic_storage in folder: C:\Program Files (x86)\Arduino\libraries\generic_storage (legacy)
exit status 1
Error compiling for board Arduino Mega ADK.

xxxajk, 2017-08-14:
This is an annyoing AVR-gcc bug. A way to avoid it is to downgrade AVR boards
```
tools -> board -> boards manager
```
find
```
software tools Arduino AVR boards by Arduino 
```
downgrade to 
```
version 1.6.11
```

Helps to also complain to the Arduino people, it is a KNOWN issue and nothing has been done to fix the optimizer.

https://github.com/arduino/toolchain-avr/issues
should be the right place to do this

TienHuyIoT, 2017-08-14:
Thank you verry much!
I had build successful.

xxxajk, 2017-08-14:
Excellent! Please follow up with the arduino people about the error.
The more noises made about it, the quicker it will be fixed.

closed 2017-08-14 by xxxajk