felis/USB_Host_Shield_2.0 #312: PS3 Controller paired, bug stops at "Waiting for incoming connection request".

toptensoftware, 2017-09-01:
Hi,

Firstly thanks so much for the work you guys have put into this library - such a great effort and truly appreciated.

Anyway, I'm trying to connect a PS3 via Bluetooth to USB shield and it seems to pair, but then stops at "Waiting for incoming connection request".

Here's the log:

~~~
PS3 Bluetooth Library Started
Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:13
Wait For Incoming Connection Request
** Unplug dongle, plug controller **
Dualshock 3 Controller Connected
Bluetooth Address was set to: 00:1A:7D:DA:71:13
** Unplug controller, plugin dongle **
Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:13
Wait For Incoming Connection Request              
** Nothing more after this **
~~~

The line "No response to HCI Reset" look suspicious but not sure what that means.

I've tried this with two dongles and both behave identically.  The second one I bought hoping it would work because it uses the same chip set as one listed on the compatible dongles list. ie: CSR 4.0 with CSR8510 chipset.  Purchased [here](http://www.ebay.com.au/itm/Mini-USB-2-0-Bluetooth-V4-0-Dongle-Wireless-Adapter-For-PC-Laptop-3Mbps-Speed/181602349327?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649).

Anything else I can try to get to the bottom of this.  (I'm a developer and happy to help investigate but not familiar with the library).  Any help greatly appreciated.

Brad

Lauszus, 2017-09-06:
@toptensoftware can you try to enable extra debugging: https://github.com/felis/USB_Host_Shield_2.0/blob/master/BTD.cpp#L20 and re-upload the code?

Lauszus, 2017-10-12:
Closing due to lack of response.

closed 2017-10-12 by Lauszus